isaac-sim/IsaacLab #1441: run isaacsim omni.isaac.sim.python.kit

Taiyu-Jin, 2024-11-20:
`(lab) terry@terry:~isaacsim omni.isaac.sim.python.kitit
[Info] [carb] Logging to file: /home/terry/.nvidia-omniverse/logs/Kit/Isaac-Sim/4.2/kit_20241120_135902.log
2024-11-20 05:59:02 [0ms] [Warning] [omni.kit.app.plugin] No crash reporter present, dumps uploading isn't available.
[0.077s] Failed to solve some dependencies locally, syncing with extension registry...
[0.078s] [ext: omni.kit.async_engine-0.0.1] startup
[0.081s] [ext: omni.client-1.2.0] startup
[0.100s] [ext: omni.kit.registry.nucleus-0.0.0] startup
syncing registry: 'https://ovextensionsprod.blob.core.windows.net/exts/kit/prod/106/shared/v2' Downloading 305 files...
2024-11-20 05:59:08 [6,472ms] [Error] [omni.ext.plugin] Failed to resolve extension dependencies. Failure hints:
	[omni.isaac.sim.python-4.2.0] dependency: 'omni.anim.graph.schema' = { version='^' } can't be satisfied. Available versions:
	 (none found)
 Platform incompatible packages:
	 - [omni.anim.graph.schema-106.3.0+106.3.0.lx64.d.cp310] (registry)
	 - [omni.anim.graph.schema-106.3.0+106.3.0.wx64.r.cp310] (registry)
	 - [omni.anim.graph.schema-106.3.0+106.3.0.lx64.r.cp310] (registry)
	 - [omni.anim.graph.schema-106.3.0+106.3.0.wx64.d.cp310] (registry)
	 - [omni.anim.graph.schema-106.2.0+106.2.0.wx64.d.cp310] (registry)
	 - [omni.anim.graph.schema-106.2.0+106.2.0.lx64.d.cp310] (registry)
	 - [omni.anim.graph.schema-106.2.0+106.2.0.wx64.r.cp310] (registry)
	 - [omni.anim.graph.schema-106.2.0+106.2.0.lx64.r.cp310] (registry)
 Synced registries:
	 - kit/default         : found 382 packages 
	 - kit/sdk             : found 320 packages 
	 - kit/community       : found 384 packages 

2024-11-20 05:59:08 [6,473ms] [Error] [omni.kit.app.plugin] Exiting app because of dependency solver failure...`
### I followed the install instructions,and ran "isaacsim omni.isaac.sim.python.kit",then it showed like this. How can I resolve it?

Andrew-Luo1, 2024-11-23:
What fixed this for me was the following:

```
 pip install isaacsim-extscache-physics==4.2.0.2 isaacsim-extscache-kit==4.2.0.2 isaacsim-extscache-^Ct-sdk==4.2.0.2 --extra-index-url https://pypi.nvidia.com
```

Good luck!

RandomOakForest, 2025-01-27:
It seems this is resolved. We'll close the issue for now, but feel free to reopen it may you need further help. Thanks.

closed 2025-01-27 by RandomOakForest