isaac-sim/IsaacLab #1449: How to achieve six degrees of freedom motion through state machine？

rese-eser, 2024-11-22:
Hello everyone! I am following Isaaclab's tutorial to reproduce the script related to lift_cube_ sm. py, but I have found that the angle of each step in the process of grabbing the cube by the robotic arm remains basically unchanged. In other words, I have changed the rotation angle of the cube about the z-axis. I hope that the end of the robotic arm can also calculate the corresponding rotation angle during the grabbing process to achieve more accurate grabbing.

I tried to add the pose quaternion of an object in the state machine, but the calculated action looks strange. Why is this?
[image]

[录屏 2024年11月22日 15时02分54秒.webm](https://github.com/user-attachments/assets/587da6f4-f02b-444a-9f81-09b6d196bff1)

rese-eser, 2024-11-23:
> It looks to me like the orientation of the manipulator is close but is running up against joint limits and it looks like there's something wrong with the desired_position calculation that's not in the code you shared.
> 
> Have you looked at the object following sample?
Thank you for your reply. What do I mean by implementing the six degree of freedom motion of the robotic arm in the lift cube sm script? Currently, the motion of the robotic arm seems to only change in the xyz position, while the rotation angle seems to be fixed.

RandomOakForest, 2025-01-27:
Thank you for following up. We ran into the holidays, our apologies for the delay. If you still have this issue, we may move this post into our _Discussions_ for follow up. Let us know if this would help. Thanks.

closed 2025-02-11 by RandomOakForest