isaac-sim/IsaacLab #1457: [Question] Im getting a gymnasium.error.NameNotFound, why is that?

bull-stack, 2024-11-24:
### Why is it that im getting this error, when i have added gym.register for the environment in the __init__.py 
 this is the error:
 
 Traceback (most recent call last):
  File "/media/bull/Data/Thesis/bin-picking-rl/isaacLab/bin_pick_sm.py", line 299, in <module>
    main()
  File "/media/bull/Data/Thesis/bin-picking-rl/isaacLab/bin_pick_sm.py", line 247, in main
    env_cfg: BinPickingEnvCfg = parse_env_cfg(
  File "/media/bull/Data/isaac-sim/pkg/IsaacLab/source/extensions/omni.isaac.lab_tasks/omni/isaac/lab_tasks/utils/parse_cfg.py", line 120, in parse_env_cfg
    cfg = load_cfg_from_registry(task_name, "env_cfg_entry_point")
  File "/media/bull/Data/isaac-sim/pkg/IsaacLab/source/extensions/omni.isaac.lab_tasks/omni/isaac/lab_tasks/utils/parse_cfg.py", line 55, in load_cfg_from_registry
    cfg_entry_point = gym.spec(task_name).kwargs.get(entry_point_key)
  File "/media/bull/Data/isaac-sim/pkg/isaac-sim-4.2.0/exts/omni.isaac.ml_archive/pip_prebundle/gymnasium/envs/registration.py", line 1029, in spec
    _check_version_exists(ns, name, version)
  File "/media/bull/Data/isaac-sim/pkg/isaac-sim-4.2.0/exts/omni.isaac.ml_archive/pip_prebundle/gymnasium/envs/registration.py", line 393, in _check_version_exists
    _check_name_exists(ns, name)
  File "/media/bull/Data/isaac-sim/pkg/isaac-sim-4.2.0/exts/omni.isaac.ml_archive/pip_prebundle/gymnasium/envs/registration.py", line 370, in _check_name_exists
    raise error.NameNotFound(
gymnasium.error.NameNotFound: Environment `Isaac-Bin-Picking-UR10-IK-Abs` doesn't exist.

RandomOakForest, 2024-12-06:
Thanks for posting this. Is the error found with Isaac Lab 1.3? Thanks

JoHo2410, 2024-12-06:
Yes, the error is found in IsaacLab 1.3. I did also solve it now: I put my main and my train function as well as the environment configuration in the same python-file. Separating the environment file in a env.py and the training into a train.py and importing the environment into the train.py I did not have a problem anymore. I can't explain, why it is like that but now it works...

psh9002, 2025-02-08:
@RandomOakForest 

https://isaac-sim.github.io/IsaacLab/main/source/tutorials/03_envs/modify_direct_rl_env.html

The same issue appears again in version 2.0.

RandomOakForest, 2025-02-11:
Have you followed the steps to [register the environment as described in this doc](https://isaac-sim.github.io/IsaacLab/main/source/tutorials/03_envs/register_rl_env_gym.html#tutorial-register-rl-env-gym)? Have you upgraded Isaac Sim as well to 4.5? It seems something is off in your registration or name conventions. Could you share relevant parts of your code? Thanks.

closed 2025-02-15 by RandomOakForest