isaac-sim/IsaacLab #1461: [Question] Are there any reinforcement learning libraries recommended by Isaac Lab?

H-Hisamichi, 2024-11-26:
Hello everyone,

I was working on a reinforcement learning task and wondered which RL libraries would be suitable for Isaac Lab.
Currently, we are using rl-games based on the ANYmal C demonstration code. However, when reading the tutorial on applying policies to Isaac Sim, I noticed that rsl-rl is mentioned frequently. It also states that for libraries other than rsl-rl, users should refer to their respective documentation. Does this mean that rsl-rl is the standard library for use with Isaac Sim and Isaac Lab?

thank you!

YannickP99, 2024-11-27:
Hey,
in this [tutorial](https://isaac-sim.github.io/IsaacLab/main/source/tutorials/03_envs/run_rl_training.html#) they use [Stable-Baselines3](https://stable-baselines3.readthedocs.io/en/master/), but also mention more references for possible libraries. I think at the end it depends on your preference, but it might be easier to follow a tutorial in the beginning to get familiar with IsaacLab.

H-Hisamichi, 2024-11-27:
Hi @YannickP99,

Thanks for the advice!
The “Deploying Policies in Isaac Sim” tutorial uses rsl-rl, so I will use this RL library.

closed 2024-11-27 by H-Hisamichi

YannickP99, 2024-11-28:
Hey, I just found this page comparing the different libraries. I think this can be interesting for you: https://isaac-sim.github.io/IsaacLab/main/source/overview/reinforcement-learning/rl_frameworks.html