isaac-sim/IsaacLab #1462: [Bug Report] Failing to download assets from nucleus cloud

YannickP99, 2024-11-26:
### Describe the bug
I installed IsaacLab according to the documentation using pip. When trying to run simulations, IsaacLab fails to download the assets from the Nucleus server (I am currently located in China, but also with VPN I cannot access the assets; by using wget and the url of the error message I am able to download the assets manually without VPN). Therefore I think that it is not related to any bans.

Both "Verifying the Isaac * installation" work fine.

### Steps to reproduce
Running: 
```
./isaaclab.sh -p source/standalone/tutorials/03_envs/create_quadruped_base_env.py --num_envs 32
```
```
[INFO] Using python from: /home/yannick/anaconda3/envs/isaaclab/bin/python                                                             
[INFO][AppLauncher]: Loading experience file: /home/yannick/IsaacLab/source/apps/isaaclab.python.kit
[Warning] [omni.isaac.kit.simulation_app] Modules: ['omni.kit_app'] were loaded before SimulationApp was started and might not be loaded correctly.
[Warning] [omni.isaac.kit.simulation_app] Please check to make sure no extra omniverse or pxr modules are imported before the call to SimulationApp(...)
Loading user config located at: '/home/yannick/anaconda3/envs/isaaclab/lib/python3.10/site-packages/omni/data/Kit/Isaac-Sim/4.2/user.config.json'
[Info] [carb] Logging to file: /home/yannick/anaconda3/envs/isaaclab/lib/python3.10/site-packages/omni/logs/Kit/Isaac-Sim/4.2/kit_20241126_114823.log
2024-11-26 03:48:23 [0ms] [Warning] [omni.kit.app.plugin] No crash reporter present, dumps uploading isn't available.

|---------------------------------------------------------------------------------------------|
| Driver Version: 550.120       | Graphics API: Vulkan
|=============================================================================================|
| GPU | Name                             | Active | LDA | GPU Memory | Vendor-ID | LUID       |
|     |                                  |        |     |            | Device-ID | UUID       |
|     |                                  |        |     |            | Bus-ID    |            |
|---------------------------------------------------------------------------------------------|
| 0   | NVIDIA GeForce RTX 3090          | Yes: 0 |     | 24822   MB | 10de      | 0          |
|     |                                  |        |     |            | 2204      | 44fed211.. |
|     |                                  |        |     |            | 1         |            |
|=============================================================================================|
| OS: 22.04.5 LTS (Jammy Jellyfish) ubuntu, Version: 22.04.5, Kernel: 6.8.0-49-generic
| XServer Vendor: The X.Org Foundation, XServer Version: 12101004 (1.21.1.4)
| Processor: 13th Gen Intel(R) Core(TM) i7-13790F | Cores: 16 | Logical: 32
|---------------------------------------------------------------------------------------------|
| Total Memory (MB): 31904 | Free Memory: 25963
| Total Page/Swap (MB): 2047 | Free Page/Swap: 2047
|---------------------------------------------------------------------------------------------|
2024-11-26 03:48:23 [394ms] [Warning] [gpu.foundation.plugin] IOMMU is enabled.
2024-11-26 03:48:26 [3,695ms] [Warning] [omni.replicator.core.scripts.annotators] Annotator PostProcessDispatch is already registered, overwriting annotator template
2024-11-26 03:48:27 [3,856ms] [Warning] [omni.kit.widget.cache_indicator.cache_state_menu] Unable to detect Omniverse Cache Server. Consider installing it for better IO performance.
2024-11-26 03:48:29 [6,678ms] [Warning] [omni.isaac.lab.envs.manager_based_env] Seed not set for the environment. The environment creation may not be deterministic.
[INFO]: Base environment:
	Environment device    : cuda:0
	Environment seed      : None
	Physics step-size     : 0.005
	Rendering step-size   : 0.005
	Environment step-size : 0.02
2024-11-26 03:48:30 [6,732ms] [Warning] [omni.isaac.lab.envs.manager_based_env] The render interval (1) is smaller than the decimation (4). Multiple render calls will happen for each environment step. If this is not intended, set the render interval to be equal to the decimation.
2024-11-26 03:55:08 [405,384ms] [Warning] [omni.usd] Warning: in _ReportErrors at line 2890 of /builds/omniverse/usd-ci/USD/pxr/usd/usd/stage.cpp -- In </World/envs/env_0/Robot/base/collisions>: Could not open asset @http://omniverse-content-production.s3-us-west-2.amazonaws.com/Assets/Isaac/4.2/Isaac/IsaacLab/Robots/ANYbotics/ANYmal-C/Props/instanceable_meshes.usd@ for reference introduced by @http://omniverse-content-production.s3-us-west-2.amazonaws.com/Assets/Isaac/4.2/Isaac/IsaacLab/Robots/ANYbotics/ANYmal-C/anymal_c.usd@</anymal/base/collisions>. (recomposing stage on stage @anon:0x16ec4bc0:World0.usd@ <0x15192bc0>)

2024-11-26 03:55:08 [405,384ms] [Warning] [omni.usd] Warning: in _ReportErrors at line 2890 of /builds/omniverse/usd-ci/USD/pxr/usd/usd/stage.cpp -- In </World/envs/env_0/Robot/LF_HIP/visuals>: Could not open asset @http://omniverse-content-production.s3-us-west-2.amazonaws.com/Assets/Isaac/4.2/Isaac/IsaacLab/Robots/ANYbotics/ANYmal-C/Props/instanceable_meshes.usd@ for reference introduced by @http://omniverse-content-production.s3-us-west-2.amazonaws.com/Assets/Isaac/4.2/Isaac/IsaacLab/Robots/ANYbotics/ANYmal-C/anymal_c.usd@</anymal/LF_HIP/visuals>. (recomposing stage on stage @anon:0x16ec4bc0:World0.usd@ <0x15192bc0>)

```

When downloading the assets manually and copying them into IsaacLab/source/extensions/omni.isaac.lab_assets/data/Robots/ and adapting the code to use ISAACLAB_ASSETS_DATA_DIR instead of ISAACLAB_NUCLEUS_DIR
```
ANYMAL_B_CFG = ArticulationCfg(
    spawn=sim_utils.UsdFileCfg(
        # usd_path=f"{ISAACLAB_NUCLEUS_DIR}/Robots/ANYbotics/ANYmal-B/anymal_b.usd",
        usd_path=f"{ISAACLAB_ASSETS_DATA_DIR}/Robots/ANYbotics/ANYmal-B/anymal_b.usd",
        activate_contact_sensors=True,
        rigid_props=sim_utils.RigidBodyPropertiesCfg(
            disable_gravity=False,
            retain_accelerations=False,
            linear_damping=0.0,
            angular_damping=0.0,
            max_linear_velocity=1000.0,
            max_angular_velocity=1000.0,
            max_depenetration_velocity=1.0,
        ),
[... 52 lines cut ...]
  File "/home/yannick/IsaacLab/source/extensions/omni.isaac.lab/omni/isaac/lab/assets/asset_base.py", line 265, in _initialize_callback
  File "/home/yannick/IsaacLab/source/extensions/omni.isaac.lab/omni/isaac/lab/assets/articulation/articulation.py", line 948, in _initialize_impl
  File "/home/yannick/IsaacLab/source/extensions/omni.isaac.lab/omni/isaac/lab/assets/articulation/articulation.py", line 1132, in _process_actuators_cfg
  File "/home/yannick/IsaacLab/source/extensions/omni.isaac.lab/omni/isaac/lab/actuators/actuator_net.py", line 49, in __init__
  File "/home/yannick/IsaacLab/source/extensions/omni.isaac.lab/omni/isaac/lab/utils/assets.py", line 125, in read_file
  File "/home/yannick/anaconda3/envs/isaaclab/lib/python3.10/site-packages/omni/extscore/omni.client/omni/client/__init__.py", line 610, in read_file

2024-11-26 04:06:02 [151,140ms] [Error] [omni.hydra] [UsdToMdl] Prim '/__Prototype_16/Looks/material_shank/Shader' parameter 'diffuse_texture': References an asset that can not be found: './materials/shank.jpg'
2024-11-26 04:06:02 [151,140ms] [Error] [omni.hydra] [UsdToMdl] Prim '/__Prototype_16/Looks/material_shank/Shader' parameter 'diffuse_texture': References an asset that can not be found: './materials/shank.jpg'
2024-11-26 04:06:02 [151,379ms] [Error] [omni.hydra] [UsdToMdl] Prim '/__Prototype_10/Looks/material_drive/Shader' parameter 'diffuse_texture': References an asset that can not be found: './materials/drive.jpg'
2024-11-26 04:06:02 [151,379ms] [Error] [omni.hydra] [UsdToMdl] Prim '/__Prototype_10/Looks/material_drive/Shader' parameter 'diffuse_texture': References an asset that can not be found: './materials/drive.jpg'

``

-->

### System Info

Describe the characteristic of your environment:
- Commit: 4d99147
- Isaac Sim Version: 4.2.0.2
- OS: Ubuntu 22.04.5 LTS
- GPU: GeForce RTX 3090
- CUDA: 12.4
- GPU Driver: 550.120

closed 2024-11-28 by YannickP99

zkj623, 2025-01-17:
Hello! How did you solve this problem?

YannickP99, 2025-01-22:
I think for me it was a VPN problem and after I downloaded the data once it worked.