isaac-sim/IsaacLab #1483: 【WinError 206】Issue on page /source/setup/installation/pip_installation.html

181404010226, 2024-11-30:
I installed isaaclab on windows and still encountered errors with long path support enabled

```
(isaaclab) PS D:\VisutalStudio\repository\ROSdevelop> reg query "HKEY_LOCAL_MACHINE\SYSTEM\CurrentControlSet\Control\FileSystem" /v LongPathsEnabled

HKEY_LOCAL_MACHINE\SYSTEM\CurrentControlSet\Control\FileSystem
    LongPathsEnabled    REG_DWORD    0x1

(isaaclab) PS D:\VisutalStudio\repository\ROSdevelop> 
```

The entire log content below:

[log.txt](https://github.com/user-attachments/files/17964637/log.txt)

I also tried running on a short path, but that didn't seem to work

```
2024-11-30 08:23:34 [5,977ms] [Error] [omni.ext.plugin] [ext: omni.kit.material.library-1.5.6] Failed to startup python extension.
Windows fatal exception: access violation

Current thread 0x000026c4 (most recent call first):
  File "D:\conda\envs\isaaclabmini\Lib\site-packages\isaacsim\exts\omni.isaac.kit\omni\isaac\kit\simulation_app.py", line 411 in _start_app
  File "D:\conda\envs\isaaclabmini\Lib\site-packages\isaacsim\exts\omni.isaac.kit\omni\isaac\kit\simulation_app.py", line 200 in __init__
  File "D:\IsaacLab\IsaacLab\source\extensions\omni.isaac.lab\omni\isaac\lab\app\app_launcher.py", line 607 in _create_app
  File "D:\IsaacLab\IsaacLab\source\extensions\omni.isaac.lab\omni\isaac\lab\app\app_launcher.py", line 113 in __init__
  File "D:\IsaacLab\IsaacLab\source\standalone\tutorials\00_sim\create_empty.py", line 29 in <module>
(isaaclabmini) PS D:\IsaacLab\IsaacLab> 
```

181404010226, 2024-12-02:
@shanereetz My conda environment is already the root directory of disk D, perhaps a better way is to find a solution that can run on a long path, which the window platform always faces.

And then I tried running with administrator:
[image]

RandomOakForest, 2024-12-06:
This seems to be a cache issue or related to previous installations of Isaac Sim in your system. Could you try a clean reinstall?

closed 2025-02-11 by RandomOakForest