isaac-sim/IsaacLab #1487: [Question] How to do blocking control?

arhanjain, 2024-12-01:
### Question

Basic questions, related to robot learning, that are not bugs or feature requests will be closed without reply, because GitHub issues are not an appropriate venue for these.

Advanced/nontrivial questions, especially in areas where documentation is lacking, are very much welcome.

For questions that are related to running and understanding Isaac Sim, please post them at the official [Isaac Sim forums](https://docs.omniverse.nvidia.com/app_isaacsim/app_isaacsim/isaac_sim_forums.html).

RandomOakForest, 2024-12-06:
Thanks for asking. Could you elaborate with an example of what you would like to achieve? Thanks

arhanjain, 2024-12-07:
> Thanks for asking. Could you elaborate with an example of what you would like to achieve? Thanks

Basically most real-world robot stacks for manipulation have a step function with a blocking option which ensures the robot reaches its desired joint positions or end effector pose, before being able to run the next step function. Presumably this could help with sim2real gap.

RandomOakForest, 2025-02-15:
Thanks for clarifying. It doesn't seem we have a standard way of implementing blocking control. If you have implemented this already, we can only encourage you to share your approach in our [show-and-tell section](https://github.com/isaac-sim/IsaacLab/discussions/categories/show-and-tell).

closed 2025-02-19 by RandomOakForest