isaac-sim/IsaacLab #1492: [Question] Is the Terrain Curriculum Generation Function Working?

H-Hisamichi, 2024-12-03:
Dear All,

I have a question about the terrain curriculaum generation function.

My `TerrainGeneratorCfg()` config:
```
ROUGH_TERRAINS_walk_CFG = TerrainGeneratorCfg(
    curriculum=True,
    size=(8.0, 8.0),
    border_width=20.0,
    num_rows=15,
    num_cols=25,
    horizontal_scale=0.1,
    vertical_scale=0.005,
    slope_threshold=0.75,
    use_cache=False,
    difficulty_range=(0.0, 1.0), # add
    sub_terrains={
        "pyramid_stairs": terrain_gen.MeshPyramidStairsTerrainCfg(
            proportion=0.2,
            step_height_range=(0.05, 0.1),
            step_width=0.3,
            platform_width=3.0,
            border_width=1.0,
            holes=False,
        ),
        "pyramid_stairs_inv": terrain_gen.MeshInvertedPyramidStairsTerrainCfg(
            proportion=0.2,
            step_height_range=(0.05, 0.1),
            step_width=0.3,
            platform_width=3.0,
            border_width=1.0,
            holes=False,
        ),
        "boxes": terrain_gen.MeshRandomGridTerrainCfg(
            proportion=0.2, grid_width=0.45, grid_height_range=(0.05, 0.2), platform_width=2.0
        ),
        "random_rough": terrain_gen.HfRandomUniformTerrainCfg(
            proportion=0.2, noise_range=(0.02, 0.10), noise_step=0.02, border_width=0.25
        ),
        "hf_pyramid_slope": terrain_gen.HfPyramidSlopedTerrainCfg(
            proportion=0.1, slope_range=(0.0, 0.4), platform_width=2.0, border_width=0.25
        ),
        "hf_pyramid_slope_inv": terrain_gen.HfInvertedPyramidSlopedTerrainCfg(
            proportion=0.1, slope_range=(0.0, 0.4), platform_width=2.0, border_width=0.25
        ),
    },
)
```

Image of the terrain just after RL started:
[image]

This image seems to show that difficult terrain was generated from the beginning.
Is there something wrong with my configs?

Thank you!

RandomOakForest, 2024-12-06:
Thank you for posting this. Could you tell us more about which is your desired output? Thanks

H-Hisamichi, 2024-12-07:
Hello @RandomOakForest,

I want to train the policy step by step, from flat terrain (or slight steps) to rough terrain as defined in TerrainGeneratorCfg(). According to the documentation, the curriculum generation feature seems to be working, but the early training renderings don't show this.

H-Hisamichi, 2024-12-10:
Resolved by migrating to the manager-based code.
Thank you very much!

closed 2024-12-10 by H-Hisamichi