isaac-sim/IsaacLab #1493: [Bug Report] TGS Velocity Discrepancy

ori-gadot, 2024-12-03:
### Bug Description
The observed joint velocity is incorrect.

### Steps to Reproduce

1. Use the provided minimal single-file script to reproduce the issue.
2. The script runs the CartPole environment with joint position as actions and prints the joint position, joint velocity, and the numerical derivative of the position.
3. The CartPole moves from its initial pose to the target pose and remains stationary.
4. Despite reaching a static state, the reported joint velocity is a positive and relatively large value.

### Additional Context

Switching the solver type from TGS (1) to PGS (0) in the following line shows a notable difference in accuracy:
`env_cfg.sim.physx.solver_type = 0`

Using PGS yields more accurate velocity reporting in the static state.

### Acceptance Criteria
When the system is in a static state, the reported joint velocity should be as close to zero as possible.

### Code

```
from omni.isaac.lab.app import AppLauncher

app_launcher = AppLauncher(headless=True)
simulation_app = app_launcher.app

import torch
import omni.isaac.lab.envs.mdp as mdp
from omni.isaac.lab.envs import ManagerBasedEnv, ManagerBasedEnvCfg
from omni.isaac.lab.managers import ObservationGroupCfg as ObsGroup
from omni.isaac.lab.managers import ObservationTermCfg as ObsTerm
from omni.isaac.lab.utils import configclass
from omni.isaac.lab_tasks.manager_based.classic.cartpole.cartpole_env_cfg import CartpoleSceneCfg
@configclass
class ActionsCfg:
    joint_pos = mdp.JointPositionActionCfg(asset_name="robot", joint_names=["cart_to_pole"], scale=1.0, use_default_offset=False)

@configclass
class ObservationsCfg:
    @configclass
    class PolicyCfg(ObsGroup):
        joint_pos_rel = ObsTerm(func=mdp.joint_pos_rel)
        joint_vel_rel = ObsTerm(func=mdp.joint_vel_rel)

        def __post_init__(self) -> None:
            self.enable_corruption = False
            self.concatenate_terms = False

    policy: PolicyCfg = PolicyCfg()

@configclass
class CartpoleEnvCfg(ManagerBasedEnvCfg):
    scene = CartpoleSceneCfg(num_envs=16, env_spacing=2.5)
    observations = ObservationsCfg()
    actions = ActionsCfg()

    def __post_init__(self):
        self.viewer.eye = [4.5, 0.0, 6.0]
        self.viewer.lookat = [0.0, 0.0, 2.0]
        self.decimation = 1
        self.sim.dt = 0.005

def main():
    env_cfg = CartpoleEnvCfg()
    env_cfg.scene.robot.actuators["pole_actuator"].stiffness = 100.0
    env_cfg.scene.robot.actuators["pole_actuator"].damping = 5.0
    env_cfg.sim.physx.solver_type = 1
    env = ManagerBasedEnv(cfg=env_cfg)
    obs, _ = env.reset()
    prev_pos = curr_pos = obs["policy"]["joint_pos_rel"][0][1].item()

    while simulation_app.is_running():
        with torch.inference_mode():
            joint_pos = torch.ones_like(env.action_manager.action) * 5.0
            obs, _ = env.step(joint_pos)
            curr_pos = obs["policy"]["joint_pos_rel"][0][1].item()
            curr_vel = obs["policy"]["joint_vel_rel"][0][1].item()

            dt = env_cfg.sim.dt * env_cfg.decimation
            numerical_vel = (curr_pos - prev_pos) / dt
            print(f"[Env 0]: Pole joint pos: {curr_pos:.8f}, numerical velocity: {numerical_vel:.8f}, measured velocity: {curr_vel:.8f}")
            prev_pos = curr_pos

    env.close()

if __name__ == "__main__":
    main()
    simulation_app.close()

```

RandomOakForest, 2024-12-06:
Thank you for posting this. We will review and follow up. Can you confirm the versions of Isaac Lab and Isaac Sim you are using? Thanks.

ori-gadot, 2024-12-06:
Isaac Sim 4.2.0
Isaac Lab 1.3.0

ori-gadot, 2024-12-23:
Hi @RandomOakForest , I was wondering if there are any updates on this bug report.
I'm also curious about why it was tagged as a question, as it seems to be a bug rather than a question.
Thank you!

RandomOakForest, 2024-12-23:
Thank you for following up. We are experiencing some delay with the new release and the holidays. The question tag has been removed. We will aim to repro this on our side. Thanks!