isaac-sim/IsaacLab #1502: [Bug Report] Modifying an existing Direct RL Environment tutorial error

jh-s5974, 2024-12-05:
### Describe the bug

I'm following the Modifying an existing Direct RL Environment tutorial and I'm getting an error.

i couldn't find H1EnvCfg.

and Can't get rid of error "Could not find category 'Replicator:Annotators' for removal", "Could not find category 'Replicator:Core' for removal", "Recursive unloadAllPlugins() detected!"

How can I complete the tutorial without errors?

### Steps to reproduce

```
./isaaclab.sh -p source/standalone/workflows/rl_games/train.py --task Isaac-H1-Direct-v0
[INFO] Using python from: /home/jh/anaconda3/envs/lab/bin/python                                                                                 
[INFO][AppLauncher]: Loading experience file: /home/jh/IsaacLab/source/apps/isaaclab.python.kit
Loading user config located at: '/home/jh/.local/share/ov/pkg/isaac-sim-4.2.0/kit/data/Kit/Isaac-Sim/4.2/user.config.json'
[Info] [carb] Logging to file: /home/jh/.local/share/ov/pkg/isaac-sim-4.2.0/kit/logs/Kit/Isaac-Sim/4.2/kit_20241206_065701.log
2024-12-05 21:57:01 [0ms] [Warning] [omni.kit.app.plugin] No crash reporter present, dumps uploading isn't available.

|---------------------------------------------------------------------------------------------|
| Driver Version: 535.183.01    | Graphics API: Vulkan
|=============================================================================================|
| GPU | Name                             | Active | LDA | GPU Memory | Vendor-ID | LUID       |
|     |                                  |        |     |            | Device-ID | UUID       |
|     |                                  |        |     |            | Bus-ID    |            |
|---------------------------------------------------------------------------------------------|
| 0   | NVIDIA GeForce RTX 4070 SUPER    | Yes: 0 |     | 12528   MB | 10de      | 0          |
|     |                                  |        |     |            | 2783      | 259267ba.. |
|     |                                  |        |     |            | 1         |            |
|=============================================================================================|
| OS: 20.04.6 LTS (Focal Fossa) ubuntu, Version: 20.04.6, Kernel: 5.15.0-125-generic
| XServer Vendor: The X.Org Foundation, XServer Version: 12013000 (1.20.13.0)
| Processor: AMD Ryzen 5 7500F 6-Core Processor | Cores: 6 | Logical: 12
|---------------------------------------------------------------------------------------------|
| Total Memory (MB): 31726 | Free Memory: 23390
| Total Page/Swap (MB): 30517 | Free Page/Swap: 30512
|---------------------------------------------------------------------------------------------|
2024-12-05 21:57:02 [792ms] [Warning] [gpu.foundation.plugin] IOMMU is enabled.
2024-12-05 21:57:07 [6,031ms] [Warning] [omni.replicator.core.scripts.annotators] Annotator PostProcessDispatch is already registered, overwriting annotator template
Traceback (most recent call last):
  File "/home/jh/IsaacLab/source/standalone/workflows/rl_games/train.py", line 178, in <module>
    main()
  File "/home/jh/IsaacLab/source/extensions/omni.isaac.lab_tasks/omni/isaac/lab_tasks/utils/hydra.py", line 79, in wrapper
    env_cfg, agent_cfg = register_task_to_hydra(task_name, agent_cfg_entry_point)
  File "/home/jh/IsaacLab/source/extensions/omni.isaac.lab_tasks/omni/isaac/lab_tasks/utils/hydra.py", line 42, in register_task_to_hydra
    env_cfg = load_cfg_from_registry(task_name, "env_cfg_entry_point")
  File "/home/jh/IsaacLab/source/extensions/omni.isaac.lab_tasks/omni/isaac/lab_tasks/utils/parse_cfg.py", line 87, in load_cfg_from_registry
    cfg_cls = getattr(mod, attr_name)
AttributeError: module 'omni.isaac.lab_tasks.direct.humanoid' has no attribute 'H1EnvCfg'
2024-12-05 21:57:11 [9,781ms] [Warning] [omni.graph.core.plugin] Could not find category 'Replicator:Annotators' for removal
2024-12-05 21:57:11 [9,781ms] [Warning] [omni.graph.core.plugin] Could not find category 'Replicator:Annotators' for removal
2024-12-05 21:57:11 [9,781ms] [Warning] [omni.graph.core.plugin] Could not find category 'Replicator:Annotators' for removal
2024-12-05 21:57:11 [9,781ms] [Warning] [omni.graph.core.plugin] Could not find category 'Replicator:Annotators' for removal
2024-12-05 21:57:11 [9,781ms] [Warning] [omni.graph.core.plugin] Could not find category 'Replicator:Annotators' for removal
2024-12-05 21:57:11 [9,781ms] [Warning] [omni.graph.core.plugin] Could not find category 'Replicator:Annotators' for removal
2024-12-05 21:57:11 [9,781ms] [Warning] [omni.graph.core.plugin] Could not find category 'Replicator:Annotators' for removal
2024-12-05 21:57:11 [9,781ms] [Warning] [omni.graph.core.plugin] Could not find category 'Replicator:Annotators' for removal
2024-12-05 21:57:11 [9,781ms] [Warning] [omni.graph.core.plugin] Could not find category 'Replicator:Annotators' for removal
2024-12-05 21:57:11 [9,781ms] [Warning] [omni.graph.core.plugin] Could not find category 'Replicator:Annotators' for removal
2024-12-05 21:57:11 [9,781ms] [Warning] [omni.graph.core.plugin] Could not find category 'Replicator:Annotators' for removal
2024-12-05 21:57:11 [9,781ms] [Warning] [omni.graph.core.plugin] Could not find category 'Replicator:Annotators' for removal
2024-12-05 21:57:11 [9,781ms] [Warning] [omni.graph.core.plugin] Could not find category 'Replicator:Core' for removal
2024-12-05 21:57:11 [9,817ms] [Warning] [carb] Recursive unloadAllPlugins() detected!

```

### System Info

Describe the characteristic of your environment:

- Commit: [???]
- Isaac Sim Version: [4.2.0]
- OS: [Ubuntu 20.04]
- GPU: [RTX 4070 Super]
- CUDA: [11.8]
- GPU Driver: [535.183.01]

RandomOakForest, 2024-12-06:
Thanks for posting this. Can you verify your installation of Isaac Lab 1.3 (latest) went well? also, if you are following the steps to properly register the environment, for example, could you post the contents of:
```
source/extensions/omni.isaac.lab_tasks/omni/isaac/lab_tasks/direct/humanoid/__init__.py
```
Thank you.

jh-s5974, 2024-12-07:
# Copyright (c) 2022-2024, The Isaac Lab Project Developers.
# All rights reserved.
#
# SPDX-License-Identifier: BSD-3-Clause

"""
Humanoid locomotion environment.
"""

import gymnasium as gym

from . import agents

##
# Register Gym environments.
##

gym.register(
    id="Isaac-H1-Direct-v0",
    entry_point="omni.isaac.lab_tasks.direct.humanoid:H1Env",
    disable_env_checker=True,
    kwargs={
        "env_cfg_entry_point": "omni.isaac.lab_tasks.direct.humanoid:H1EnvCfg",
        "rl_games_cfg_entry_point": f"{agents.__name__}:rl_games_ppo_cfg.yaml",
        "rsl_rl_cfg_entry_point": f"{agents.__name__}.rsl_rl_ppo_cfg:HumanoidPPORunnerCfg",
        "skrl_cfg_entry_point": f"{agents.__name__}:skrl_ppo_cfg.yaml",
    },
)

this code is __init__.py

and

# Copyright (c) 2022-2024, The Isaac Lab Project Developers.
# All rights reserved.
#
# SPDX-License-Identifier: BSD-3-Clause

from __future__ import annotations

from omni.isaac.lab_assets import HUMANOID_CFG

import omni.isaac.lab.sim as sim_utils
from omni.isaac.lab.assets import ArticulationCfg
from omni.isaac.lab.envs import DirectRLEnvCfg
from omni.isaac.lab.scene import InteractiveSceneCfg
from omni.isaac.lab.sim import SimulationCfg
from omni.isaac.lab.terrains import TerrainImporterCfg
from omni.isaac.lab.utils import configclass

from omni.isaac.lab_tasks.direct.locomotion.locomotion_env import LocomotionEnv

@configclass
class H1EnvCfg(DirectRLEnvCfg):
    # env
    episode_length_s = 15.0
    decimation = 2
    action_scale = 1.0
    action_space = 19
    observation_space = 69
    state_space = 0

    # simulation
    sim: SimulationCfg = SimulationCfg(dt=1 / 120, render_interval=decimation)
    terrain = TerrainImporterCfg(
        prim_path="/World/ground",
        terrain_type="plane",
        collision_group=-1,
        physics_material=sim_utils.RigidBodyMaterialCfg(
            friction_combine_mode="average",
            restitution_combine_mode="average",
            static_friction=1.0,
            dynamic_friction=1.0,
            restitution=0.0,
        ),
        debug_vis=False,
    )

    # scene
    scene: InteractiveSceneCfg = InteractiveSceneCfg(num_envs=4096, env_spacing=4.0, replicate_physics=True)

    # robot
    robot: ArticulationCfg = HUMANOID_CFG.replace(
        prim_path="/World/envs/env_.*/Robot",
        spawn=sim_utils.UsdFileCfg(
            usd_path="~/IsaacLab/source/extensions/omni.isaac.lab_assets/data/Robots/Unitree/h1.usd",
            )
        )
    joint_gears: list = [
        67.5000,  # lower_waist
        67.5000,  # lower_waist
        67.5000,  # right_upper_arm
        67.5000,  # right_upper_arm
        67.5000,  # left_upper_arm
        67.5000,  # left_upper_arm
        67.5000,  # pelvis
        45.0000,  # right_lower_arm
        45.0000,  # left_lower_arm
        45.0000,  # right_thigh: x
        135.0000,  # right_thigh: y
        45.0000,  # right_thigh: z
        45.0000,  # left_thigh: x
        135.0000,  # left_thigh: y
        45.0000,  # left_thigh: z
        90.0000,  # right_knee
        90.0000,  # left_knee
        22.5,  # right_foot
        22.5,  # right_foot
        22.5,  # left_foot
        22.5,  # left_foot
    ]

    heading_weight: float = 0.5
    up_weight: float = 0.1

    energy_cost_scale: float = 0.05
    actions_cost_scale: float = 0.01
    alive_reward_scale: float = 2.0
    dof_vel_scale: float = 0.1

    death_cost: float = -1.0
    termination_height: float = 0.8

    angular_velocity_scale: float = 0.25
    contact_force_scale: float = 0.01

class H1Env(LocomotionEnv):
    cfg: H1EnvCfg

    def __init__(self, cfg: H1EnvCfg, render_mode: str | None = None, **kwargs):
        super().__init__(cfg, render_mode, **kwargs)

this is h1_env.py

please check it

thank you

jh-s5974, 2024-12-09:
I solved this problem

Thank you

closed 2024-12-09 by jh-s5974

IonaGuyomarch, 2024-12-09:
Hey @jh-s5974 , can you explin how you solved this issue please? I'm also stuck there

jh-s5974, 2024-12-09:
I add this import text.

```
from .h1_env import H1Env, H1EnvCfg
```

IonaGuyomarch, 2024-12-09:
Thanks a lot for your help!! ^^

jh-s5974, 2024-12-09:
you're welcome

panhainan2020, 2024-12-10:
Isaac lab uses docker to install the configuration. I modified the existing Direct RL environment according to the tutorial, and the following error occurred. Has anyone encountered similar problems?
```
isaaclab -p source/standalone/workflows/rl_games/train.py --task Isaac-H1-Direct-v0 --headless
[INFO] Using python from: /workspace/isaaclab/_isaac_sim/python.sh                                     
[INFO][AppLauncher]: Loading experience file: /workspace/isaaclab/source/apps/isaaclab.python.headless.kit
Loading user config located at: '/isaac-sim/kit/data/Kit/Isaac-Sim/4.2/user.config.json'
[Info] [carb] Logging to file: /isaac-sim/kit/logs/Kit/Isaac-Sim/4.2/kit_20241210_174805.log
2024-12-10 17:48:05 [0ms] [Warning] [omni.kit.app.plugin] No crash reporter present, dumps uploading isn't available.

|---------------------------------------------------------------------------------------------|
| Driver Version: 560.35.03     | Graphics API: Vulkan
|=============================================================================================|
| GPU | Name                             | Active | LDA | GPU Memory | Vendor-ID | LUID       |
|     |                                  |        |     |            | Device-ID | UUID       |
|     |                                  |        |     |            | Bus-ID    |            |
|---------------------------------------------------------------------------------------------|
| 0   | NVIDIA GeForce RTX 4090 D        | Yes: 0 |     | 24810   MB | 10de      | 0          |
|     |                                  |        |     |            | 2685      | 91299c12.. |
|     |                                  |        |     |            | 1         |            |
|=============================================================================================|
| OS: 22.04.3 LTS (Jammy Jellyfish) ubuntu, Version: 22.04.3, Kernel: 5.15.0-67-generic
| XServer Vendor: The X.Org Foundation, XServer Version: 12013000 (1.20.13.0)
| Processor: 11th Gen Intel(R) Core(TM) i7-11700F @ 2.50GHz | Cores: 8 | Logical: 16
|---------------------------------------------------------------------------------------------|
| Total Memory (MB): 64031 | Free Memory: 56984
| Total Page/Swap (MB): 15624 | Free Page/Swap: 15624
|---------------------------------------------------------------------------------------------|
Traceback (most recent call last):
  File "/workspace/isaac_lab/source/standalone/workflows/rl_games/train.py", line 178, in <module>
    main()
  File "/workspace/isaaclab/source/extensions/omni.isaac.lab_tasks/omni/isaac/lab_tasks/utils/hydra.py", line 79, in wrapper
    env_cfg, agent_cfg = register_task_to_hydra(task_name, agent_cfg_entry_point)
  File "/workspace/isaaclab/source/extensions/omni.isaac.lab_tasks/omni/isaac/lab_tasks/utils/hydra.py", line 42, in register_task_to_hydra
    env_cfg = load_cfg_from_registry(task_name, "env_cfg_entry_point")
  File "/workspace/isaaclab/source/extensions/omni.isaac.lab_tasks/omni/isaac/lab_tasks/utils/parse_cfg.py", line 55, in load_cfg_from_registry
    cfg_entry_point = gym.spec(task_name).kwargs.get(entry_point_key)
  File "/isaac-sim/exts/omni.isaac.ml_archive/pip_prebundle/gymnasium/envs/registration.py", line 1029, in spec
    _check_version_exists(ns, name, version)
  File "/isaac-sim/exts/omni.isaac.ml_archive/pip_prebundle/gymnasium/envs/registration.py", line 393, in _check_version_exists
    _check_name_exists(ns, name)
  File "/isaac-sim/exts/omni.isaac.ml_archive/pip_prebundle/gymnasium/envs/registration.py", line 370, in _check_name_exists
    raise error.NameNotFound(
gymnasium.error.NameNotFound: Environment `Isaac-H1-Direct` doesn't exist. Did you mean: `Isaac-Ant-Direct`?
2024-12-10 17:48:10 [4,838ms] [Warning] [carb] Recursive unloadAllPlugins() detected!
```

jh-s5974, 2024-12-13:
Did you solve it?

I think this error can be solved by changing the id in init.py

Sorry I saw your comment late