isaac-sim/IsaacLab #1506: [Question] Problem with Installation

anyRandomHuman, 2024-12-06:
### Question
I have posted with question in the forum [https://forums.developer.nvidia.com/t/isaaclab-starts-but-does-not-render/312891/13](url).
And they recommneded me to ask for help here.
Details are stated in the link above.
If anyone knows how to solve the problem or knows how to install isaaclab to a working isaacsim installed via omniverse, please help me with this.

RandomOakForest, 2024-12-06:
Thank you for posting this. The note lists Isaac Lab Version 1.2. Could you try the latest release v1.3 of Isaac Lab? Thank you.

anyRandomHuman, 2024-12-06:
I have just reinstalled isaacsim through terminal and i already got errors while verifying its installation.
After runnning `isaacsim omni.isaac.sim.python.kit`, the gui poped up, but nothing could be rendered.
The errors that i got are

> 2024-12-06 15:59:16 [979ms] [Error] [carb.graphics-vulkan.plugin] Flags 0x6 must be the same for both device and instance.
> 
> 2024-12-06 15:59:16 [1,216ms] [Error] [carb.graphics-vulkan.plugin] VK_EXT_memory_budget is not supported on this platform. Can't get the VRAM memory usage info
> 
> 2024-12-06 15:59:16 [1,612ms] [Error] [gpu.foundation.plugin] Cannot load shader file 'rtx/system/GenerateMipMap.comp.hlsl'.
> 2024-12-06 15:59:16 [1,613ms] [Error] [gpu.foundation.plugin] buildShaderAsync() failed: rtx/system/GenerateMipMap.comp.hlsl
> 2024-12-06 15:59:16 [1,613ms] [Error] [gpu.foundation.plugin] Failed to async compile compute shader: rtx/system/GenerateMipMap.comp.hlsl
> 2024-12-06 15:59:17 [1,614ms] [Error] [gpu.foundation.plugin] Failed to compile compute shader: rtx/system/GenerateMipMap.comp.hlsl
> 
> 
> 2024-12-06 15:59:17 [2,217ms] [Error] [carb.graphics-vulkan.plugin] VK_EXT_memory_budget is not supported on this platform. Can't get the VRAM memory usage info
> 2024-12-06 15:59:17 [2,217ms] [Warning] [gpu.foundation.plugin] ResourceAllocator: Can't get memory usage info. All unused memory will be released
> 
> 2024-12-06 15:59:21 [6,475ms] [Error] [carb.scripting-python.plugin] UnicodeDecodeError: 'gbk' codec can't decode byte 0xa0 in position 1096: illegal multibyte sequence
> 
> At:
>   C:\Users\51388\anaconda3\envs\isaaclab\lib\json\__init__.py(293): load
>   c:/users/51388/anaconda3/envs/isaaclab/lib/site-packages/isaacsim/exts/omni.isaac.sensor/omni/isaac/sensor/scripts/menu.py(232): __init__
>   c:/users/51388/anaconda3/envs/isaaclab/lib/site-packages/isaacsim/exts/omni.isaac.sensor/omni/isaac/sensor/scripts/extension.py(37): on_startup
>   C:\Users\51388\anaconda3\envs\isaaclab\Lib\site-packages\omni\kernel\py\omni\ext\_impl\_internal.py(164): _startup_ext
>   C:\Users\51388\anaconda3\envs\isaaclab\Lib\site-packages\omni\kernel\py\omni\ext\_impl\_internal.py(224): startup
>   C:\Users\51388\anaconda3\envs\isaaclab\Lib\site-packages\omni\kernel\py\omni\ext\_impl\_internal.py(328): startup_extension
>   PythonExtension.cpp::startup()(2): <module>
>   C:\Users\51388\anaconda3\envs\isaaclab\lib\site-packages\omni\kit_app.py(114): startup
>   C:\Users\51388\anaconda3\envs\isaaclab\lib\site-packages\isaacsim\__init__.py(127): main
>   C:\Users\51388\anaconda3\envs\isaaclab\Scripts\isaacsim.exe\__main__.py(7): <module>
>   C:\Users\51388\anaconda3\envs\isaaclab\lib\runpy.py(86): _run_code
>   C:\Users\51388\anaconda3\envs\isaaclab\lib\runpy.py(196): _run_module_as_main
> 
> 2024-12-06 15:59:21 [6,476ms] [Error] [omni.ext.plugin] [ext: omni.isaac.sensor-12.9.1] Failed to startup python extension.
> 
>

RandomOakForest, 2024-12-06:
This may be a driver issue. Can you post `nvidia-smi` output? Thanks

anyRandomHuman, 2024-12-07:
Sat Dec  7 20:33:59 2024
+-----------------------------------------------------------------------------------------+
| NVIDIA-SMI 560.94                 Driver Version: 560.94         CUDA Version: 12.6     |
|-----------------------------------------+------------------------+----------------------+
| GPU  Name                  Driver-Model | Bus-Id          Disp.A | Volatile Uncorr. ECC |
| Fan  Temp   Perf          Pwr:Usage/Cap |           Memory-Usage | GPU-Util  Compute M. |
|                                         |                        |               MIG M. |
|=========================================+========================+======================|
|   0  NVIDIA GeForce RTX 4060 ...  WDDM  |   00000000:01:00.0 Off |                  N/A |
| N/A   47C    P8              1W /   55W |       0MiB /   8188MiB |      0%      Default |
|                                         |                        |                  N/A |
+-----------------------------------------+------------------------+----------------------+

+-----------------------------------------------------------------------------------------+
| Processes:                                                                              |
|  GPU   GI   CI        PID   Type   Process name                              GPU Memory |
|        ID   ID                                                               Usage      |
|=========================================================================================|
|    0   N/A  N/A     19796    C+G   ...inaries\Win64\EpicGamesLauncher.exe      N/A      |
+-----------------------------------------------------------------------------------------+

RandomOakForest, 2024-12-09:
Thanks. The documentation mentions that [on Windows with CUDA 12, the GPU driver version 552.86 is required](https://isaac-sim.github.io/IsaacLab/main/source/setup/installation/pip_installation.html). Please give this a try and see if the issue persists.

anyRandomHuman, 2024-12-09:
Well, i just checked and found that i could not simply roll back my driver.
And more importantly, i have an IsaacSim installed via Omniverse that works with my current driver.
So i think the driver really isn't the problem.
[image]

anyRandomHuman, 2024-12-24:
Does this not prove the compatiablity of my driver with isaac sim?
Should i try other measures?

RandomOakForest, 2025-01-02:
In such a case, could you try adding the following flag to when running Isaac Lab or Isaac Sim?
```
--/renderer/multiGpu/enabled=false
```
It may be a workaround if using the recommended drivers is not possible.

anyRandomHuman, 2025-01-03:
I have tried to run 
`isaacsim omni.isaac.sim.python.kit --/renderer/multiGpu/enabled=false` and it does not work.
And i have additionally tried to disable the gpu in device manager (once for both gpus) and doesn't seem to solve the problem.

And following errors are reported for running the command

_2025-01-03 08:42:17 [14,506ms] [Error] [carb.scripting-python.plugin] UnicodeDecodeError: 'gbk' codec can't decode byte 0xa0 in position 1096: illegal multibyte sequence

At:
  C:\Users\51388\anaconda3\envs\isaaclab\lib\json\__init__.py(293): load
  c:/users/51388/anaconda3/envs/isaaclab/lib/site-packages/isaacsim/exts/omni.isaac.sensor/omni/isaac/sensor/scripts/menu.py(232): __init__
  c:/users/51388/anaconda3/envs/isaaclab/lib/site-packages/isaacsim/exts/omni.isaac.sensor/omni/isaac/sensor/scripts/extension.py(37): on_startup
  C:\Users\51388\anaconda3\envs\isaaclab\Lib\site-packages\omni\kernel\py\omni\ext\_impl\_internal.py(164): _startup_ext
  C:\Users\51388\anaconda3\envs\isaaclab\Lib\site-packages\omni\kernel\py\omni\ext\_impl\_internal.py(224): startup
  C:\Users\51388\anaconda3\envs\isaaclab\Lib\site-packages\omni\kernel\py\omni\ext\_impl\_internal.py(328): startup_extension
  PythonExtension.cpp::startup()(2): <module>
  C:\Users\51388\anaconda3\envs\isaaclab\lib\site-packages\omni\kit_app.py(114): startup
  C:\Users\51388\anaconda3\envs\isaaclab\lib\site-packages\isaacsim\__init__.py(127): main
  C:\Users\51388\anaconda3\envs\isaaclab\Scripts\isaacsim.exe\__main__.py(7): <module>
  C:\Users\51388\anaconda3\envs\isaaclab\lib\runpy.py(86): _run_code
  C:\Users\51388\anaconda3\envs\isaaclab\lib\runpy.py(196): _run_module_as_main

2025-01-03 08:42:17 [14,506ms] [Error] [omni.ext.plugin] [ext: omni.isaac.sensor-12.9.1] Failed to startup python extension._

_2025-01-03 08:42:31 [29,015ms] [Error] [gpu.foundation.plugin] Raytracing is not supported on GPU: 0. Hardware, driver or Operating system upgrade is required.
2025-01-03 08:42:31 [29,015ms] [Error] [carb.scenerenderer-rtx.plugin] rtx driver verification failed._

rynem89, 2025-01-29:
Hey @anyRandomHuman , got any updates to overcome these errors? I am facing the same errors.

anyRandomHuman, 2025-01-29:
Unfortunately not.

> Hey [@anyRandomHuman](https://github.com/anyRandomHuman) , got any updates to overcome these errors? I am facing the same errors.

RandomOakForest, 2025-02-15:
Thanks for following up. This is likely an issue that would be resolved using the recommended driver. Please try the latest version of the tools and the driver recommended for your platform. If the issue persists, please open a _bug report_ so that we can reproduce the issue. Thank you.

closed 2025-02-15 by RandomOakForest