isaac-sim/IsaacLab #1511: Windows bug issue

vinomutty, 2024-12-07:
when giving commands in the command prompt
pip install isaacsim==4.2.0.2 --extra-index-url https://pypi.nvidia.com
facing errors, and warnings and could not find a version that required satisfaction. I am using Python 3.11.0
I am using an upgraded pip
but my pip list not showing Isaacs and Omni
module not found error
I have used the documentation and full Python commands on Windows
please help

Toni-SM, 2024-12-07:
From Isaac Lab docs

[image]

RandomOakForest, 2024-12-09:
Thanks, yes, the [documentation specifies Python 3.10 as a requirement](https://isaac-sim.github.io/IsaacLab/main/source/setup/installation/pip_installation.html).

closed 2025-02-15 by RandomOakForest