isaac-sim/IsaacLab #1512: Abnormal speed display when external force is applied

mjjhand123, 2024-12-07:
Hello esteemed developers
I was using isaaclab to apply an external force to an intelligent body to make it move along a pipe deployed in the environment when observing the intelligent body through root_pos_w and root_lin_vel_w, and I noticed that when moving backwards there is a velocity value towards the front, but from the page the robot is really moving backwards.
USE IsaacSim 4.2.0
`@configclass
class ObservationsCfg:
    """Observation specifications for the MDP."""

    @configclass
    class PolicyCfg(ObsGroup):
        """Observations for policy group."""

        joint_pos_rel = ObsTerm(func=mdp.joint_pos_rel)
        joint_vel_rel = ObsTerm(func=mdp.joint_vel_rel)
        body_vel_lin = ObsTerm(func=mdp.root_lin_vel_w, params = {"asset_cfg": SceneEntityCfg("robot")})
        body_position = ObsTerm(func=mdp.root_pos_w)
     
        actions = ObsTerm(func=mdp.last_action)

        def __post_init__(self):
            self.enable_corruption = True
            self.concatenate_terms = True

    # observation groups
    policy: PolicyCfg = PolicyCfg()
`

[image]

mjjhand123, 2024-12-07:
[image]
[image]
The model looks like this and then the GPU uses a 4070Ti

RandomOakForest, 2024-12-09:
Thank you for posting this. Could you elaborate on the relationship in the red boxes values you highlight? is this happening in the beginning frames or later ones?