isaac-sim/IsaacLab #1514: [Bug Report] ERROR: Failed to build installable wheels for some pyproject.toml based projects (egl_probe)

Oliverchhhh, 2024-12-08:
### [bug]

I followed the official guide to install Isaac Lab step by step, and encountered an error when running ./isaaclab.sh --install. The error is: ERROR: Failed to build installable wheels for some pyproject.toml based projects (egl_probe). I updated pip, installed the required packages, and even tried installing the egl_probe package separately, but none of these solutions worked.

### [step]
first
```
./isaaclab.sh --install
```

then,error
```
 Building wheel for egl_probe (setup.py) ... error
  error: subprocess-exited-with-error
  
  × python setup.py bdist_wheel did not run successfully.
  │ exit code: 1
  ╰─> [365 lines of output]
      /home/ch/miniconda3/envs/lab/lib/python3.10/site-packages/setuptools/_distutils/dist.py:261: UserWarning: Unknown distribution option: 'tests_require'
        warnings.warn(msg)
      running bdist_wheel
      running build
      running build_py
      creating build/lib.linux-x86_64-cpython-310/egl_probe
      copying egl_probe/get_available_devices.py -> build/lib.linux-x86_64-cpython-310/egl_probe
      copying egl_probe/__init__.py -> build/lib.linux-x86_64-cpython-310/egl_probe
      running egg_info
      writing egl_probe.egg-info/PKG-INFO
      writing dependency_links to egl_probe.egg-info/dependency_links.txt
      writing top-level names to egl_probe.egg-info/top_level.txt
      ERROR setuptools_scm._file_finders.git listing git files failed - pretending there aren't any
      reading manifest file 'egl_probe.egg-info/SOURCES.txt'
      reading manifest template 'MANIFEST.in'
      adding license file 'LICENSE'
      writing manifest file 'egl_probe.egg-info/SOURCES.txt'
      /home/ch/miniconda3/envs/lab/lib/python3.10/site-packages/setuptools/command/build_py.py:218: _Warning: Package 'egl_probe.glad' is absent from the `packages` configuration.
      !!
      
              ********************************************************************************
              ############################
              # Package would be ignored #
              ############################
              Python recognizes 'egl_probe.glad' as an importable package[^1],
              but it is absent from setuptools' `packages` configuration.
      
              This leads to an ambiguous overall configuration. If you want to distribute this
              package, please make sure that 'egl_probe.glad' is explicitly added
              to the `packages` configuration field.
      
              Alternatively, you can also rely on setuptools' discovery methods
              (for example by using `find_namespace_packages(...)`/`find_namespace:`
              instead of `find_packages(...)`/`find:`).
      
              You can read more about "package discovery" on setuptools documentation page:
      
              - https://setuptools.pypa.io/en/latest/userguide/package_discovery.html
      
              If you don't want 'egl_probe.glad' to be distributed and are
              already explicitly excluding 'egl_probe.glad' via
              `find_namespace_packages(...)/find_namespace` or `find_packages(...)/find`,
              you can try to use `exclude_package_data`, or `include-package-data=False` in
              combination with a more fine grained `package-data` configuration.
      
              You can read more about "package data files" on setuptools documentation page:
      
              - https://setuptools.pypa.io/en/latest/userguide/datafiles.html
      
      
              [^1]: For Python, any directory (with suitable naming) can be imported,
                    even if it does not contain any `.py` files.
[... 312 lines cut ...]
      subprocess.CalledProcessError: Command 'cmake ..; make -j' returned non-zero exit status 2.
      [end of output]
  
  note: This error originates from a subprocess, and is likely not a problem with pip.
  ERROR: Failed building wheel for egl_probe
  Running setup.py clean for egl_probe
Successfully built omni-isaac-lab_tasks robomimic rsl-rl
Failed to build egl_probe
ERROR: ERROR: Failed to build installable wheels for some pyproject.toml based projects (egl_probe)
```

### System Info

Describe the characteristic of your environment:

- Commit: [e.g. 8f3b9ca]
- Isaac Sim Version: 4.0.0-rc.21+4.0.13872.3e3cb0c9.gl(lab)
- OS:  Ubuntu 22.04
- GPU: RTX 3060 ti
- CUDA: 11.6
- GPU Driver:565.57.01

### Checklist

-  I have checked that there is no similar issue in the repo

Oliverchhhh, 2024-12-11:
I tried the 4.2.0 version of Isaac Sim, and the installation was very smooth. Thank you very much for your suggestion!

RandomOakForest, 2024-12-16:
Great to hear. I will close this issue for now, please reopen it if needed. Thanks.

closed 2024-12-16 by RandomOakForest

KyleM73, 2025-03-27:
I ran into this issue when setting up on a new machine, but resolved it with `sudo apt  install cmake` in case anyone runs into this issue again

XuSeeYoung, 2025-04-12:
I also ran into this issue when installing isaac Sim 4.5.0 and isaac lab. Even I have run `sudo apt isntall cmake`.
Finally, I solved this problem by reinstalling cmake 3.22 version. i.e., `pip install cmake==3.22`

Zzl3, 2025-07-18:
> Hi, I'm currently setting up on a new machine. The `pip install cmake==3.22` command (thank you [@XuSeeYoung](https://github.com/XuSeeYoung) for the fix!) allowed me to complete the Isaaclab installation. However, I still receive two error messages regarding incompatible versions of gymmnasium and stable baselines from running `./isaaclab.sh --install.` So far, the tutorials have run as expected. Should I be concerned about this issue if I don't plan to use SB3?您好，我目前正在一台新计算机上进行设置。`pip install cmake==3.22` 命令（感谢您的修复！）允许我完成 Isaaclab 安装。但是，我仍然收到两条错误消息，分别是不兼容的 gymmnasium 版本和运行 `./isaaclab.sh --install` 的稳定基线。到目前为止，教程已按预期运行。如果我不打算使用 SB3，我应该担心这个问题吗？
> 
> <img alt="Image" width="822" height="23.765625" src="https://private-user-images.githubusercontent.com/82644963/465427927-2b1f5a39-f75f-4886-850e-91d3e46023e9.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3NTI4MTA1ODksIm5iZiI6MTc1MjgxMDI4OSwicGF0aCI6Ii84MjY0NDk2My80NjU0Mjc5MjctMmIxZjVhMzktZjc1Zi00ODg2LTg1MGUtOTFkM2U0NjAyM2U5LnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNTA3MTglMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjUwNzE4VDAzNDQ0OVomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWViYjUyMjBhYzk0Y2ZmOWI1OGFiMzZmZWUwZThiMzA0NWFjY2Y5NWM5MGViMGQyOTVkYWJiYzJmNDNmOGU5M2QmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.sOFqU_GuhV-l1XYKM_LslRqvGfObhs7MeUzGZhmpusc"> <img alt="Image" width="822" height="24.296875" src="https://private-user-images.githubusercontent.com/82644963/465428378-b23810eb-cfc9-4b6b-93da-35897ccfa8ae.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3NTI4MTA1ODksIm5iZiI6MTc1MjgxMDI4OSwicGF0aCI6Ii84MjY0NDk2My80NjU0MjgzNzgtYjIzODEwZWItY2ZjOS00YjZiLTkzZGEtMzU4OTdjY2ZhOGFlLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNTA3MTglMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjUwNzE4VDAzNDQ0OVomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTdlMDkxNDM4ZDlhZDdhZGI1NmJhZTcyMzNiNWVlZWUzM2Y5MmE4Zjg5OTRlMDdhMzFhOWUzMDhlZDdmNTFkYWImWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.AwDpUc01nupotOuamNf8X84TzNANva6sULVna45aoFw">
> ```
> Collecting cmake==3.22
>   Downloading cmake-3.22.0-py2.py3-none-manylinux_2_12_x86_64.manylinux2010_x86_64.whl.metadata (7.3 kB)
> Downloading cmake-3.22.0-py2.py3-none-manylinux_2_12_x86_64.manylinux2010_x86_64.whl (21.8 MB)
>    ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 21.8/21.8 MB 9.6 MB/s eta 0:00:00
> Installing collected packages: cmake
> Successfully installed cmake-3.22.0
> (env_isaaclab) virgilio@virgilio-laptop:~/IsaacLab$ ./isaaclab.sh --install
> [INFO] Installing extensions inside the Isaac Lab repository...                                                                                                                                                                                                
> 	 module: /home/virgilio/IsaacLab/source/isaaclab_rl
> Obtaining file:///home/virgilio/IsaacLab/source/isaaclab_rl
>   Installing build dependencies ... done
>   Checking if build backend supports build_editable ... done
>   Getting requirements to build editable ... done
>   Preparing editable metadata (pyproject.toml) ... done
> Requirement already satisfied: numpy<2 in /home/virgilio/miniconda3/envs/env_isaaclab/lib/python3.10/site-packages (from isaaclab_rl==0.1.8) (1.26.4)
> Requirement already satisfied: torch==2.5.1 in /home/virgilio/miniconda3/envs/env_isaaclab/lib/python3.10/site-packages (from isaaclab_rl==0.1.8) (2.5.1+cu121)
> Requirement already satisfied: torchvision>=0.14.1 in /home/virgilio/miniconda3/envs/env_isaaclab/lib/python3.10/site-packages (from isaaclab_rl==0.1.8) (0.20.1+cu121)
> Requirement already satisfied: protobuf!=5.26.0,>=3.20.2 in /home/virgilio/miniconda3/envs/env_isaaclab/lib/python3.10/site-packages (from isaaclab_rl==0.1.8) (3.20.3)
> Requirement already satisfied: hydra-core in /home/virgilio/miniconda3/envs/env_isaaclab/lib/python3.10/site-packages (from isaaclab_rl==0.1.8) (1.3.2)
> Requirement already satisfied: h5py in /home/virgilio/miniconda3/envs/env_isaaclab/lib/python3.10/site-packages (from isaaclab_rl==0.1.8) (3.14.0)
> Requirement already satisfied: tensorboard in /home/virgilio/miniconda3/envs/env_isaaclab/lib/python3.10/site-packages (from isaaclab_rl==0.1.8) (2.19.0)
> Requirement already satisfied: moviepy in /home/virgilio/miniconda3/envs/env_isaaclab/lib/python3.10/site-packages (from isaaclab_rl==0.1.8) (2.2.1)
> Requirement already satisfied: pillow==11.0.0 in /home/virgilio/miniconda3/envs/env_isaaclab/lib/python3.10/site-packages (from isaaclab_rl==0.1.8) (11.0.0)
> Requirement already satisfied: filelock in /home/virgilio/miniconda3/envs/env_isaaclab/lib/python3.10/site-packages (from torch==2.5.1->isaaclab_rl==0.1.8) (3.13.1)
> Requirement already satisfied: typing-extensions>=4.8.0 in /home/virgilio/miniconda3/envs/env_isaaclab/lib/python3.10/site-packages (from torch==2.5.1->isaaclab_rl==0.1.8) (4.12.2)
> Requirement already satisfied: networkx in /home/virgilio/miniconda3/envs/env_isaaclab/lib/python3.10/site-packages (from torch==2.5.1->isaaclab_rl==0.1.8) (3.3)
> Requirement already satisfied: jinja2 in /home/virgilio/miniconda3/envs/env_isaaclab/lib/python3.10/site-packages (from torch==2.5.1->isaaclab_rl==0.1.8) (3.1.4)
[... cut ...]
> [WARN] Could not find Isaac Sim VSCode settings: /home/virgilio/miniconda3/envs/env_isaaclab/lib/python3.10/site-packages/isaacsim/.vscode/settings.json.
> 	This will result in missing 'python.analysis.extraPaths' in the VSCode
> 	settings, which limits the functionality of the Python language server.
> 	However, it does not affect the functionality of the Isaac Lab project.
> 	We are working on a fix for this issue with the Isaac Sim team.
> ```

Hi, I had the same problem as you, have you solved it yet?