isaac-sim/IsaacLab #1519: [Bug Report] low quality of rendered image with gpu like A100? May because DLSS-RR is not supported???

yichaoshen-MS, 2024-12-09:
If you are submitting a bug report, please fill in the following details and use the tag [bug].

### Describe the bug

Low quality of rendered images with A100, and there is a warning of 

' [Warning] [rtx.postprocessing.plugin] DLSS-RR is not supported in the current build due to some known bugs with this GPU: NVIDIA A100 PCIe.Note that RTX Real-Time raytracing mode will not be able to denoise the output correctly'

[image]

### System Info

Describe the characteristic of your environment:

- Commit: [e.g. 8f3b9ca]
- Isaac Sim Version: [e.g. 2022.2.0, this can be obtained by `cat ${ISAACSIM_PATH}/VERSION`]
- OS: [e.g. Ubuntu 20.04]
- GPU: [e.g. A100 PCIe]
- CUDA: [e.g. 11.4]
- GPU Driver: [e.g. 470.82.01, this can be seen by using `nvidia-smi` command.]

RandomOakForest, 2024-12-10:
This may be due to the version of Isaac Sim you are using. Please try version Isaac Sim 4.2.

yichaoshen-MS, 2024-12-10:
> This may be due to the version of Isaac Sim you are using. Please try version Isaac Sim 4.2.
Thanks for your response, however, I have used the latest Issacsim 4.2.
[image]

Did Isaaclab’ developers test to use A100s to render?

KyleM73, 2024-12-10:
I would also look at updating the driver version if possible, I find more stable results with 550. I also run into trouble with A100 rendering due to lack of RTX cores.

yichaoshen-MS, 2024-12-10:
> I would also look at updating the driver version if possible, I find more stable results with 550. I also run into trouble with A100 rendering due to lack of RTX cores.

Thank you for your suggestion. However, my driver version is already 550.127.05, and Vulkan is also installed. It seems that the lack of RTX cores might be the core cause of the issue? Interestingly, this problem doesn't occur in Isaac Gym, for instance, when running Isaac Gym with the A100.

celestialdr4g0n, 2024-12-11:
Please use headless mode to gain more training speed (much better). For visualizing purpose, please use play.py script and set a low number of environment (e.g: 120).

closed 2024-12-13 by yichaoshen-MS

KyleM73, 2024-12-13:
@yichaoshen-MS how did you resolve the problem?