isaac-sim/IsaacLab #1522: gymnasium.error.NameNotFound: Environment `Isaac-H1-Direct` doesn't exist.

panhainan2020, 2024-12-10:
Isaac lab uses docker to install the configuration. I modified the existing Direct RL environment according to the tutorial, and the following error occurred. Has anyone encountered similar problems?
```
isaaclab -p source/standalone/workflows/rl_games/train.py --task Isaac-H1-Direct-v0 --headless
[INFO] Using python from: /workspace/isaaclab/_isaac_sim/python.sh                                     
[INFO][AppLauncher]: Loading experience file: /workspace/isaaclab/source/apps/isaaclab.python.headless.kit
Loading user config located at: '/isaac-sim/kit/data/Kit/Isaac-Sim/4.2/user.config.json'
[Info] [carb] Logging to file: /isaac-sim/kit/logs/Kit/Isaac-Sim/4.2/kit_20241210_174805.log
2024-12-10 17:48:05 [0ms] [Warning] [omni.kit.app.plugin] No crash reporter present, dumps uploading isn't available.

|---------------------------------------------------------------------------------------------|
| Driver Version: 560.35.03     | Graphics API: Vulkan
|=============================================================================================|
| GPU | Name                             | Active | LDA | GPU Memory | Vendor-ID | LUID       |
|     |                                  |        |     |            | Device-ID | UUID       |
|     |                                  |        |     |            | Bus-ID    |            |
|---------------------------------------------------------------------------------------------|
| 0   | NVIDIA GeForce RTX 4090 D        | Yes: 0 |     | 24810   MB | 10de      | 0          |
|     |                                  |        |     |            | 2685      | 91299c12.. |
|     |                                  |        |     |            | 1         |            |
|=============================================================================================|
| OS: 22.04.3 LTS (Jammy Jellyfish) ubuntu, Version: 22.04.3, Kernel: 5.15.0-67-generic
| XServer Vendor: The X.Org Foundation, XServer Version: 12013000 (1.20.13.0)
| Processor: 11th Gen Intel(R) Core(TM) i7-11700F @ 2.50GHz | Cores: 8 | Logical: 16
|---------------------------------------------------------------------------------------------|
| Total Memory (MB): 64031 | Free Memory: 56984
| Total Page/Swap (MB): 15624 | Free Page/Swap: 15624
|---------------------------------------------------------------------------------------------|
Traceback (most recent call last):
  File "/workspace/isaac_lab/source/standalone/workflows/rl_games/train.py", line 178, in <module>
    main()
  File "/workspace/isaaclab/source/extensions/omni.isaac.lab_tasks/omni/isaac/lab_tasks/utils/hydra.py", line 79, in wrapper
    env_cfg, agent_cfg = register_task_to_hydra(task_name, agent_cfg_entry_point)
  File "/workspace/isaaclab/source/extensions/omni.isaac.lab_tasks/omni/isaac/lab_tasks/utils/hydra.py", line 42, in register_task_to_hydra
    env_cfg = load_cfg_from_registry(task_name, "env_cfg_entry_point")
  File "/workspace/isaaclab/source/extensions/omni.isaac.lab_tasks/omni/isaac/lab_tasks/utils/parse_cfg.py", line 55, in load_cfg_from_registry
    cfg_entry_point = gym.spec(task_name).kwargs.get(entry_point_key)
  File "/isaac-sim/exts/omni.isaac.ml_archive/pip_prebundle/gymnasium/envs/registration.py", line 1029, in spec
    _check_version_exists(ns, name, version)
  File "/isaac-sim/exts/omni.isaac.ml_archive/pip_prebundle/gymnasium/envs/registration.py", line 393, in _check_version_exists
    _check_name_exists(ns, name)
  File "/isaac-sim/exts/omni.isaac.ml_archive/pip_prebundle/gymnasium/envs/registration.py", line 370, in _check_name_exists
    raise error.NameNotFound(
gymnasium.error.NameNotFound: Environment `Isaac-H1-Direct` doesn't exist. Did you mean: `Isaac-Ant-Direct`?
2024-12-10 17:48:10 [4,838ms] [Warning] [carb] Recursive unloadAllPlugins() detected!
```

_Originally posted by @panhainan2020 in https://github.com/isaac-sim/IsaacLab/issues/1502#issuecomment-2532466488_

panhainan2020, 2024-12-11:
Incorrect mounting path of docker container, which has been solved.

panhainan2020, 2024-12-11:
> ```
>           Isaac lab uses docker to install the configuration. I modified the existing Direct RL environment according to the tutorial, and the following error occurred. Has anyone encountered similar problems?
> ```
> 
> ```
> isaaclab -p source/standalone/workflows/rl_games/train.py --task Isaac-H1-Direct-v0 --headless
> [INFO] Using python from: /workspace/isaaclab/_isaac_sim/python.sh                                     
> [INFO][AppLauncher]: Loading experience file: /workspace/isaaclab/source/apps/isaaclab.python.headless.kit
> Loading user config located at: '/isaac-sim/kit/data/Kit/Isaac-Sim/4.2/user.config.json'
> [Info] [carb] Logging to file: /isaac-sim/kit/logs/Kit/Isaac-Sim/4.2/kit_20241210_174805.log
> 2024-12-10 17:48:05 [0ms] [Warning] [omni.kit.app.plugin] No crash reporter present, dumps uploading isn't available.
> 
> |---------------------------------------------------------------------------------------------|
> | Driver Version: 560.35.03     | Graphics API: Vulkan
> |=============================================================================================|
> | GPU | Name                             | Active | LDA | GPU Memory | Vendor-ID | LUID       |
> |     |                                  |        |     |            | Device-ID | UUID       |
> |     |                                  |        |     |            | Bus-ID    |            |
> |---------------------------------------------------------------------------------------------|
> | 0   | NVIDIA GeForce RTX 4090 D        | Yes: 0 |     | 24810   MB | 10de      | 0          |
> |     |                                  |        |     |            | 2685      | 91299c12.. |
> |     |                                  |        |     |            | 1         |            |
> |=============================================================================================|
> | OS: 22.04.3 LTS (Jammy Jellyfish) ubuntu, Version: 22.04.3, Kernel: 5.15.0-67-generic
> | XServer Vendor: The X.Org Foundation, XServer Version: 12013000 (1.20.13.0)
> | Processor: 11th Gen Intel(R) Core(TM) i7-11700F @ 2.50GHz | Cores: 8 | Logical: 16
> |---------------------------------------------------------------------------------------------|
> | Total Memory (MB): 64031 | Free Memory: 56984
> | Total Page/Swap (MB): 15624 | Free Page/Swap: 15624
> |---------------------------------------------------------------------------------------------|
> Traceback (most recent call last):
>   File "/workspace/isaac_lab/source/standalone/workflows/rl_games/train.py", line 178, in <module>
>     main()
>   File "/workspace/isaaclab/source/extensions/omni.isaac.lab_tasks/omni/isaac/lab_tasks/utils/hydra.py", line 79, in wrapper
>     env_cfg, agent_cfg = register_task_to_hydra(task_name, agent_cfg_entry_point)
>   File "/workspace/isaaclab/source/extensions/omni.isaac.lab_tasks/omni/isaac/lab_tasks/utils/hydra.py", line 42, in register_task_to_hydra
>     env_cfg = load_cfg_from_registry(task_name, "env_cfg_entry_point")
>   File "/workspace/isaaclab/source/extensions/omni.isaac.lab_tasks/omni/isaac/lab_tasks/utils/parse_cfg.py", line 55, in load_cfg_from_registry
>     cfg_entry_point = gym.spec(task_name).kwargs.get(entry_point_key)
>   File "/isaac-sim/exts/omni.isaac.ml_archive/pip_prebundle/gymnasium/envs/registration.py", line 1029, in spec
>     _check_version_exists(ns, name, version)
>   File "/isaac-sim/exts/omni.isaac.ml_archive/pip_prebundle/gymnasium/envs/registration.py", line 393, in _check_version_exists
>     _check_name_exists(ns, name)
>   File "/isaac-sim/exts/omni.isaac.ml_archive/pip_prebundle/gymnasium/envs/registration.py", line 370, in _check_name_exists
>     raise error.NameNotFound(
> gymnasium.error.NameNotFound: Environment `Isaac-H1-Direct` doesn't exist. Did you mean: `Isaac-Ant-Direct`?
> 2024-12-10 17:48:10 [4,838ms] [Warning] [carb] Recursive unloadAllPlugins() detected!
> ```
> 
> _Originally posted by @panhainan2020 in [#1502 (comment)](https://github.com/isaac-sim/IsaacLab/issues/1502#issuecomment-2532466488)_

Incorrect mounting path of docker container, which has been solved.

RandomOakForest, 2024-12-11:
Thanks for posting this. Can you close this if it is no longer an issue? Thanks!

closed 2024-12-11 by panhainan2020