isaac-sim/IsaacLab #1527: [Bug Report] Unable to render

Huisouan, 2024-12-12:
If you are submitting a bug report, please fill in the following details and use the tag [bug].

### Describe the bug
[image]
Unable to render the scene with the command 
./isaaclab.sh -p source/standalone/workflows/rsl_rl/train.py --task Isaac-Reach-Franka-v0

### System Info

Describe the characteristic of your environment:

- Commit: [e.g. [762f4e3]]
- Isaac Sim Version: [4.2.0.2]
- OS: [Ubuntu 22.04]
- GPU: [e.g. RTX 4090
- CUDA: [e.g. 12.4]
- GPU Driver: [e.g. 550.120]

Mayankm96, 2024-12-12:
You need to also pass the command `--enable_cameras`.

Huisouan, 2024-12-12:
tried，still the same problem。

Huisouan, 2024-12-12:
ps :I was able to train with --headless mode.

RandomOakForest, 2024-12-12:
Thanks for posting this. Could you please post the full output of the error message you get? It seems the stdout you posted is a partial one. There should be a line with a type of error such as `AssertionError`, etc. for example. Also, if you could try the release [commit for Isaac Lab 1.3](https://github.com/isaac-sim/IsaacLab/releases/tag/v1.3.0) would be appreciated. Thanks!

Huisouan, 2024-12-14:
it seems like there is also an asset problem.
I uesd another computer,and reporformed this problem
`
(isaaclab) (base) hsh@hsh-MS-7E01:~/Code/IsaacLab$ ./isaaclab.sh -p source/standalone/workflows/rsl_rl/train.py --task Isaac-Reach-Franka-v0
[INFO] Using python from: /home/hsh/miniconda3/envs/isaaclab/bin/python                                                                                          
[INFO][AppLauncher]: Loading experience file: /home/hsh/Code/isaac-rl-lab/source/apps/isaaclab.python.kit
[Warning] [omni.isaac.kit.simulation_app] Modules: ['omni.kit_app'] were loaded before SimulationApp was started and might not be loaded correctly.
[Warning] [omni.isaac.kit.simulation_app] Please check to make sure no extra omniverse or pxr modules are imported before the call to SimulationApp(...)
Loading user config located at: '/home/hsh/miniconda3/envs/isaaclab/lib/python3.10/site-packages/omni/data/Kit/Isaac-Sim/4.2/user.config.json'
[Info] [carb] Logging to file: /home/hsh/miniconda3/envs/isaaclab/lib/python3.10/site-packages/omni/logs/Kit/Isaac-Sim/4.2/kit_20241214_142554.log
2024-12-14 06:25:54 [0ms] [Warning] [omni.kit.app.plugin] No crash reporter present, dumps uploading isn't available.
2024-12-14 06:25:54 [4ms] [Warning] [omni.ext.plugin] [ext: unitree_s2r] Extensions config 'extension.toml' doesn't exist '/home/hsh/Code/isaac-rl-lab/source/extensions/unitree_s2r' or '/home/hsh/Code/isaac-rl-lab/source/extensions/unitree_s2r/config'
2024-12-14 06:25:54 [4ms] [Warning] [omni.ext.plugin] [ext: unitree_sdk2_python] Extensions config 'extension.toml' doesn't exist '/home/hsh/Code/isaac-rl-lab/source/extensions/unitree_sdk2_python' or '/home/hsh/Code/isaac-rl-lab/source/extensions/unitree_sdk2_python/config'

|---------------------------------------------------------------------------------------------|
| Driver Version: 550.120       | Graphics API: Vulkan
|=============================================================================================|
| GPU | Name                             | Active | LDA | GPU Memory | Vendor-ID | LUID       |
|     |                                  |        |     |            | Device-ID | UUID       |
|     |                                  |        |     |            | Bus-ID    |            |
|---------------------------------------------------------------------------------------------|
| 0   | NVIDIA GeForce RTX 4070 Ti SUPER | Yes: 0 |     | 16376   MB | 10de      | 0          |
|     |                                  |        |     |            | 2705      | ec0ca200.. |
|     |                                  |        |     |            | 1         |            |
|=============================================================================================|
| OS: 24.04.1 LTS (Noble Numbat) ubuntu, Version: 24.04.1, Kernel: 6.8.0-50-generic
| XServer Vendor: The X.Org Foundation, XServer Version: 12101011 (1.21.1.11)
| Processor: 13th Gen Intel(R) Core(TM) i7-13700KF | Cores: 16 | Logical: 24
|---------------------------------------------------------------------------------------------|
| Total Memory (MB): 64148 | Free Memory: 56052
| Total Page/Swap (MB): 8191 | Free Page/Swap: 8191
|---------------------------------------------------------------------------------------------|
2024-12-14 06:25:58 [3,715ms] [Warning] [omni.replicator.core.scripts.annotators] Annotator PostProcessDispatch is already registered, overwriting annotator template
2024-12-14 06:25:58 [4,012ms] [Warning] [omni.kit.widget.cache_indicator.cache_state_menu] Unable to detect Omniverse Cache Server. File /home/hsh/.nvidia-omniverse/config/omniverse.toml is not found. Consider installing it for better IO performance.
Path http://omniverse-content-production.s3-us-west-2.amazonaws.com/Assets/Isaac/4.2 exists and is set as NUCLEUS_ASSET_ROOT_DIR.
[INFO]: Parsing configuration from: <class 'omni.isaac.lab_tasks.manager_based.manipulation.reach.config.franka.joint_pos_env_cfg.FrankaReachEnvCfg'>
[INFO]: Parsing configuration from: omni.isaac.lab_tasks.manager_based.manipulation.reach.config.franka.agents.rsl_rl_ppo_cfg:FrankaReachPPORunnerCfg
[INFO] Logging experiment in directory: /home/hsh/Code/IsaacLab/logs/rsl_rl/franka_reach
Setting seed: 42
[2024-12-14 14:26:03,522][ogn_registration][INFO] - Looking for Python nodes to register in omni.physx.fabric-106.1.9
[2024-12-14 14:26:03,522][ogn_registration][INFO] -  -> Registered nodes from module omni.physxfabric at /home/hsh/miniconda3/envs/isaaclab/lib/python3.10/site-packages/isaacsim/extsPhysics/omni.physx.fabric
[2024-12-14 14:26:03,522][ogn_registration][INFO] - Registering nodes in /home/hsh/miniconda3/envs/isaaclab/lib/python3.10/site-packages/isaacsim/extsPhysics/omni.physx.fabric imported as omni.physxfabric with AutoNode config {}
[2024-12-14 14:26:03,522][ogn_registration][INFO] - Registering Python Node Types from omni.physxfabric at /home/hsh/miniconda3/envs/isaaclab/lib/python3.10/site-packages/isaacsim/extsPhysics/omni.physx.fabric in omni.physx.fabric
[2024-12-14 14:26:03,522][ogn_registration][INFO] - ========================================================================================================================
[2024-12-14 14:26:03,522][ogn_registration][INFO] - No dependency on omni.graph, therefore no nodes to register in omni.physx.fabric
[2024-12-14 14:26:03,522][ogn_registration][INFO] - ...None found, no registration to do
[... cut ...]
2024-12-14 06:29:07 [192,420ms] [Warning] [omni.graph.core.plugin] Could not find category 'Replicator:Annotators' for removal
2024-12-14 06:29:07 [192,420ms] [Warning] [omni.graph.core.plugin] Could not find category 'Replicator:Annotators' for removal
2024-12-14 06:29:07 [192,420ms] [Warning] [omni.graph.core.plugin] Could not find category 'Replicator:Annotators' for removal
2024-12-14 06:29:07 [192,420ms] [Warning] [omni.graph.core.plugin] Could not find category 'Replicator:Core' for removal
2024-12-14 06:29:07 [192,470ms] [Warning] [carb] Recursive unloadAllPlugins() detected!
`

RandomOakForest, 2025-02-15:
Following up on this, we ran into the holidays and release work, our apologies for the late reply. If this is something you still see in Isaac Sim 4.5 and Isaac Lab 2.0 with the recommended drivers, please let us know and we will investigate. Thank you.

RandomOakForest, 2025-02-19:
I hope this issue has been resolved with Isaac Sim 4.5 and Isaac Lab 2.0. If this is not the case, please reopen the issue and update your output log. Thanks.