isaac-sim/IsaacLab #1531: [Question] How to get cartesian positions of joints in world frame

nico-bohlinger, 2024-12-12:
Hello!

I was wondering how to get the cartesian position of a joint in the world frame.
So I'm not referring to the joint position in its possible range of motion but to its actual position in the world, similar to body_pos_w.

I would greatly appreciate any help on that.

Kind regards
Nico

RandomOakForest, 2025-01-02:
Thanks for posting this. It seems you have to access [ArticulationData](https://isaac-sim.github.io/IsaacLab/main/source/api/lab/omni.isaac.lab.assets.html#omni.isaac.lab.assets.ArticulationData), which _data is stored in the simulation world frame unless otherwise specified_. [This tutorial may be of help](https://isaac-sim.github.io/IsaacLab/main/source/tutorials/01_assets/run_articulation.html). Let us know if you need further help.

nico-bohlinger, 2025-01-02:
Unfortunately, the ArticulationData doesn't store the pose of the joint, it only holds the "joint_pos" which is the position of the joint in its range of motion in radians

StrainFlow, 2025-01-15:
I'm looking through the api for the exact calls that might do this, but it seems like you could do this by combining the link information in the ArticulationData class with the joint definition in USD. It seems like you should be able to get the link position, and then offset that by the joint position relative to that link.

StrainFlow, 2025-01-23:
You can do this directly with the `body_link_state_w` function. First you will need to get the names the the bodies in your articulation class using `Articulation.find_bodies` function. Then, if you want the position of a joint between a given parent and child body, you will pass in the name of the child body to get the position of the joint between it and its parent.

dantes-ai, 2025-02-03:
I'd like to know this as well as it could simplify imitation learning from mocap data.

> you will pass in the name of the child body to get the position of the joint between it and its parent.

Could you please specify to what function exactly we pass in the child body? body_link_state_w doesnt accept any arguments...

StrainFlow, 2025-02-14:
lol, so sorry, it's not a function but it does return that information for each link. I believe it is a dictionary so you can get a specific link's data by using its name as a key.

RandomOakForest, 2025-02-19:
Thanks for following up. Let us know if you need further help. I will close this issue for now.

closed 2025-02-19 by RandomOakForest