isaac-sim/IsaacLab #1534: [Proposal] Viewer tracking for non-root bodies during RL training

KyleM73, 2024-12-12:
### Proposal

In commit #59e6c68 the viewer config was added such that the default viewer could track the world origin, the environment origin, or the asset_root during training. I would like to be able to specify the origin of a different body in an articulation or rigid body to track as the origin, for example the end effector in a manipulation task. I am happy to work on this as a PR if there is no concurrent work being done to this effect.

### Motivation

The currently documented setup for wheeled navigation tasks suggests using a virtual arm to control the SO2 position of the robot instead of simulating the wheel dynamics directly, since the GPU pipeline currently struggles with smooth cylinders. This is also an easy way to implement quadcopter dynamics, which would suffer from the same problem (the robot is not the root). There is currently no way to track the robot (the way quadrupeds are tracked with asset_root) since the vehicle is not the asset_root in this setup. Additionally, it would be helpful to directly track the end effector in manipulation tasks to observe the grasps in high detail, no matter where the object is located prior to the grasp.

### Alternatives

Cameras can be configured to save videos e.g. in a play script that can take any origin. But this is only useful for analyzing policies after the fact, not during training.

### Checklist

- [X] I have checked that there is no similar issue in the repo

### Acceptance Criteria

- [ ] Support "asset_body" as an origin mode in the viewer cfg with an optional parameter "body_name" to specify which body to place the origin at for the camera.

RandomOakForest, 2024-12-17:
Thank you for posting this. I will tag this for the team to follow up upon.

RandomOakForest, 2024-12-19:
After reviewing this with the team, we will welcome your PR on this feature. Thank you for your work and for helping us improve Isaac Lab.

KyleM73, 2025-01-02:
Thanks for the feedback @RandomOakForest ! I have submitted a PR resolving this issue, I will close this issue once the PR is merged.

closed 2025-01-05 by KyleM73