isaac-sim/IsaacLab #1536: [Question] ValueError: No contact sensors added to the prim: '/World/envs/env_0/Robot'. This means that no rigid bodies are present under this prim. Please check the prim path.

jh-s5974, 2024-12-13:
```
./isaaclab.sh -p source/standalone/workflows/rsl_rl/train.py --task Isaac-Velocity-Rough-robotnl-v0
```
i use this commend.

```
2024-12-13 08:17:52 [12,356ms] [Warning] [omni.usd] Warning: in _ReportErrors at line 2890 of /builds/omniverse/usd-ci/USD/pxr/usd/usd/stage.cpp -- In </World/envs/env_0/Robot>: Unresolved reference prim path @/home/jh/IsaacLab/source/extensions/omni.isaac.lab_assets/data/Robots/robot_nl/robot_nl.usd@<defaultPrim> introduced by @anon:0x378b0450:World0.usd@</World/envs/env_0/Robot> (recomposing stage on stage @anon:0x378b0450:World0.usd@ <0x35df3b60>)

2024-12-13 08:17:52 [12,393ms] [Warning] [omni.isaac.lab.sim.utils] Could not perform 'modify_rigid_body_properties' on any prims under: '/World/envs/env_0/Robot'. This might be because of the following reasons:
	(1) The desired attribute does not exist on any of the prims.
	(2) The desired attribute exists on an instanced prim.
		Discovered list of instanced prim paths: []
2024-12-13 08:17:52 [12,393ms] [Warning] [omni.isaac.lab.sim.utils] Could not perform 'modify_articulation_root_properties' on any prims under: '/World/envs/env_0/Robot'. This might be because of the following reasons:
	(1) The desired attribute does not exist on any of the prims.
	(2) The desired attribute exists on an instanced prim.
		Discovered list of instanced prim paths: []
[INFO]: Time taken for scene creation : 0.040183 seconds
Error executing job with overrides: []
Traceback (most recent call last):
  File "/home/jh/IsaacLab/source/extensions/omni.isaac.lab_tasks/omni/isaac/lab_tasks/utils/hydra.py", line 99, in hydra_main
    func(env_cfg, agent_cfg, *args, **kwargs)
  File "/home/jh/IsaacLab/source/standalone/workflows/rsl_rl/train.py", line 101, in main
    env = gym.make(args_cli.task, cfg=env_cfg, render_mode="rgb_array" if args_cli.video else None)
  File "/home/jh/.local/share/ov/pkg/isaac-sim-4.2.0/exts/omni.isaac.ml_archive/pip_prebundle/gymnasium/envs/registration.py", line 802, in make
    env = env_creator(**env_spec_kwargs)
  File "/home/jh/IsaacLab/source/extensions/omni.isaac.lab/omni/isaac/lab/envs/manager_based_rl_env.py", line 75, in __init__
    super().__init__(cfg=cfg)
  File "/home/jh/IsaacLab/source/extensions/omni.isaac.lab/omni/isaac/lab/envs/manager_based_env.py", line 121, in __init__
    self.scene = InteractiveScene(self.cfg.scene)
  File "/home/jh/IsaacLab/source/extensions/omni.isaac.lab/omni/isaac/lab/scene/interactive_scene.py", line 150, in __init__
    self._add_entities_from_cfg()
  File "/home/jh/IsaacLab/source/extensions/omni.isaac.lab/omni/isaac/lab/scene/interactive_scene.py", line 584, in _add_entities_from_cfg
    self._articulations[asset_name] = asset_cfg.class_type(asset_cfg)
  File "/home/jh/IsaacLab/source/extensions/omni.isaac.lab/omni/isaac/lab/assets/articulation/articulation.py", line 99, in __init__
    super().__init__(cfg)
  File "/home/jh/IsaacLab/source/extensions/omni.isaac.lab/omni/isaac/lab/assets/asset_base.py", line 77, in __init__
    self.cfg.spawn.func(
  File "/home/jh/IsaacLab/source/extensions/omni.isaac.lab/omni/isaac/lab/sim/utils.py", line 276, in wrapper
    schemas.activate_contact_sensors(prim_paths[0], cfg.activate_contact_sensors)
  File "/home/jh/IsaacLab/source/extensions/omni.isaac.lab/omni/isaac/lab/sim/schemas/schemas.py", line 517, in activate_contact_sensors
    raise ValueError(
ValueError: No contact sensors added to the prim: '/World/envs/env_0/Robot'. This means that no rigid bodies are present under this prim. Please check the prim path.

Set the environment variable HYDRA_FULL_ERROR=1 for a complete stack trace.
```

but, i got this error.
and scene is shutdown.

how can i solve this error.

my code is this.
i copied and used h1 code in isaaclab.

### robotnl.py
```
from __future__ import annotations

import omni.isaac.lab.sim as sim_utils
from omni.isaac.lab.actuators import ImplicitActuatorCfg
from omni.isaac.lab.assets.articulation import ArticulationCfg

robotnl_CFG = ArticulationCfg(
    spawn=sim_utils.UsdFileCfg(
        usd_path="/home/jh/IsaacLab/source/extensions/omni.isaac.lab_assets/data/Robots/robot_nl/robot_nl.usd",
        activate_contact_sensors=True,
        rigid_props=sim_utils.RigidBodyPropertiesCfg(
            disable_gravity=False,
            retain_accelerations=False,
            linear_damping=0.0,
[... 308 lines cut ...]
        self.scene.terrain.terrain_generator = None
        # no height scan
        self.scene.height_scanner = None
        self.observations.policy.height_scan = None
        # no terrain curriculum
        self.curriculum.terrain_levels = None
        self.rewards.feet_air_time.weight = 1.0
        self.rewards.feet_air_time.params["threshold"] = 0.6

class robotnlFlatEnvCfg_PLAY(robotnlFlatEnvCfg):
    def __post_init__(self) -> None:
        # post init of parent
        super().__post_init__()

        # make a smaller scene for play
        self.scene.num_envs = 50
        self.scene.env_spacing = 2.5
        # disable randomization for play
        self.observations.policy.enable_corruption = False
        # remove random pushing
        self.events.base_external_force_torque = None
        self.events.push_robot = None
```

Thank you.

jh-s5974, 2024-12-13:
my robot link and joint is this.

[image]

ozhanozen, 2024-12-13:
Is the "robot_nl.usd" file your own asset? It seems there might be an issue with it. The initial warnings, such as `Could not perform 'modify_rigid_body_properties'/'modify_articulation_root_properties'`, suggest that some settings are not being correctly applied.

Could you verify the following for the "robot_nl.usd" file in Isaac Sim?

1. Ensure that all Xforms with the `_link` postfix have a rigid body attached and are not instanceable.
2. Confirm that the `robotnl` parent Xform has an articulation root attached, and that no other prims have an articulation root.
3. Choose `robotnl` default prim to make sure it is the one imported.

jh-s5974, 2024-12-13:
Everything is done.
But it doesn't work.

it sets are so hardly

jh-s5974, 2024-12-16:
@ozhanozen 

I got another error but I fixed this one.

It was because of the default prim setting.

Thank you.

I will try to solve the other error.

closed 2024-12-16 by jh-s5974

RandomOakForest, 2024-12-17:
Thank you for posting this. I will tag this to improve upon our documentation. Please feel free to open a new question or bug report on our _Issues_ if you can't resolve the other error you got.

SnoopyGeeek, 2025-01-04:
> @ozhanozen
> 
> I got another error but I fixed this one.
> 
> It was because of the default prim setting.
> 
> Thank you.
> 
> I will try to solve the other error.

I met the same error, could you plz tell me how dd you solve it? 
[image]

Catarina-lab, 2025-05-19:
> [@ozhanozen](https://github.com/ozhanozen)
> 
> I got another error but I fixed this one.
> 
> It was because of the default prim setting.
> 
> Thank you.
> 
> I will try to solve the other error.
Hello！How do you change the settings? Could you explain it in more detail?

LuXiangyu666, 2025-06-09:
> > > [@ozhanozen](https://github.com/ozhanozen)
> > > I got another error but I fixed this one.
> > > It was because of the default prim setting.
> > > Thank you.
> > > I will try to solve the other error.
> > > Hello！How do you change the settings? Could you explain it in more detail?
> 
> Sorry for the late reply.
> 
> I got an error because I didn't set the default prim of the robot usd file.

May I ask how to modify the default prim and what to change it to

ozhanozen, 2025-06-09:
> May I ask how to modify the default prim and what to change it to

You can right click on a new prim, and choose "set as default prim". 

When you import a usd, only the default prim and its subtree is imported, so you should choose it accordingly.