isaac-sim/IsaacLab #1539: [Bug Report] Malloc Error

SanskarSoni8, 2024-12-13:
Description

When I train with more than 512 environments, I get a malloc error that says: `malloc(): invalid size (unsorted)`. There is more than enough VRAM not being used in the 3080 GPU I have. Here is the output of nvidia-smi:

### Steps to reproduce

Please try to provide a minimal example to reproduce the bug. Error messages and stack traces are also helpful.

### System Info

- Isaac Sim Version: 4.2.0
- OS:Ubuntu 22.04
- GPU: NVIDIA GeForce RTX 3080
- CUDA:  11.2
- GPU Driver: 550.120

### Checklist

- [YES] I have checked that there is no similar issue in the repo (**required**)
- [YES] I have checked that the issue is not in running Isaac Sim itself and is related to the repo

### Acceptance Criteria

- [ ] Criteria 1
- [ ] Criteria 2

KyleM73, 2024-12-15:
I would look at these errors/warnings first:

`2024-12-13 20:20:02 [3,701ms] [Warning] [omni.isaac.lab.envs.manager_based_env] The render interval (4) is smaller than the decimation (12). Multiple render calls will happen for each environment step. If this is not intended, set the render interval to be equal to the decimation.`

`2024-12-13 20:20:02 [3,936ms] [Warning] [omni.isaac.lab.sim.utils] Could not perform 'modify_fixed_tendon_properties' on any prims under: '/World/envs/env_0/robot'. This might be because of the following reasons:
	(1) The desired attribute does not exist on any of the prims.
	(2) The desired attribute exists on an instanced prim.
		Discovered list of instanced prim paths: []`

`2024-12-13 20:20:02 [3,942ms] [Warning] [omni.isaac.lab.sim.utils] Could not perform 'modify_mass_properties' on any prims under: '/World/envs/env_0/sphere'. This might be because of the following reasons:
	(1) The desired attribute does not exist on any of the prims.
	(2) The desired attribute exists on an instanced prim.
		Discovered list of instanced prim paths: []`

The "last" error message in the traceback isn't always the most descriptive. If you resolve those errors and still get the malloc error then there might be something else going on, but when I get a nondescript error message like that it's almost always either my robot model or how I configured the scene.

RandomOakForest, 2025-01-02:
Thank you for posting this. @SanskarSoni8 let us know if you still have this issue.

closed 2025-02-15 by RandomOakForest