isaac-sim/IsaacLab #1541: [Question] Can I choose different low-level policy actions?

H-Hisamichi, 2024-12-14:
Dear All,

I'm working on a task to learn a High-Level Policy that selects the output of two Low-Level Policies.
I am referring to the navigation task demo code, but this code seems to only support a single Low-Level Policy.
How can I implement the ability to select the inference results of two learned policies by the action of a High-Level Policy?
I would be glad to get your opinion!

Best regards,

H-Hisamichi, 2024-12-15:
The learning detail I'm trying to implement is to choose whether to enter policy 1 or policy 2 for each actor by the High-Level Policy action (normalized to 0, 1 or -1, 1).

closed 2024-12-20 by H-Hisamichi

ZhangHan0301, 2025-02-17:
> The learning detail I'm trying to implement is to choose whether to enter policy 1 or policy 2 for each actor by the High-Level Policy action (normalized to 0, 1 or -1, 1).

ZhangHan0301, 2025-02-17:
hello! 
"I want to implement a navigation task, similar to the high-level policy you mentioned earlier. Could you tell me which open-source navigation task you referred to?"

I would be glad to get your opinion!

Best regards,

H-Hisamichi, 2025-03-06:
Hi @ZhangHan0301,

Sorry for the delayed response.  
I referred to the `navigation` task in `manager_based`. Currently, the project involving a high-level policy that selects multiple low-level policies is on hold, and there has been no progress.  

I hope this helps!