isaac-sim/IsaacLab #1549: [Question] Observation that depends on an event

kousheekc, 2024-12-16:
Hello, 

I had a question that does not seem to be covered in the documentation and I am a bit stuck with the implementation details. What I am trying to achieve is to train an agent for impulse rejection using a feedforward based technique. To provide some context I want to train an aerial robot to counteract **predictable** impulse disturbances by providing a feedforward "warning" signal to the controller so that the drone can preemptively perform a corrective maneuver to not get destabilized when the actual impulse takes place. For the moment I have managed to add impulse disturbances to the robot using **EventCfg** in the following way:

```
push_robot = EventTerm(
        func=mdp.push_by_setting_velocity,
        mode="interval",
        interval_range_s=(10.0, 15.0),
        params={"velocity_range": {"x": (-0.5, 0.5), "y": (-0.5, 0.5)}},
    )
```

but I am now stuck with how I can relate this to an observation term. The idea is that I want to create an "artificial warning signal" that starts for example 1 second before the impulse takes place, and counts down to 0 till the impulse actually takes place. 

To clarify what I am trying to say I have attached a figure below. The red line represents the impulse disturbance (which may or may not be included in the observations) and the blue signal is a feedforward term that stays constant at some value and starts counting down as soon as the simulation time is for example 1 second away from the impulse taking place, till it reaches 0 when the impulse actually happens. The hope is that the agent can use this signal to learn the corrective maneuver it needs to perform to not get destabilized and "brace itself" preemptively.

[image]

Any help regarding this would be greatly appreciated. FYI I have implemented this using a **direct env configuration** but I would like to be able to do this using **manager based envs**. 

Thank you all!

KyleM73, 2024-12-17:
One way I have implemented something like this is by using commands:
- Write a command to trigger in some range (currently what you have as the sample range for the event)
- Have an observation report the status of the command (ie your warning system)
- X time steps after the command triggers (can be counted by the env step counter or an internal step counter if you make the event class-based) trigger the event (a push in your case)

So you would need to write your own command, event, and observation terms, but doing so would fit nicely within the manager-based framework. This is only one option of course, others may have different ideas on the "right" way to do this.

kousheekc, 2024-12-18:
That's an interesting approach, thanks for the pointer! Will try to implement it and keep posted about how it goes. Closing the issue for the moment, might reopen later if required.

closed 2024-12-18 by kousheekc