isaac-sim/IsaacLab #1552: [Bug] Articulation's default_joint_stiffness doesn't get set by code if Implicit Actuator is used

zoctipus, 2024-12-17:
If you are submitting a bug report, please fill in the following details and use the tag [bug].

### Describe the bug

line: 1333 - 1353 in lab.assets.articulation.py
```
         if isinstance(actuator, ImplicitActuator):
                self._has_implicit_actuators = True
                # the gains and limits are set into the simulation since actuator model is implicit
                self.write_joint_stiffness_to_sim(actuator.stiffness, joint_ids=actuator.joint_indices)
                self.write_joint_damping_to_sim(actuator.damping, joint_ids=actuator.joint_indices)
                self.write_joint_effort_limit_to_sim(actuator.effort_limit, joint_ids=actuator.joint_indices)
                self.write_joint_velocity_limit_to_sim(actuator.velocity_limit, joint_ids=actuator.joint_indices)
                self.write_joint_armature_to_sim(actuator.armature, joint_ids=actuator.joint_indices)
                self.write_joint_friction_to_sim(actuator.friction, joint_ids=actuator.joint_indices)
            else:
                # the gains and limits are processed by the actuator model
                # we set gains to zero, and torque limit to a high value in simulation to avoid any interference
                self.write_joint_stiffness_to_sim(0.0, joint_ids=actuator.joint_indices)
                self.write_joint_damping_to_sim(0.0, joint_ids=actuator.joint_indices)
                self.write_joint_effort_limit_to_sim(1.0e9, joint_ids=actuator.joint_indices)
                self.write_joint_velocity_limit_to_sim(actuator.velocity_limit, joint_ids=actuator.joint_indices)
                self.write_joint_armature_to_sim(actuator.armature, joint_ids=actuator.joint_indices)
                self.write_joint_friction_to_sim(actuator.friction, joint_ids=actuator.joint_indices)
                # Store the actual default stiffness and damping values for explicit actuators (not written the sim)
                self._data.default_joint_stiffness[:, actuator.joint_indices] = actuator.stiffness
                self._data.default_joint_damping[:, actuator.joint_indices] = actuator.damping
```

is it a design choice to not able to set cfg's stiffness and damping to default_joint_stiffness and default_joint_damping?

what will happen if a joint-drive-less usd uses this code is that the default_joint_stiffness and default_joint_damping will be 0, then a hard to debug issue will arise when we use eventCfg like `randomize_actuator_gains`, the randomization will apply with respect to default_joint_stiffness((which is now 0) not the current joint stiffness or damping. As shown below:

```
# Randomize stiffness
        if stiffness_distribution_params is not None:
            stiffness = actuator.stiffness[env_ids].clone()
            stiffness[:, actuator_indices] = asset.data.default_joint_stiffness[env_ids][:, global_indices].clone()
            randomize(stiffness, stiffness_distribution_params)
            actuator.stiffness[env_ids] = stiffness
            if isinstance(actuator, ImplicitActuator):
                asset.write_joint_stiffness_to_sim(stiffness, joint_ids=actuator.joint_indices, env_ids=env_ids)
        # Randomize damping
        if damping_distribution_params is not None:
            damping = actuator.damping[env_ids].clone()
            damping[:, actuator_indices] = asset.data.default_joint_damping[env_ids][:, global_indices].clone()
            randomize(damping, damping_distribution_params)
            actuator.damping[env_ids] = damping
            if isinstance(actuator, ImplicitActuator):
                asset.write_joint_damping_to_sim(damping, joint_ids=actuator.joint_indices, env_ids=env_ids)
```

the result stiffness and damping will be close to 0 and robot will likly to not move at all

Mayankm96, 2024-12-21:
That looks like a bug to me 👀

zoctipus, 2024-12-22:
Thanks for viewing! I will create a pull request regarding this then.

evanzijianhe, 2024-12-26:
Thanks for pointing out this bug. I also met the same problem. The `default_joint_damping`  and `default_joint_stiffness` cannot correctly read the default joint damping and stiffness which are set in the asset configuration file. This will result bad randomization in the `mdp.randomize_actuator_gains`. The proposed the method can correct this bug.

RandomOakForest, 2025-01-02:
Thanks for posting this and for the pull request.

commit e8ea1850 (2025-01-03): Fixes error in Articulation where default_joint_stiffness and default_joint_damping is not correctly set if actuator is instance of ImplicitActuator (#1580)

# Description

Before the fix, the field, default_joint_stiffness, and
default_joint_damping in omni.isaac.lab.assets.Articulation will be
the default value configured Joint Drive property in usd file, and
defining a new value with actuator config will not have effect
if the actuator is type ImplicitActuator

after the fix, all actuator types will be able to override
default_joint_stiffness and default_joint_damping

Fixes #1552 

## Type [... cut]
closed 2025-01-03 by kellyguo11