isaac-sim/IsaacLab #1556: [Bug Report] Ray caster failed to measure distance of given mesh properly

raincold23, 2024-12-17:
Hello, IsaacLab Team, 

While I was working on the `ManagerBasedRLEnvCfg`, I found that if a new terrain cfg is added into EnvCfg, the terrain scanner `RayCaster` cannot measure the hit pos properly given the new terrain prim path.

### Steps to reproduce

I want the new terrain be added on certain position and orientation, so the terrain is attached to a `Xform`. The new terrain is added as follow: 
```python
@configclass
class MyEnvCfg(ManagerBasedRLEnvCfg):

    def __post_init__(self):
        """Post initialization."""
        self.scene.upper = AssetBaseCfg(
            prim_path="/World/upper",
            spawn=sim_utils.DiskLightCfg(visible=False, , prim_type="Xform"),
            init_state=AssetBaseCfg.InitialStateCfg(pos=(0.0, 0.0, 10.2), rot=(0.0, 0.0, 1.0, 0.0)),
        )
        self.scene.upper_terrain = TerrainImporterCfg(
            prim_path="/World/upper/ground",
            terrain_type="generator",
            terrain_generator=UPPER_MASK_TERRAINS_CFG,
            max_init_terrain_level=9,
            collision_group=-1,
            physics_material=sim_utils.RigidBodyMaterialCfg(
                friction_combine_mode="multiply",
                restitution_combine_mode="multiply",
                static_friction=1.0,
                dynamic_friction=1.0,
            ),
            visual_material=sim_utils.MdlFileCfg(
                mdl_path=f"{ISAACLAB_NUCLEUS_DIR}/Materials/TilesMarbleSpiderWhiteBrickBondHoned/TilesMarbleSpiderWhiteBrickBondHoned.mdl",
                project_uvw=True,
                texture_scale=(0.25, 0.25),
            ),
            debug_vis=False,
        )
        self.scene.lower_terrain_scanner = RayCasterCfg(
            prim_path = "{ENV_REGEX_NS}/Robot/body",
            offset=RayCasterCfg.OffsetCfg(pos=(0.0, 0.0, -20.0), rot=(0.0,0.0,1.0,0.0)),
            attach_yaw_only=True,
            pattern_cfg=patterns.GridPatternCfg(resolution=0.1, size=[1.6, 1.0]),
            debug_vis=True,
            mesh_prim_paths=["/World/upper/ground"],
        )
```

While running this env, the new terrain is added to the World and collision of new terrain works well.
[image]

However, the `RayCaster` result is weird. It seems that from the perspective of `lower_terrain_scanner`, the new added terrain is still at the initial pos and ori.
[image]

So, the problem is why `RayCaster` assumes the terrain remains the initial pos and ori, even if the terrain is already translated and rotated.
### System Info

Describe the characteristic of your environment:

- Commit: 5d86a8bf6bd1d6b8b13ab157148e4596e109261b
- Isaac Sim Version:4.2.0
- OS: Ubuntu 20.04
- GPU: RTX 3070

### Additional context

Add any other context about the problem here.

### Checklist

- [x] I have checked that there is no similar issue in the repo (**required**)
- [x] I have checked that the issue is not in running Isaac Sim itself and is related to the repo

amatsukaze233, 2024-12-24:
You can refer to https://github.com/isaac-sim/IsaacLab/pull/1448, and calling self._initialize_warp_meshes() in _update_buffers_impl seems to detect dynamic objects.

RandomOakForest, 2025-01-02:
Thank you for posting this. Please reopen the bug report if you still face this issue, will close for now.