isaac-sim/IsaacLab #1558: [Question] applied_torque vs root_physx_view.get_measured_joint_efforts

Jdowdy05, 2024-12-17:
### applied_torque vs get_measured_joint_efforts vs get_measured_joint_forces

Hello, I am checking my understanding of the following three functions and what they are returning. 

"Applied_torque" is the calculated torque from the actuator that does not account for any external wrenches on the actuator.

"[get_measured_joint_efforts](https://docs.omniverse.nvidia.com/py/isaacsim/source/extensions/omni.isaac.core/docs/index.html?highlight=get_measured_#omni.isaac.core.articulations.Articulation.get_measured_joint_efforts)" is the external torque applied to the joints from the environment. It is essentially the "[get_measured_joint_forces](https://docs.omniverse.nvidia.com/py/isaacsim/source/extensions/omni.isaac.core/docs/index.html?highlight=get_measured_#omni.isaac.core.articulations.Articulation.get_measured_joint_forces)" wrench converted to joint space for the articulation. Im also assuming that the "get_measured_joint_efforts" is just the reaction torque and not a sum of applied_torque

I have read the documentation for both Isaac lab and Isaac sim a few times, but I am still a bit confused about "get_measured_joint_efforts," as it doesn't explicitly say you receive the reaction forces in torque like "get_measured_joint_forces"  does.

Another question along the same line (though I think I know the answer): The observation function "body_incoming_wrench" has a function call inside called "get_link_incoming_joint_force". Does this give a future reaction force by some +sim.dt ahead?

Thank you in advance for the help!

RandomOakForest, 2025-01-02:
Thank you for posting this. I'll review with the team and post here any findings.

heleiduan, 2025-01-18:
Referring to this [API doc](https://docs.omniverse.nvidia.com/py/isaacsim/source/extensions/omni.isaac.core/docs/index.html?highlight=get_measured_joint_efforts#omni.isaac.core.articulations.ArticulationView.get_measured_joint_efforts), it seems that `get_measured_joint_efforts()` and `get_max_efforts()` from `ArticulatioView` cannot be accessed properly with the following errors, 
```  
torques = asset.data._root_physx_view.get_measured_joint_efforts()[:, asset_cfg.joint_ids]
AttributeError: 'ArticulationView' object has no attribute 'get_measured_joint_efforts'
```
While others such as `get_mass_matrices()` works properly.

RandomOakForest, 2025-02-15:
Thank you for following up. I will move this to our _Discussions_ section for the team to address.