isaac-sim/IsaacLab #1560: [Bug Report] Interpenetration of rigid object and robot in Factory Nut and Bot environment

kbkartik, 2024-12-18:
newbie to the sim2real world. This bug is similar to the question I raised earlier #1515. I am not sure if this qualifies as a bug or if I am making some mistake.

I import the newly added Factory Nut and Bolt environment and replace the Franka to Shadow Hand. Unlike #1515, I am not experimenting with contact sensor and adding physics material to objects. I simply check if physical interactions between robot and objects are valid.

### Describe the bug

There are two observations/bugs I want to highlight.

1) When dropped from a height, nut penetrates through the hand and falls on the floor.
2) When it falls on the bolt, nut and bolt interpenetrate each other.

In both scenarios, collisions should take place between nut and hand, and between nut and bolt.

### Steps to reproduce

1. Set the initial state of [shadow_hand_config](https://github.com/isaac-sim/IsaacLab/blob/main/source/extensions/omni.isaac.lab_assets/omni/isaac/lab_assets/shadow_hand.py) to the following:

```
init_state=ArticulationCfg.InitialStateCfg(
        pos=(0.0, 0.0, 0.5),
        rot=(0.0, 0.0, 0.0, 0.0),
        joint_pos={".*": 0.0},
    ),
```

2. Set the initial state of [fixed_asset](https://github.com/isaac-sim/IsaacLab/blob/main/source/extensions/omni.isaac.lab_tasks/omni/isaac/lab_tasks/direct/factory/factory_tasks_cfg.py) in `NutThread` class.
```
init_state=ArticulationCfg.InitialStateCfg(
            pos=(0.0, -0.35, 0.0), rot=(1.0, 0.0, 0.0, 0.0), joint_pos={}, joint_vel={}
        ),
```

3. Set the initial state of [held_asset](https://github.com/isaac-sim/IsaacLab/blob/main/source/extensions/omni.isaac.lab_tasks/omni/isaac/lab_tasks/direct/factory/factory_tasks_cfg.py) in `NutThread` class.
```
init_state=ArticulationCfg.InitialStateCfg(
            pos=(0.0, -0.35, 3.0), rot=(1.0, 0.0, 0.0, 0.0), joint_pos={}, joint_vel={}
        ),
```
4. Set `disable_gravity` to False in [held_asset](https://github.com/isaac-sim/IsaacLab/blob/main/source/extensions/omni.isaac.lab_tasks/omni/isaac/lab_tasks/direct/factory/factory_tasks_cfg.py) in `NutThread` class.

5. Add `from omni.isaac.lab_assets import SHADOW_HAND_CFG` in [factory_env_cfg](https://github.com/isaac-sim/IsaacLab/blob/main/source/extensions/omni.isaac.lab_tasks/omni/isaac/lab_tasks/direct/factory/factory_env_cfg.py) and set `robot = SHADOW_HAND_CFG` in `FactoryEnvCfg` class.

### System Info

Describe the characteristic of your environment:

- Commit: 019e3523cee5597946210e5d7584e636c469b38b
- Isaac Sim Version: 4.2.0-rc.17+release.15988.c99988b1.gl
- OS: Ubuntu 20.04
- GPU: RTX4500
- CUDA: 12.6
- GPU Driver: 560.35.03

### Checklist

- [x] I have checked that there is no similar issue in the repo (**required**)
- [ ] I have checked that the issue is not in running Isaac Sim itself and is related to the repo

### Acceptance Criteria

Collision should happen when:

- [ ] When nut falls on the hand.
- [ ] When nut falls on the bolt.

RandomOakForest, 2025-01-02:
Thank you for posting this. I've tagged this for the team to investigate. As we return from time away and the latest release work, we should be able to address this soon.

kbkartik, 2025-01-06:
What are collision messes? And how to visualize them?

Also, please see [factory_nutbolt_check.py](https://gist.github.com/kbkartik/9333b658f9233d4092318e3d4bd41cc2) gist to reproduce above issue.

mpgussert, 2025-02-18:
Hello Kbartik!

Collision meshes are simplified meshes that are used to actually perform the simulation.  There are numerous ways to calculate if two meshes are intersecting, but regardless of the method, it scales with mesh complexity:  Generally, the more vertexes in the mesh, the more costly it is to figure out collisions. 

Collision meshes can be visualized in the GUI by clicking on the "eye" icon at the top of the viewport and then selecting **show by type > physics > colliders > all**

kbkartik, 2025-03-05:
So, higher the mesh complexity, more natural the physics simulation is right? At the expense of computation time.

When I checked the collision type of shadow hand, they use very simple mesh types like cylinder and cuboid.  Then, how do they achieve high reward performance for cube reorientation task? The physics simulation should be inaccurate due to simpler meshes models, leading to poor reward performance?

ooctipus, 2025-06-26:
Hello Kbartik,

Simpler mesh doesn't not necessarily results in poor reward performance. Imagine a hand made of purely rectangular cubes, and capsules. "Mine Craft Hand", it is simplified but it is still able to perform task like reorient, pick and place, etc. To optimize for the best world between realism(natural) and speed, we often pick a reasonable mesh simplification so we can move forward. A lot of time when simplification are done correctly, there is actually very little sim to real gap. 

mesh complexity changes as we choose different collision type. For example, triangle collider for static mesh is very accurate and fast, sdf collider is very accurate but a bit slower and more expensive.  Convex Hull if fast but can be inaccurate if your object is obviously non convex. Convex composition is also very fast and can somewhat coarsely represent the non_convex objects through multiple Convex Hulls, but it can still be not quite accurate if your objects has thin part or hole-like part,

I hope this anwsers your questions