isaac-sim/IsaacLab #1563: [Question] Can I generate actions that are not input to the actor?

H-Hisamichi, 2024-12-18:
Dear All,

I'm working on a project to train a high-level policy that selects multiple low-level policies. I want to generate actions that are not input to the actor as joint positions. I found `BinaryJointAction`, but it seems to be a type where actions are input to the actor. Is there a way to generate actions that are not input to the actor? If it's not possible by default, I’m thinking of implementing it as a custom MDP.
Thank you!

Jdowdy05, 2024-12-18:
If you use the direct workflow environment, I think you can do this as there is a function to apply the action directly to the robot. If you look at the locomotion environment under "IsaacLab/source/extensions/omni.isaac.lab_tasks/omni/isaac/lab_tasks/direct/" you'll see how there is pre_physics_step where you generate your action and then a apply_action step where you can then do whatever you need with the action.

H-Hisamichi, 2024-12-19:
Hello @Jdowdy05,

Actually, I was working with the direct workflow before. However, I switched to the manager-based workflow because I couldn't resolve the following error that occurred when feeding the inference values of the trained policy (JIT *.pt) into the robot joints. If the manager-based workflow does not have a function to generate actions that do not directly input to the joints, I think I will try to address this error further.

```
################################################################################
                       Learning iteration 14/5000

                       Computation: 5451 steps/s (collection: 4.392s, learning 0.116s)
               Value function loss: 720.8562
                    Surrogate loss: -0.0014
             Mean action noise std: 1.01
                       Mean reward: 1.08
               Mean episode length: 105.16
Episode_Reward/track_lin_vel_w_xy_exp: 0.0633
Episode_Reward/track_ang_vel_z_exp: 0.0165
        Episode_Reward/lin_vel_z_b: -0.0180
     Episode_Reward/dof_torques_l2: -0.0055
         Episode_Reward/dof_acc_l2: -0.1054
     Episode_Reward/action_rate_l2: -0.0419
 Episode_Reward/undesired_contacts: 0.0000
Episode_Reward/flat_orientation_l2: 0.0427
     Episode_Reward/z_position_err: 0.0999
  Episode_Termination/base_contact: 3.7500
      Episode_Termination/time_out: 0.6667
--------------------------------------------------------------------------------
                   Total timesteps: 368640
                    Iteration time: 4.51s
                        Total time: 70.99s
                               ETA: 23598.0s

_processed_actions_test_NaN: False
_processed_actions_test_Inf: False
_processed_actions_test_NaN: False
_processed_actions_test_Inf: False
_processed_actions_test_NaN: False
_processed_actions_test_Inf: False
_processed_actions_test_NaN: False
_processed_actions_test_Inf: False
_processed_actions_test_NaN: False
_processed_actions_test_Inf: False
_processed_actions_test_NaN: False
_processed_actions_test_Inf: False
Error executing job with overrides: []
Traceback (most recent call last):
  File "/home/robot/IsaacLab_v1.2.0/IsaacLab/source/extensions/omni.isaac.lab_tasks/omni/isaac/lab_tasks/utils/hydra.py", line 99, in hydra_main
    func(env_cfg, agent_cfg, *args, **kwargs)
  File "/home/robot/IsaacLab_v1.2.0/IsaacLab/source/standalone/workflows/rsl_rl/train.py", line 144, in main
    runner.learn(num_learning_iterations=agent_cfg.max_iterations, init_at_random_ep_len=True)
  File "/home/robot/anaconda3/envs/iccar2025/lib/python3.10/site-packages/rsl_rl/runners/on_policy_runner.py", line 111, in learn
    actions = self.alg.act(obs, critic_obs)
  File "/home/robot/anaconda3/envs/iccar2025/lib/python3.10/site-packages/rsl_rl/algorithms/ppo.py", line 73, in act
    self.transition.actions = self.actor_critic.act(obs).detach()
  File "/home/robot/anaconda3/envs/iccar2025/lib/python3.10/site-packages/rsl_rl/modules/actor_critic.py", line 104, in act
    self.update_distribution(observations)
  File "/home/robot/anaconda3/envs/iccar2025/lib/python3.10/site-packages/rsl_rl/modules/actor_critic.py", line 101, in update_distribution
    self.distribution = Normal(mean, mean * 0.0 + self.std)
  File "/home/robot/.local/share/ov/pkg/isaac-sim-4.2.0/exts/omni.isaac.ml_archive/pip_prebundle/torch/distributions/normal.py", line 57, in __init__
    super().__init__(batch_shape, validate_args=validate_args)
  File "/home/robot/.local/share/ov/pkg/isaac-sim-4.2.0/exts/omni.isaac.ml_archive/pip_prebundle/torch/distributions/distribution.py", line 70, in __init__
    raise ValueError(
ValueError: Expected parameter loc (Tensor of shape (1024, 18)) of distribution Normal(loc: torch.Size([1024, 18]), scale: torch.Size([1024, 18])) to satisfy the constraint Real(), but found invalid values:
tensor([[ 0.1859, -0.1425,  0.1355,  ..., -0.1009,  0.1226,  0.1512],
        [ 0.1282, -0.0371,  0.6706,  ..., -0.0529,  0.4659, -0.1631],
        [-0.2442, -0.3320,  0.1529,  ..., -0.0195,  0.6803,  0.7541],
        ...,
        [-0.3731, -0.3420,  0.5202,  ...,  0.4025,  0.5078,  0.0750],
        [ 0.3426, -0.0826,  0.1956,  ..., -0.3085, -0.0036,  0.1367],
        [-0.0207, -0.3074,  0.3149,  ..., -0.0626,  0.2548, -0.0025]],
       device='cuda:0')
```

H-Hisamichi, 2024-12-19:
Dear All,

The error related to replaying the trained policy has been resolved. However, the trained policy does not function as expected. My trained policy was learned using a manager-based workflow, and it is challenging to reproduce the observations in a direct workflow, especially replicating `mdp.JointPositionActionCfg`.

Is there a way to generate binary actions that are not directly input to the joints in the manager-based workflow?

I would appreciate it if you could help me.

closed 2024-12-22 by H-Hisamichi