isaac-sim/IsaacLab #1569: [Question] How to Tune the DifferentialIKController

amatsukaze233, 2024-12-20:
### Question
When tuning the DifferentialIKController, it can get stuck when calculating a distant target position. It would be helpful to set up an alert for target points that are out of the working range.
` actuators={
        "joint": ImplicitActuatorCfg(
            joint_names_expr=["joint[1-6]"],
            velocity_limit=6.0,
            stiffness=1e5,
            damping=1e3,
        ),`
`diff_ik_cfg = DifferentialIKControllerCfg(command_type="pose", use_relative_mode=False, ik_method="dls")`
[image]

[image]

RandomOakForest, 2025-01-02:
Thanks for the suggestion. Currently, it seems doing this check before the `compute` step would be a good workaround. [This tutorial may help](https://isaac-sim.github.io/IsaacLab/main/source/tutorials/05_controllers/run_diff_ik.html).

amatsukaze233, 2025-01-09:
> Thanks for the suggestion. Currently, it seems doing this check before the `compute` step would be a good workaround. [This tutorial may help](https://isaac-sim.github.io/IsaacLab/main/source/tutorials/05_controllers/run_diff_ik.html).

Thank you for your response. Currently, I can only try to clip the input error values in omni.isaac.lab.utils.math.compute_pose_error and record the previous results in omni.isaac.lab.controllers.differential_ik._compute_delta_joint_pos to determine if the joints are jittering. I hope there will be better methods in the future.

closed 2025-01-09 by amatsukaze233

JerryBonjour, 2025-06-26:
> ### Question
> When tuning the DifferentialIKController, it can get stuck when calculating a distant target position. It would be helpful to set up an alert for target points that are out of the [...]

how to drag the end effector like your demonstration