isaac-sim/IsaacLab #1571: [Question] UniformPoseCommandCfg in world frame for floating base environments.

kousheekc, 2024-12-20:
Hello, trying to implement something but it does not seem to be covered in the documentation so I am a bit confused. I am creating a manager based drone environment (inspired by the direct env that exists already) for position control i.e the commands in the manager based configuration are 3D poses that act as targets for the drone. So I implemented the following CommandsCfg by taking inspiration of the various manipulation tasks that have similar functions.

```
@configclass
class CommandsCfg:
    """Command terms for the MDP."""

    target_pose = mdp.UniformPoseCommandCfg(
        asset_name="robot",
        body_name="body",
        resampling_time_range=(1.0e9, 1.0e9), # no resampling in the episode
        debug_vis=True,
        ranges=mdp.UniformPoseCommandCfg.Ranges(
            pos_x=(-0.5, 0.5), pos_y=(-0.5, 0.5), pos_z=(0.1, 0.5), roll=(0.0, 0.0), pitch=(0.0, 0.0), yaw=(0.0, 0.0)
        ),
    )
```

The issue is that this creates a target pose wrt the body frame as you can see from the gif attached below. Which means that since the drone is a floating body, the target pose moves with the drone which is not what I want. How could I create a command that is "fixed" in the world frame the **UniformPoseCommandCfg**.

[Screencast from 12-20-2024 04:14:28 PM.webm](https://github.com/user-attachments/assets/b033d351-1e5f-4bec-bfc8-510419eae2ba)

Of course I understand that I can implement my own command config but I want hoping I could use this since it exists already.

Thank you in advance for the help and I look forward to the responses!

RandomOakForest, 2025-01-02:
Thank you for posting this. We are experiencing a natural delay with the holidays. We should be able to review this soon.

StrainFlow, 2025-01-15:
I just wanted to let you know we haven't forgotten about you. I know how to do this in a direct environment, but not in a managed environment yet. Quite a few people were out for the holidays and I'm reaching out to a few people as they return.

closed 2025-02-19 by RandomOakForest

quacklemtduck, 2025-02-26:
Did you manage to solve this? And if so, how? @RandomOakForest

StrainFlow, 2025-03-04:
@quacklemtduck are you doing something similar? Managed or direct environment?

kousheekc, 2025-03-12:
@quacklemtduck I have a hacky solution which is good enough for me for the moment. In the scene configuration of the manager based environment I define a "fixed" dummy object and I create my UniformPoseCommands based on this dummy object and while passing the observations, I do a frame transform to get the pose of the target in the drones body frame.