isaac-sim/IsaacLab #1574: [Question] How to load custom articulation with joint into isaaclab

lxy270, 2024-12-21:
hi, I am trying to load an articulation(simply a sphere with a D6 joint) into the scene, but I got the following error indicating the joint is not recognized, I wonder where could went wrong

the error I got:
`Traceback (most recent call last):
  File "/home/xinyili/IsaacLab/source/extensions/omni.isaac.lab/omni/isaac/lab/assets/asset_base.py", line 92, in <lambda>
    lambda event, obj=weakref.proxy(self): obj._initialize_callback(event),
  File "/home/xinyili/IsaacLab/source/extensions/omni.isaac.lab/omni/isaac/lab/assets/asset_base.py", line 263, in _initialize_callback
    self._initialize_impl()
  File "/home/xinyili/IsaacLab/source/extensions/omni.isaac.lab/omni/isaac/lab/assets/articulation/articulation.py", line 942, in _initialize_impl
    self._process_cfg()
  File "/home/xinyili/IsaacLab/source/extensions/omni.isaac.lab/omni/isaac/lab/assets/articulation/articulation.py", line 1069, in _process_cfg
    indices_list, _, values_list = string_utils.resolve_matching_names_values(
  File "/home/xinyili/IsaacLab/source/extensions/omni.isaac.lab/omni/isaac/lab/utils/string.py", line 364, in resolve_matching_names_values
    raise ValueError(
ValueError: Not all regular expressions are matched! Please check that the regular expressions are correct: 
	.*: []
Available strings: []

[INFO]: Setup complete...
Traceback (most recent call last):
  File "/home/xinyili/IsaacLab/3d_action.py", line 176, in <module>
    main()
  File "/home/xinyili/IsaacLab/3d_action.py", line 171, in main
    run_simulator(sim, scene)
  File "/home/xinyili/IsaacLab/3d_action.py", line 136, in run_simulator
    scene.reset()
  File "/home/xinyili/IsaacLab/source/extensions/omni.isaac.lab/omni/isaac/lab/scene/interactive_scene.py", line 347, in reset
    articulation.reset(env_ids)
  File "/home/xinyili/IsaacLab/source/extensions/omni.isaac.lab/omni/isaac/lab/assets/articulation/articulation.py", line 166, in reset
    for actuator in self.actuators.values():
AttributeError: 'Articulation' object has no attribute 'actuators'`

I save the articulation in the isaacsim GUI by ‘save selected’ of the sphere and the joint where the sphere is set to be the articulation root, since I cannot upload usd file in the forum, here I attach the usda version of my articulation
`#usda 1.0
(
    defaultPrim = "Root"
    upAxis = "Z"
)

def Xform "Root"
{
    def Sphere "Sphere" (
        apiSchemas = ["PhysicsArticulationRootAPI", "PhysxArticulationAPI", "PhysicsRigidBodyAPI", "PhysxRigidBodyAPI", "PhysicsCollisionAPI", "PhysxCollisionAPI"]
    )
    {
        float3[] extent = [(-0.5, -0.5, -0.5), (0.5, 0.5, 0.5)]
        bool physics:collisionEnabled = 1
        bool physics:kinematicEnabled = 0
        bool physics:rigidBodyEnabled = 1
        double radius = 0.5
        custom bool refinementEnableOverride = 1
        custom int refinementLevel = 2
        token ui:displayGroup = "Material Graphs"
        token ui:displayName = "Sphere"
        int ui:order = 1024
        float3 xformOp:rotateXYZ = (0, -0, 0)
        float3 xformOp:scale = (1, 1, 1)
        double3 xformOp:translate = (0, 0, 0)
        uniform token[] xformOpOrder = ["xformOp:translate", "xformOp:rotateXYZ", "xformOp:scale"]

        def PhysicsJoint "D6Joint" (
            apiSchemas = ["PhysicsLimitAPI:transX", "PhysicsLimitAPI:transY", "PhysicsLimitAPI:transZ", "PhysicsLimitAPI:rotX", "PhysicsLimitAPI:rotY", "PhysicsLimitAPI:rotZ"]
        )
        {
            float limit:rotX:physics:high = -1
            float limit:rotX:physics:low = 1
            float limit:rotY:physics:high = -1
            float limit:rotY:physics:low = 1
            float limit:rotZ:physics:high = -1
            float limit:rotZ:physics:low = 1
            float limit:transX:physics:high = -1
            float limit:transX:physics:low = 1
            float limit:transY:physics:high = -1
            float limit:transY:physics:low = 1
            float limit:transZ:physics:high = -1
            float limit:transZ:physics:low = 1
            rel physics:body1 = </Root/Sphere>
            float physics:breakForce = inf
            float physics:breakTorque = inf
            point3f physics:localPos0 = (0, 0, 0)
            point3f physics:localPos1 = (0, 0, 0)
            quatf physics:localRot0 = (1, 0, 0, 0)
            quatf physics:localRot1 = (1, 0, 0, 0)
        }
    }
}`

the code I use to load the articulation:
`@configclass
class BouncingCfg(InteractiveSceneCfg):
    sphere = ArticulationCfg(
        prim_path="{ENV_REGEX_NS}/Sphere",
        # spawn=sim_utils.SphereCfg(
        spawn=sim_utils.UsdFileCfg(usd_path="/home/xinyili/IsaacLab/Sphere_joint.usd",
           rigid_props=sim_utils.RigidBodyPropertiesCfg(
               rigid_body_enabled=True,
               max_linear_velocity=1000.0,
               max_angular_velocity=1000.0,
               max_depenetration_velocity=100.0,
               enable_gyroscopic_forces=True,
           ),
           articulation_props=sim_utils.ArticulationRootPropertiesCfg(
               enabled_self_collisions=False,
               solver_position_iteration_count=4,
               solver_velocity_iteration_count=0,
               sleep_threshold=0.005,
               stabilization_threshold=0.001,
           ),
        ),
        init_state=ArticulationCfg.InitialStateCfg(pos=[0, 0, 1], lin_vel=[10, 0.3, 0]),
        actuators={
           "actuator": ImplicitActuatorCfg(
               joint_names_expr=["D6Joint"],
               effort_limit=400.0,
               velocity_limit=100.0,
               stiffness=0.0,
               damping=10.0,
           ),
        }
    )

def run_simulator(sim: sim_utils.SimulationContext, scene: InteractiveScene):
    sphere = scene["sphere"]
    sim_dt = sim.get_physics_dt()
    scene.reset()
    
    # Simulation loop
    while simulation_app.is_running():
        scene.write_data_to_sim()
        sim.step()
        scene.update(sim_dt)

def main():
    sim_cfg = sim_utils.SimulationCfg(device=args_cli.device, dt=0.01)
    sim = SimulationContext(sim_cfg)
    sim.set_camera_view([5, 0.0, 8], [0.0, 0.0, 2.0])
    scene_cfg = BouncingCfg(num_envs=args_cli.num_envs, env_spacing=20)
    scene = InteractiveScene(scene_cfg)
    sim.reset()
    run_simulator(sim, scene)`

any help would be appreciated, thanks!

RandomOakForest, 2025-01-03:
Thanks for posting this. There seems to be an issue with your saved articulation because you got `Available strings: []`. Can you confirm that you can open the newly saved assets you saved using "save selected" in Isaac Sim?

jiangranlv, 2025-01-06:
I met the same issue

RandomOakForest, 2025-02-15:
Following up, maybe using the latest release of Isaac Sim and Isaac Lab in a fresh install would help. Let us know if we can be of further help. If you still face this issue, please open a _bug report_. Thanks.

closed 2025-02-15 by RandomOakForest

omarrayyann, 2025-03-22:
I am facing the same issue.

omarrayyann, 2025-03-22:
@lxy270 Did you figure this out?

lozoRUST, 2025-07-02:
@lxy270 @jiangranlv @omarrayyann @RandomOakForest  Facing the same problem. Have you solved this?

lozoRUST, 2025-07-02:
@RandomOakForest hi, but i'm wondering why other joints (PhysicsRevoluteJoint, PhysicsPrimaticJoint, PhysicsFixedJoint) are not recognized by the ImplicitActuatorCfg. Here's my post describing my problem.[https://forums.developer.nvidia.com/t/joint-match-problems-in-scripts/337828](url)