isaac-sim/IsaacLab #1583: [Proposal] Many naming inconsistencies in the joint properties inside Articulation

Mayankm96, 2024-12-22:
### Proposal

Going over recent changes in the Articulation class and there are a few inconsistencies that have popped up in the recent merges.

* Variables such as `joint_armature` and `joint_friction` are not a part of ArticulationData
* It is unclear what `joint_limits` refer to unless we start digging into the code?
* There are attributes for `joint_velocity_limits` but at the same time there isn't `joint_efforts_limits`
* There are attributes like `soft_joint_pos_limits` and `soft_joint_vel_limits` which are not getting updated when one calls `write_xxx_to_sim` methods
* As attributes are exploding, it is getting very unclear what is meant by "default" and non-default members in the class. We should be documenting this very explicitly and clearly.

### Motivation

Would be great to make sure we stay consistent in our code and operations inside the classes.

### Checklist

- [x] I have checked that there is no similar issue in the repo (**required**)

### Acceptance Criteria

- [ ] The above issues get resolved.

RandomOakForest, 2025-01-03:
Thank you for posting this suggestion. It has been duly noted.

Mayankm96, 2025-02-15:
The more I work with the Articulation class, the huger the mess seems.

The `default_` parameters, for instance. They are set to the values from inside the USD and not the values passed from the configuration. This isn't what is expected IMO. The configured values should be the default values.

Also `default_joint_stiffness` and `joint_stiffness` are not the same "set". The former has stiffness/damping from explicit, while the latter only has the values set inside the simulation. This makes things buggy.

Mayankm96, 2025-02-15:
Made the MR #1875 for this.

To be fair, I want to deprecate all the `joint_xxx` parameters to make them `joint_xxx_sim` (for explicit clarity). But this breaks too many things and can be annoying. Since the docstrings mention they are the value provided to the simulation, I can live with that for now. Thoughts?

closed 2025-03-07 by Mayankm96