isaac-sim/IsaacLab #1592: [Bug Report] Bug title

toenails6, 2024-12-24:
If you are submitting a bug report, please fill in the following details and use the tag [bug].

### Describe the bug

Launching Isaac Sim 4.2.0 from omniverse seems to be fine, but launching it from Isaac Lab would yield "Failed to import python module omni.isaac.core". The same issue happened to some others, and this only seems to be a very recent thing. Reference: https://forums.developer.nvidia.com/t/failed-to-import-python-module-omni-isaac-core/317674. 

The Isaac Sim 4.2.0 that I installed on another computer approximately three months ago still work well with Isaac Lab and the old scripts that I have written, including the given standalone examples. But nothing works with newly installed Isaac Sim 4.2.0. 

### Steps to reproduce

In my situation it was only the installation of a new Isaac Sim 4.2.0 and Isaac Lab, basically a fresh start, but nothing works. 

### System Info

Describe the characteristic of your environment:

- Commit: [dd044a4977722d85e966c5b63949093c9b14667f]
- Isaac Sim Version: [4.2.0-rc.17+release.15988.c99988b1.gl]
- OS: [Windows]
- GPU: [RTX 4090 16GB]
- CUDA: [12.7]
- GPU Driver: [566.36]

### Additional context

Note that the given error log from the terminal is from a pip installation of Isaac Sim 4.2.0, as I attempted the method given in the mentioned Nvidia link, but with the sources from my old computer's Isaac Sim 4.2.0. That evidently did not work, and even according to the Nvidia link I was supposed to 4.0.0's files anyway. My guess is the issue runs deeper than what the Nvidia link suggested. 

So I had to resort to using a pip installed Isaac Sim 4.2.0 for the following error logs. 
According to the Nvidia link given above, this seems to be a very recent issue. 

### Checklist

- [x] I have checked that there is no similar issue in the repo (**required**)
- [x] I have checked that the issue is not in running Isaac Sim itself and is related to the repo

### Acceptance Criteria

Add the criteria for which this task is considered **done**. If not known at issue creation time, you can add this once the issue is assigned.

- [ ] Clean start install of Isaac Sim 4.2.0 and Isaac Lab runs normally. 
- [ ] Clean start pip install of Isaac Sim 4.2.0 and Isaac Lab runs normally.

Vertax42, 2024-12-25:
I have the same issue, when i try to use the ManagerBasedRLEnvWindow by import ManagerBasedRLEnvWindow from omni.isaac.lab.envs.ui, here is what i got:

> Traceback (most recent call last):
>   File "/home/vertax/IsaacLab/source/standalone/workflows/rsl_rl/play_with_keyboard.py", line 12, in <module>
>     from omni.isaac.lab.envs.ui import ManagerBasedRLEnvWindow
>   File "/home/vertax/IsaacLab/source/extensions/omni.isaac.lab/omni/isaac/lab/envs/__init__.py", line 45, in <module>
>     from . import mdp, ui
>   File "/home/vertax/IsaacLab/source/extensions/omni.isaac.lab/omni/isaac/lab/envs/mdp/__init__.py", line 18, in <module>
>     from .actions import *  # noqa: F401, F403
>   File "/home/vertax/IsaacLab/source/extensions/omni.isaac.lab/omni/isaac/lab/envs/mdp/actions/__init__.py", line 8, in <module>
>     from .actions_cfg import *
>   File "/home/vertax/IsaacLab/source/extensions/omni.isaac.lab/omni/isaac/lab/envs/mdp/actions/actions_cfg.py", line 9, in <module>
>     from omni.isaac.lab.managers.action_manager import ActionTerm, ActionTermCfg
>   File "/home/vertax/IsaacLab/source/extensions/omni.isaac.lab/omni/isaac/lab/managers/__init__.py", line 13, in <module>
>     from .action_manager import ActionManager, ActionTerm
>   File "/home/vertax/IsaacLab/source/extensions/omni.isaac.lab/omni/isaac/lab/managers/action_manager.py", line 20, in <module>
>     from omni.isaac.lab.assets import AssetBase
>   File "/home/vertax/IsaacLab/source/extensions/omni.isaac.lab/omni/isaac/lab/assets/__init__.py", line 41, in <module>
>     from .articulation import Articulation, ArticulationCfg, ArticulationData
>   File "/home/vertax/IsaacLab/source/extensions/omni.isaac.lab/omni/isaac/lab/assets/articulation/__init__.py", line 8, in <module>
>     from .articulation import Articulation
>   File "/home/vertax/IsaacLab/source/extensions/omni.isaac.lab/omni/isaac/lab/assets/articulation/articulation.py", line 16, in <module>
>     import omni.isaac.core.utils.stage as stage_utils
> ModuleNotFoundError: No module named 'omni.isaac.core'
> There was an error running python
Please help me to solve this problem

toenails6, 2024-12-25:
Thus far from what I can tell, all I can say is the Nvidia link was at least correct in part, from my error log, the torch binaries from Isaac Sim failed to load. But as for whether this is an Isaac Sim problem or if it could go deeper is beyond me. I'm also trying to find a solution to this issue.

toenails6, 2024-12-25:
I just reinstalled Isaac Sim 4.2.0 binaries. The following are the binaries install output logs: 

```
(isaaclab) PS C:\Users\renda\IsaacLab> & C:/Users/renda/anaconda3/envs/isaaclab/python.exe c:/Users/renda/IsaacLab/source/standalone/tutorials/00_sim/spawn_prims.py
[WARN][AppLauncher]: There are no arguments attached to the ArgumentParser object. If you have your own arguments, please load your own arguments before calling the `AppLauncher.add_app_launcher_args` method. This allows the method to check the validity of the arguments and perform checks for argument names.
[INFO][AppLauncher]: Loading experience file: C:\Users\renda\IsaacLab\source\apps\isaaclab.python.kit
[Info] [carb] Logging to file: c:/users/renda/isaaclab/_isaac_sim/kit/logs/Kit/Isaac-Sim/4.2/kit_20241225_105113.log
2024-12-25 15:51:13 [0ms] [Warning] [omni.kit.app.plugin] No crash reporter present, dumps uploading isn't available.
2024-12-25 15:51:15 [1,281ms] [Warning] [gpu.foundation.plugin] Skipping unsupported non-NVIDIA GPU: Intel(R) RaptorLake-S Mobile Graphics Controller
2024-12-25 15:51:15 [1,281ms] [Warning] [gpu.foundation.plugin] Skipping unsupported non-NVIDIA GPU: Intel(R) RaptorLake-S Mobile Graphics Controller

|---------------------------------------------------------------------------------------------|
| Driver Version: 566.36        | Graphics API: Vulkan
|=============================================================================================|
| GPU | Name                             | Active | LDA | GPU Memory | Vendor-ID | LUID       |
|     |                                  |        |     |            | Device-ID | UUID       |
|     |                                  |        |     |            | Bus-ID    |            |
|---------------------------------------------------------------------------------------------|
| 0   | NVIDIA GeForce RTX 4090 Laptop.. | Yes: 0 |     | 16050   MB | 10de      | a0700100.. |
|     |                                  |        |     |            | 2757      | 25e58ab1.. |
|     |                                  |        |     |            | 1         |            |
|---------------------------------------------------------------------------------------------|
| 1   | Intel(R) RaptorLake-S Mobile G.. |        |     | 32612   MB | 8086      | fe6c0100.. |
|     |                                  |        |     |            | a788      | 868088a7.. |
|     |                                  |        |     |            | N/A       |            |
|=============================================================================================|
| OS: Windows 11 Home, Version: 10.0 (24H2), Build: 26100, Kernel: 10.0.26100.2605
| Processor: Intel(R) Core(TM) i9-14900HX | Cores: 24 | Logical: 32
|---------------------------------------------------------------------------------------------|
| Total Memory (MB): 65224 | Free Memory: 44100
| Total Page/Swap (MB): 74440 | Free Page/Swap: 49701
|---------------------------------------------------------------------------------------------|
2024-12-25 15:51:17 [3,594ms] [Error] [omni.ext._impl._internal] Failed to import python module omni.isaac.core from c:\users\renda\isaaclab\_isaac_sim\exts\omni.isaac.core. Error: [WinError 126] The specified module could not be found. Error loading "c:/users/renda/isaaclab/_isaac_sim/exts/omni.isaac.ml_archive/pip_prebundle/torch\lib\fbgemm.dll" or one of its dependencies.. Traceback:
Traceback (most recent call last):
  File "C:\Users\renda\IsaacLab\_isaac_sim/kit/kernel/py\omni\ext\_impl\_internal.py", line 187, in _custom_importer
    return _import_public(ext_module.path, ext_module.name, reload_enabled)
  File "C:\Users\renda\IsaacLab\_isaac_sim/kit/kernel/py\omni\ext\_impl\_internal.py", line 94, in _import_public
    module = import_module(module_name)
  File "C:\Users\renda\IsaacLab\_isaac_sim/kit/kernel/py\omni\ext\_impl\custom_importer.py", line 76, in import_module
    return importlib.import_module(name)
  File "C:\Users\renda\anaconda3\envs\isaaclab\lib\importlib\__init__.py", line 126, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
  File "<frozen importlib._bootstrap>", line 1050, in _gcd_import
  File "<frozen importlib._bootstrap>", line 1027, in _find_and_load
  File "<frozen importlib._bootstrap>", line 1006, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 688, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 883, in exec_module
  File "<frozen importlib._bootstrap>", line 241, in _call_with_frames_removed
  File "c:/users/renda/isaaclab/_isaac_sim/exts/omni.isaac.core/omni/isaac/core/__init__.py", line 11, in <module>
    from omni.isaac.core.physics_context.physics_context import PhysicsContext
  File "c:/users/renda/isaaclab/_isaac_sim/exts/omni.isaac.core/omni/isaac/core/physics_context/__init__.py", line 9, in <module>
    from omni.isaac.core.physics_context.physics_context import PhysicsContext
  File "c:/users/renda/isaaclab/_isaac_sim/exts/omni.isaac.core/omni/isaac/core/physics_context/physics_context.py", line 15, in <module>
    from omni.isaac.core.utils.prims import get_prim_at_path, get_prim_path, is_prim_path_valid
  File "c:/users/renda/isaaclab/_isaac_sim/exts/omni.isaac.core/omni/isaac/core/utils/prims.py", line 24, in <module>
    from omni.isaac.core.utils.types import SDF_type_to_Gf
[... cut ...]
2024-12-25 15:52:08 [54,723ms] [Warning] [omni.graph.core.plugin] Could not find category 'Replicator:Annotators' for removal
2024-12-25 15:52:08 [54,723ms] [Warning] [omni.graph.core.plugin] Could not find category 'Replicator:Annotators' for removal
2024-12-25 15:52:08 [54,723ms] [Warning] [omni.graph.core.plugin] Could not find category 'Replicator:Annotators' for removal
2024-12-25 15:52:08 [54,724ms] [Warning] [omni.graph.core.plugin] Could not find category 'Replicator:Core' for removal
2024-12-25 15:52:08 [54,818ms] [Warning] [carb] Recursive unloadAllPlugins() detected!
```

kellyguo11, 2024-12-25:
Hi there, we have seen this error pop up on some Windows systems, please try the solution in the first point here - https://docs.omniverse.nvidia.com/isaacsim/latest/known_issues.html#general.

toenails6, 2024-12-25:
That was actually exactly the solution. 
Kind of stupid on my end not to think of checking up known issues, but in my defense it was kind of difficult to index this from Google using the top few lines of the error log output. 
Okay, also stupid not to changed some search keywords. 
But, nonetheless, I have modified this thread to make it more likely to be indexed by others in the future. 
Thanks a lot! 

> Hi there, we have seen this error pop up on some Windows systems, please try the solution in the first point here - https://docs.omniverse.nvidia.com/isaacsim/latest/known_issues.html#general.

closed 2024-12-25 by toenails6