isaac-sim/IsaacLab #1593: [Question] spawn different robots in different envs

BRoln7, 2024-12-24:
### Question

I can use the `multi_asset.py` demo file from the Isaac Lab demo folder to randomly spawn different robots in different environments. However, when I try the same in my manager-based reinforcement learning environment, it fails. Specifically, all the robots generated in the environments are the same, and it is the last one in the usd_path list.

kamzero, 2024-12-27:
I have encountered the same issue.

RandomOakForest, 2025-01-03:
Thanks for posting this. The team has added it to the list of improvements for an upcoming release.

kellyguo11, 2025-05-07:
Could you please share the error log for the error you are hitting? Note that although we can spawn different robots in the scene, each Articulation object can only contain robots with the same topology.

maxkra15, 2026-09-22:
Thanks @BRoln7 for reporting that manager-based scenes selected only the last USD variant.

Current PhysX manager-based `InteractiveScene` clone planning supports per-environment robot variants through `MultiUsdFileCfg` or `MultiAssetSpawnerCfg`; this workflow was added in [#5528](https://github.com/isaac-sim/IsaacLab/pull/5528). The maintained `scripts/demos/multi_asset.py` example uses ANYmal-C and ANYmal-D. For random rather than round-robin assignment, set `scene_cfg.clone_cfg.clone_strategy = cloner.random` before constructing the scene.

Variants behind one `Articulation` must have compatible links, joints, collision-body counts, and names. Newton currently requires a uniform body layout across worlds, and structurally different robots remain tracked in [#1733](https://github.com/isaac-sim/IsaacLab/issues/1733). We are closing this question with the current PhysX workflow. Please reopen with a minimal Isaac Sim 6.1 reproduction if compatible PhysX variants still all resolve to the same USD.

closed 2026-09-22 by maxkra15