isaac-sim/IsaacLab #1595: [Bug Report] Training on an MARL environment does not populate videos/train.py

Rishi-V, 2024-12-24:
### Describe the bug
I am working on an MARL project/paper and am trying to run the provided Cart Double Pendulum environment on Isaac Lab.
Command to run: `./isaaclab.sh -p source/standalone/workflows/rl_games/train.py --task Isaac-Cart-Double-Pendulum-Direct-v0 --num_envs 1024 --headless --video`
Outcome: Train successfully which can be seen through the printed messages and populated folder in `logs/rl_games/cart_double_pendulum_direct`. 
Problem: However, I cannot see any saved videos in `videos/train`.

On the flip side, running a non-MARL task (e.g. running Isaac-Cartpole-Direct-v0) does have videos populated in `vidoes/train`.

### System Info
- Commit: dd044a4
- Isaac Sim Version: 4.2.0.2
- OS: Ubuntu 20.04
- GPU: RTX 4090
- CUDA: 12.6
- GPU Driver: 560.35.03

### Fix
I was able to fix this by modifying train.py to move `if isinstance(env.unwrapped, DirectMARLEnv): env = multi_agent_to_single_agent(env)` [link](https://github.com/isaac-sim/IsaacLab/blob/dd044a4977722d85e966c5b63949093c9b14667f/source/standalone/workflows/rl_games/train.py#L145) to before `if args_cli.video` as opposed to afterwards.

I see that rsl_rl, sb3, and skrl have the same pattern and therefore implemented the same solution for their train.py's and play.py's. I verified with skrl that it has the same problem beforehand and is fixed with the change. I have not verified rls_rl and sb3 but am guessing the same change helps. Following the contributing code guidelines I have created this issue and will submit a pull request accordingly with the changes.

RandomOakForest, 2025-01-04:
Thanks for submitting this bug report. The team is following up on it.

Rishi-V, 2025-01-04:
I think this was just merged yesterday so it should be resolved / closed?

closed 2025-01-05 by RandomOakForest