isaac-sim/IsaacLab #1604: [Question] Proper way to implement domain randomization on actuator gains?

sdfzz, 2024-12-27:
Hi,

I'm trying to implement domain randomization on Unitree G1 robot.

To randomize actuator gains, I've added the following to '/manager_based/locomotion/velocity/config/g1/flat_env_cfg.py':

```
@configclass
class G1Events(EventCfg):    
    actuator_gain = EventTerm(
        func=mdp.randomize_actuator_gains,
        mode="startup",
        params={
            "asset_cfg": SceneEntityCfg("robot", joint_names=".*"),
            "stiffness_distribution_params": (0., 0.),
            "damping_distribution_params": (0., 0.),
            "operation": "add",  
            "distribution": "uniform" 
        },
    )
```
It is my understanding that this should NOT affect the training since I'm adding zero to actuator stiffness and damping gains.

However, when I run the training, with this added randomization, I get:

Learning iteration 243/3000
Mean reward: -10.49

Without this randomization:

Learning iteration 243/3000                       
Mean reward: 16.69

So, the new randomization is affecting the training somehow...I've tried this multiple times, and got the same result.

Unless this is a bug, I feel like I'm missing something

Can anyone help?

Any help would be appreciated

Regards,

Steve

sdfzz, 2024-12-27:
Hmm, after looking at events.py, I've noticed that (line 338-345):

   ```
     # Randomize stiffness
        if stiffness_distribution_params is not None:
            stiffness = actuator.stiffness[env_ids].clone()
            stiffness[:, actuator_indices] = asset.data.default_joint_stiffness[env_ids][:, global_indices].clone()
            randomize(stiffness, stiffness_distribution_params)
            actuator.stiffness[env_ids] = stiffness
            if isinstance(actuator, ImplicitActuator):
                asset.write_joint_stiffness_to_sim(stiffness, joint_ids=actuator.joint_indices, env_ids=env_ids)
```
Why are we loading default_joint_stiffness before applying randomization?

It seems that this overwrites the stiffness/damping defined in the asset, thus affecting the training

Regards,

Steve

kellyguo11, 2024-12-30:
Thanks for the feedback, I think it makes sense to separate the default values from the randomizations, similar to the proposal in https://github.com/isaac-sim/IsaacLab/pull/1416.

AntoineRichard, 2026-08-27:
@sdfzz thanks for tracing this all the way down to the reseeding behavior — that was exactly the bug.

`randomize_actuator_gains` no longer reseeds from the USD-derived `default_joint_stiffness`. It now snapshots per-actuator defaults (`default_actuator_stiffness` / `default_actuator_damping`) from the configured actuator/controller gains when the term is constructed, and randomizes relative to those (`source/isaaclab/isaaclab/envs/mdp/events.py`), so an `add` of `(0, 0)` is a genuine no-op for implicit and explicit actuators alike. That landed in https://github.com/isaac-sim/IsaacLab/pull/6839.

Closing as fixed — if you still see a reward gap on a current release, please reopen.

---
*🤖 This issue was reviewed as part of an AI-assisted triage of the Isaac Lab backlog, with automated verification against the current codebase. If we got this wrong, please reopen — we'd rather hear about it!*

closed 2026-08-27 by AntoineRichard