isaac-sim/IsaacLab #1605: [Question] How is the amount of data collected by the recorder function determined?

H-Hisamichi, 2024-12-27:
Dear All,

I’m using the newly added `recorder` feature, but I’ve noticed some inconsistencies in the HDF5 data it generates.
When using `mdp.PostStepStatesRecorderCfg` to collect data after each step with `num_envs=16`, the amount of data collected in each instance varies significantly. For example, the first dataset, `demo_0`, only contains 10 rows, while `demo_62` contains 3000 rows. The dimensions of the collected data (joint_position, joint_velocity, root_pose, root_velocity) match those specified in the documentation.
According to the recorder manager’s description, the amount of data collected should match the `num_envs`.
How does the recorder function determine the number of data entries collected at one time?

Thank you!

RandomOakForest, 2025-01-04:
Thank you for posting this. We definitively should have a better way to explain this in our docs. I have tagged this for the team to improve upon. We will aim to reply to this soon.

H-Hisamichi, 2025-01-04:
Hello @RandomOakForest,

Thank you for your response.
This question has resolved itself. It seems that there was a difference in the amount of data collected because some actors were not able to complete the episode due to insufficient training of my model.
I look forward to the documentation becoming even more comprehensive!

closed 2025-02-15 by RandomOakForest