isaac-sim/IsaacLab #1606: [Bug Report] ERROR: Could not find a version that satisfies the requirement robomimic however v0.2.0 is in the robomimic repository

mctouch, 2024-12-28:
### Describe the bug

Running ./docker/container.py start results in error  ERROR: Could not find a version that satisfies the requirement robomimic however v0.2.0 is in the robomimic repository

goal is to run.
docker build -t isaacrayimage -f ./source/standalone/workflows/ray/cluster_configs/Dockerfile .

### Steps to reproduce

```
~/code/IsaacLab$ ./docker/container.py start
...
146.9     Preparing wheel metadata: finished with status 'done'
146.9 ERROR: Could not find a version that satisfies the requirement robomimic (unavailable) (from omni-isaac-lab-tasks[all]) (from versions: 0.1.0, 0.2.0, 0.3.0)
146.9 ERROR: No matching distribution found for robomimic (unavailable)
146.9 WARNING: You are using pip version 21.2.1+nv1; however, version 24.3.1 is available.
146.9 You should consider upgrading via the '/workspace/isaaclab/_isaac_sim/kit/python/bin/python3 -m pip install --upgrade pip' command.
[+] Running 0/1 an error running python
 ⠹ Service isaac-lab-base  Building                                                       135.2s
failed to solve: process "/bin/bash -c ${ISAACLAB_PATH}/isaaclab.sh --install" did not complete successfully: exit code: 1
```

For more information on this, check: https://www.markdownguide.org/extended-syntax/#fenced-code-blocks

-->

### System Info

Describe the characteristic of your environment:

- Commit: [e.g. 8f3b9ca]
~/code/IsaacLab$ cat VERSION
1.3.0
- OS: Ubuntu 22.04
- GPU: RTX A2000
- CUDA:  12.7
- GPU Driver: nvidia-smi --query-gpu=driver_version --format=csv,noheader --id=0
566.03

### Additional context

Add any other context about the problem here.

### Checklist

- [ x] I have checked that there is no similar issue in the repo (**required**)
- [ x] I have checked that the issue is not in running Isaac Sim itself and is related to the repo

### Acceptance Criteria

Add the criteria for which this task is considered **done**. If not known at issue creation time, you can add this once the issue is assigned.

- [ ] Install of base docker container works
- [ ] Ability to run  docker build -t isaacrayimage -f ./source/standalone/workflows/ray/cluster_configs/Dockerfile . sucessfully

kellyguo11, 2025-01-02:
@hhansen-bdai

hhansen-bdai, 2025-01-02:
Hey @mctouch ,

I just tried out setting `v0.2.0` as the robomimic branch to checkout, and was able to get the correct version installed.
```
root@WS36:/workspace/isaaclab# python
Python 3.10.14 (main, Apr 12 2024, 18:16:13) [GCC 7.3.1 20180303 (Red Hat 7.3.1-5)] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import robomimic
>>> robomimic.__version__
'0.2.0'
```

I was able to do this by altering the file `IsaacLab/source/extensions/omni.isaac.lab_tasks/setup.py` on line 55.
```
# Check if the platform is Linux and add the dependency
if platform.system() == "Linux":
    EXTRAS_REQUIRE["robomimic"].append("robomimic@git+https://github.com/ARISE-Initiative/robomimic.git@v0.2.0")
```
The string itself is easy to mess up with tiny errors (I had an accidental `.` messing it up at first), so perhaps you can try this exactly and let me know what happens. Thanks :smile:

mctouch, 2025-01-03:
Thanks, The code was missing the version @v0.2.0 at the end of the git+https url and its at least trying to pull the code however I am getting RPC failed; errors when it tried the pull but that's probably something to do with my network being on a tropical island in the Philippines or some dodgy git config I've added. I'll try the pure ssh url.

The setup.py should probably be updated to use the latest pip
 WARNING: You are using pip version 21.2.1+nv1; however, version 24.3.1 is available.
    EXTRAS_REQUIRE["robomimic"].append("robomimic@git+https://github.com/ARISE-Initiative/robomimic.git")

hhansen-bdai, 2025-01-03:
@mctouch Cool, happy to help. IsaacLab is lockstep with IsaacSim on Python versioning (because they're upstream), so we try not to touch versioning for things related to that if possible. Appreciate the note though :smile:

mctouch, 2025-01-04:
@hhansen-bdai I am using Ubuntu 22.04 on WSL on my local workstation and even manually trying to clone robomimic fails with the same error RPC failed; which is normally network + large file. I know I can increase the timeouts in gitconfig but this doesn't seem to work even on a local WSL or windows terminal pull. 

(base) PS D:\code\IsaacLab> git clone https://github.com/ARISE-Initiative/robomimic.git
Cloning into 'robomimic'...
remote: Enumerating objects: 3989, done.
remote: Counting objects: 100% (75/75), done.
remote: Compressing objects: 100% (52/52), done.
error: RPC failed; curl 92 HTTP/2 stream 5 was not closed cleanly: CANCEL (err 8)
error: 2723 bytes of body are still expected
fetch-pack: unexpected disconnect while reading sideband packet
fatal: early EOF
fatal: fetch-pack: invalid index-pack output

This repo might be a candidate for git lfs as it gets stuck on 14%, even trying on my remote server with the fastest internet connection in Australia chokes on the robomimic repo a bit. Or a zip file download... in the setup.py

My understanding is that './docker/container.py start' starts a build container runs IsaacLab/source/extensions/omni.isaac.lab_tasks/setup.py which does the git clone https from within the build docker container. I have a server with A100's I can try this on however it will take some setup as I will have to use an X11 server on window, punch a hole in the firewall and export the DISPLAY to see the kit window.

hhansen-bdai, 2025-01-06:
@mctouch I'm sorry to hear things are still not working for you :frowning_face: . Unfortunately we can't alter robomimic (to be LFS) because that repo is owned by a different group. As we haven't heard issues from other people (please lmk if I'm wrong about that, @kellyguo11) I think the best option might be for you to remove the robomimic install in setup.py, download it in whatever way you prefer, and then include it in the copied files to be installed once inside the container.

Sorry this is a hassle for you!

mctouch, 2025-01-09:
Yes thanks I used git ssh to download and then ran up my own git server, I guess its a wsl networking issue. Also running it on my A100 server remotely with X11

hhansen-bdai, 2025-01-09:
Awesome, glad this worked out! I'll mark this issue as resolved

closed 2025-01-09 by hhansen-bdai