isaac-sim/IsaacLab #1618: [Bug Report] Result of RigidBodyViw.get_accelerations() is Incorrect in some cases

Csfalpha, 2025-01-02:
### Describe the bug

Hi everyone,
I found that the acceleration result of a rigid body is incorrect in some cases when I use RigidBodyViw.get_accelerations().  I collected the pose, speed and acceleration of the base primitive of the robot dog ANYMAL_C under static conditions. Theoretically the acceleration should be 0, but the actual results deviate greatly.

Below is the code to collect data.

```python
"""Launch Isaac Sim Simulator first."""

import argparse
import os

from omni.isaac.lab.app import AppLauncher

# add argparse arguments
parser = argparse.ArgumentParser(description="Tutorial on adding sensors on a robot.")
parser.add_argument("--num_envs", type=int, default=2, help="Number of environments to spawn.")
# append AppLauncher cli args
AppLauncher.add_app_launcher_args(parser)
# parse the arguments
args_cli = parser.parse_args()

# launch omniverse app
app_launcher = AppLauncher(args_cli)
simulation_app = app_launcher.app

"""Rest everything follows."""

import torch
import numpy as np

import omni.isaac.lab.sim as sim_utils
from omni.isaac.lab.assets import ArticulationCfg, AssetBaseCfg, Articulation
from omni.isaac.lab.scene import InteractiveScene, InteractiveSceneCfg
from omni.isaac.lab.utils import configclass

##
# Pre-defined configs
##
from omni.isaac.lab_assets.anymal import ANYMAL_C_CFG  # isort: skip

@configclass
class SensorsSceneCfg(InteractiveSceneCfg):
    """Design the scene with sensors on the robot."""

    # ground plane
    ground = AssetBaseCfg(prim_path="/World/defaultGroundPlane", spawn=sim_utils.GroundPlaneCfg())

    # lights
    dome_light = AssetBaseCfg(
        prim_path="/World/Light", spawn=sim_utils.DomeLightCfg(intensity=3000.0, color=(0.75, 0.75, 0.75))
    )

    # robot
    robot: ArticulationCfg = ANYMAL_C_CFG.replace(prim_path="{ENV_REGEX_NS}/Robot")

def run_simulator(sim: sim_utils.SimulationContext, scene: InteractiveScene):
    """Run the simulator."""
    # Define simulation stepping
    sim_dt = sim.get_physics_dt()
    sim_time = 0.0
    count = 0

    # Simulate physics
    while simulation_app.is_running():
        # Apply default actions to the robot
        # -- generate actions/commands
        targets = scene["robot"].data.default_joint_pos
        # -- apply action to the robot
        scene["robot"].set_joint_position_target(targets)
        # -- write data to sim
        scene.write_data_to_sim()
        # perform step
        sim.step()
        # update sim-time
        sim_time += sim_dt
        count += 1
        # update buffers
        scene.update(sim_dt)

        if count == 12000:
            stop = 1
        # save to file
        if count % 1 == 0:
            for i in range(2):
                root_state = scene["robot"].data.root_state_w[i, :].tolist()
                body_state = scene["robot"].data.body_state_w[i, :].tolist()
                body_acc   = scene["robot"].data.body_acc_w[i, :].tolist()
                num_instances = len(body_state)
                    
                robot_state_path = f"robot_state_{i}.txt"
                if os.path.exists(robot_state_path):
                    mode = "a"
                else:
                    mode = "w"
                with open(robot_state_path, mode) as f:
                    if mode == "w":
                        f.write("count, pos_w_x, pos_w_y, pos_w_z, quat_w_w, quat_w_x, quat_w_y, quat_w_z, lin_vel_w_x, lin_vel_w_y, lin_vel_w_z, ang_vel_w_x, ang_vel_w_y, ang_vel_w_z\n")
                    f.write(str(count) + ", " + ",".join(map(str, root_state)) + "\n")
                    
                for j in range(num_instances):
                    body_state_path = f"body_state_{i}_{j}.txt"
                    if os.path.exists(body_state_path):
                        mode = "a"
                    else:
                        mode = "w"
                    with open(body_state_path, mode) as f:
                        if mode == "w":
                            f.write("count, pos_w_x, pos_w_y, pos_w_z, quat_w_w, quat_w_x, quat_w_y, quat_w_z, lin_vel_w_x, lin_vel_w_y, lin_vel_w_z, ang_vel_w_x, ang_vel_w_y, ang_vel_w_z, lin_acc_w_x, lin_acc_w_y, lin_acc_w_z, ang_acc_w_x, ang_acc_w_y, ang_acc_w_z\n")
                        f.write(str(count) + ", " + ",".join(map(str, body_state[j])) + "," + ",".join(map(str, body_acc[j])) + "\n")

        if (count > 200*1800*1 + 1000):
            break

def main():
    """Main function."""

    # Initialize the simulation context
    sim_cfg = sim_utils.SimulationCfg(dt=0.005, device=args_cli.device)
    sim = sim_utils.SimulationContext(sim_cfg)
    # Set main camera
    sim.set_camera_view(eye=[3.5, 3.5, 3.5], target=[0.0, 0.0, 0.0])
    # design scene
    scene_cfg = SensorsSceneCfg(num_envs=args_cli.num_envs, env_spacing=2.0)
    scene = InteractiveScene(scene_cfg)
    # Play the simulator
    sim.reset()
    # Now we are ready!
    print("[INFO]: Setup complete...")
    # Run the simulator
    run_simulator(sim, scene)

if __name__ == "__main__":
    # run the main function
    main()
    # close sim app
    simulation_app.close()

``` 

### Additional context

robot_state_i.txt is the state of articulation (without acceleration), body_state_i_j.txt is the state of body in articulation (including acceleration), where j=0 corresponds to base.

Visualize the acceleration data of the z-axis as shown below. In a static state, the acceleration is as high as 5 or more, which is obviously unreasonable.
[image]

However, if I comment the code below, the robot dog will fall and acceleration will be correct. 
```python
        targets = scene["robot"].data.default_joint_pos
        # -- apply action to the robot
        scene["robot"].set_joint_position_target(targets)
        # -- write data to sim
        scene.write_data_to_sim()
``` 
[image]
[image]

### Checklist

- [x] I have checked that there is no similar issue in the repo (**required**)
- [ ] I have checked that the issue is not in running Isaac Sim itself and is related to the repo

### Acceptance Criteria

Add the criteria for which this task is considered **done**. If not known at issue creation time, you can add this once the issue is assigned.

- [ ] Figure out why acceleration is wrong
- [ ] Ensure that the acceleration obtained is correct under any circumstances

jtigue-bdai, 2025-01-06:
I have been able to reproduce this. The resulting acceleration seems to be dependent on the pose of the robot during controlled pose.

jtigue-bdai, 2025-01-06:
I have also reproduced behavior with spot and unitree

jtigue-bdai, 2025-01-06:
The problem does not seem to be present if we switch to `ImplicitActuators` instead of the one derived from `IdealPDActuator`

Csfalpha, 2025-01-17:
Thanks! Still using animal_c, I modified the parent class of its actuator to ImplicitActuator, but my results are still wrong. Can you explain in more detail how you modified it?

Csfalpha, 2025-01-17:
[image]
In addition, I noticed that the official documentation indicates that it should not be used. Is there any problem?

jtigue-bdai, 2025-01-17:
In `source/extensions/omni.isaac.lab_assets/omni/isaac/lab_assets/anymal.py` I changed the `actuators` argument of the `ArticualtionCfg` to the point to a new actuator:
```
ANYDRIVE_3_IMPLICIT_ACTUATOR_CFG = ImplicitActuatorCfg(
    joint_names_expr=[".*HAA", ".*HFE", ".*KFE"],
    stiffness={".*": 400.0},
    damping={".*": 50.0},
)
```

The `ImplicitActuator`class isn't technically used, If you use the `ImplicitActuatorCfg` (emphasis on Cfg) the drive parameters are passed directly to the articulation root_physics_view of the configured joints, rather than spawning an ImplicitActuator like the other actuators would. This makes it so the joint drive is handled by the physics solver rather than outside the physics step. Its closer to a "continuous" time dynamics system than a "discrete" time system that real actuator controllers behave like.

KyleM73, 2025-01-24:
+1 using the Spot config the accelerations are wrong, especially for the body link, even when the robot is at rest. The values in `asset.data.body_lin_acc_w` do not match what is happening in sim (or on hardware with the same control inputs). This is a problem if you want to calculate the total force acting on the robot, since the contact sensors don't report tangential force so you can't get the full body wrench using the sensor data alone.

jtigue-bdai, 2025-01-24:
Yeah we have see this issue as well. If you use the imu sensor's linear accelerations you will get one calculated from finite difference. Its noisy though due to the finite difference.

maxkra15, 2026-09-22:
Thanks @Csfalpha for the detailed reproducer. I reran the same ANYmal-C held-default-pose case for 12,000 steps at `dt=0.005` on current `develop` with Isaac Sim 6.2, on both CPU and GPU PhysX. The base stayed at rest and the last 500 steps remained below 0.0012 m/s² on CPU and 0.010 m/s² on GPU, rather than the steady >5 m/s² shown here. `body_acc_w` still reads the PhysX link-acceleration view, so this checks the same quantity. Closing as no longer reproducible on the current runtime; please reopen with a current 6.1/6.2 reproducer if it persists, including the robot configuration and controlled pose.

closed 2026-09-22 by maxkra15