isaac-sim/IsaacLab #1627: [Question] Suggestion for relative position calculation for large size tensor

celestialdr4g0n, 2025-01-04:
### Question

I am trying to write a version of this function but for two large sets of objects. The purpose of my function is close to this get_env_local_pose(). My problem is after implement this function my training is 5 times slower. Though the input size is large: num_env = 4096, num_obj_1 =  60, num_obj_2 = 50. Is there any way I can speed up my training while implement my function?

def get_env_local_pose(env_pos: torch.Tensor, xformable: UsdGeom.Xformable, device: torch.device):
	"""Compute pose in env-local coordinates"""
	world_transform = xformable.ComputeLocalToWorldTransform(0)
	world_pos = world_transform.ExtractTranslation()
	world_quat = world_transform.ExtractRotationQuat()

	px = world_pos[0] - env_pos[0]
	py = world_pos[1] - env_pos[1]
	pz = world_pos[2] - env_pos[2]
	qx = world_quat.imaginary[0]
	qy = world_quat.imaginary[1]
	qz = world_quat.imaginary[2]
	qw = world_quat.real

	return torch.tensor([px, py, pz, qw, qx, qy, qz], device=device)

@torch.jit.script
def get_two_obj_rev_pose(
	obj_1_rot_w: torch.Tensor,  # Shape: (num_envs, num_obj_1, 4)
	obj_1_pos_w: torch.Tensor,  # Shape: (num_envs, num_obj_1, 3)
	obj_2_rot_w: torch.Tensor,    # Shape: (num_envs, num_obj_2, 4)
	obj_2_pos_w: torch.Tensor     # Shape: (num_envs, num_obj_2, 3)
):
	"""
	For each OBB object, calculate the sphere positions in the OBB frame.
	Args:
		obj_1_pos_w (torch.Tensor): Shape: (num_envs, num_obj_1, 3)
		obj_1_rot_w (torch.Tensor): Shape: (num_envs, num_obj_1, 4)
		obj_2_rot_w (torch.Tensor): Shape: (num_envs, num_obj_2, 4)
		obj_2_pos_w (torch.Tensor): Shape: (num_envs, num_obj_2, 3)
	Returns:
		obj_1_rot_b (torch.Tensor): Shape: (num_envs, num_obj_1, num_obj_2, 4)
		obj_1_pos_b (torch.Tensor): Shape: (num_envs, num_obj_1, num_obj_2, 3)
	"""
	num_envs, num_obj_1, _ = obj_1_rot_w.shape
	num_obj_2 = obj_2_rot_w.shape[1]

	# Expand obj_1 and obj_2s to match for pairwise computation
	obj_1_rot_exp = obj_1_rot_w.unsqueeze(2).expand(-1, -1, num_obj_2, -1)  # (num_envs, num_obj_1, num_obj_2, 4)
	obj_1_pos_exp = obj_1_pos_w.unsqueeze(2).expand(-1, -1, num_obj_2, -1)  # (num_envs, num_obj_1, num_obj_2, 3)
	
	obj_2_rot_exp = obj_2_rot_w.unsqueeze(1).expand(-1, num_obj_1, -1, -1)  # (num_envs, num_obj_1, num_obj_2, 4)
	obj_2_pos_exp = obj_2_pos_w.unsqueeze(1).expand(-1, num_obj_1, -1, -1)  # (num_envs, num_obj_1, num_obj_2, 3)

	# Inputs are reshaped to (:, 4) and (:, 3) 
	inv_obj_1_rot_w, inv_obj_1_pos_w = tf_inverse(obj_1_rot_exp, obj_1_pos_exp)

	# Inputs are reshaped to (:, 4) and (:, 3) 
	obj_1_rot_in_obj_2, obj_1_pos_in_obj_2 = tf_combine(inv_obj_1_rot_w, inv_obj_1_pos_w, obj_2_rot_exp, obj_2_pos_exp)

	# Reshape back to original structure
	obj_1_rot_in_obj_2 = obj_1_rot_in_obj_2.view(num_envs, num_obj_1, num_obj_2, 4)  # (num_envs, num_obj_1, num_obj_2, 4)
	obj_1_pos_in_obj_2 = obj_1_pos_in_obj_2.view(num_envs, num_obj_1, num_obj_2, 3)  # (num_envs, num_obj_1, num_obj_2, 3)
	return obj_1_rot_in_obj_2, obj_1_pos_in_obj_2

RandomOakForest, 2025-01-10:
Thank you for posting this. We are reviewing this issue. Have you tried any of [these speed-ups](https://docs.omniverse.nvidia.com/isaacsim/latest/reference_material/speedup_cheat_sheet.html)?

celestialdr4g0n, 2025-01-11:
Thank you for your feedback! I am making fun of my self. I though @torch.jit.script decorator would move all the data of a function to GPU, but no, I need to specific the device.
So most of the performance drop comes from my checking functions after this coordinate transformation.
I accepted the cost for coordinate transformation and reduce the amount of collision object.

As for the speed up check list, I will try changing some parameters to see if it helps. 
And I have a question? In headless mode should I care about the light objects?

RandomOakForest, 2025-01-23:
Good to hear. Headless execution [docs are here](https://isaac-sim.github.io/IsaacLab/main/source/tutorials/03_envs/run_rl_training.html#headless-execution), give it a shot. If something comes up, please post a question or start a new discussion. Thanks.

closed 2025-01-23 by RandomOakForest