isaac-sim/IsaacLab #1631: [Question] How to Compute Forward Kinematics Without Changing the Object's State in the Simulation Environment?

yilin404, 2025-01-05:
Hello,
I am currently writing code to control a robot arm for opening a drawer. After the arm grasps the drawer handle, I want to compute the change in the handle's pose based on the changes in the prismatic joint values corresponding to the drawer. Using this, I aim to calculate the pose of the drawer handle via the drawer's forward kinematics, which will allow me to deduce the trajectory of the robot arm's end effector.

Could anyone suggest how I should approach calculating the forward kinematics of the drawer, given the change in the prismatic joint values? Any guidance on how to implement this would be greatly appreciated!

celestialdr4g0n, 2025-01-05:
If you check the config for that task, you will see that the drawer is an Articulation. The method to get the joint position of the drawer is similar to the manipulator. Articulation().data.(...)

kellyguo11, 2025-01-06:
You can use the [forward](https://github.com/isaac-sim/IsaacLab/blob/main/source/extensions/omni.isaac.lab/omni/isaac/lab/sim/simulation_context.py#L419) API in SimulationContext.

closed 2025-02-15 by RandomOakForest