isaac-sim/IsaacLab #1646: [Question] Factory Environment Live Visualizer

anikmuhury, 2025-01-08:
### Question

Hello,
At first for the great work.
I was working with the contact rich environment you shared.
(https://github.com/isaac-sim/IsaacLab/blob/main/source/extensions/omni.isaac.lab_tasks/omni/isaac/lab_tasks/direct/factory/factory_env_cfg.py)

I could load the environment using this command in command promt.
 ./isaaclab.sh -p source/standalone/environments/random_agent.py --task Isaac-Factory-NutThread-Direct-v0 --num_envs 1
 
 But I can't change camera view or the rendering for this environment in Isaaclab as I found out live visualization is not active. How can I turn it on. (How can I add action manager and observation manager?).
 Here I have attached the ss.
 Thank you.
 
[image]

RandomOakForest, 2025-01-08:
Thanks for posting this. Could you share the versions of Isaac Sim and Isaac Lab you are using? Thanks.

anikmuhury, 2025-01-08:
Thank you for your response. Here I included the versions.
Isaac Sim Version - 4.2.0
Isaac Lab Version - 1.4.0

RandomOakForest, 2025-01-09:
Thanks. Have you passed `--enable_cameras`? An example is included in [this how-to doc](https://isaac-sim.github.io/IsaacLab/main/source/tutorials/03_envs/run_rl_training.html).

anikmuhury, 2025-01-10:
In the Random_agent code, the camera enabled option was not there. I included that but it didn't change anything.

RandomOakForest, 2025-02-15:
Thank you for following up. If the latest releases of Isaac Sim 4.5 and Isaac Lab 2.0 have not fixed this, could you please file a _bug report_ so we can investigate? Thanks.

closed 2025-02-15 by RandomOakForest