isaac-sim/IsaacLab #1647: [Question] Simulate wind/air blowing

luju97, 2025-01-08:
### Question

Hi,

I would like to now whether it is possible to simulate wind / air blowing objects on IsaacLab.

Thank you in advance!

RandomOakForest, 2025-01-08:
Thanks for submitting this question. The [Omniverse Force Fields](https://docs.omniverse.nvidia.com/extensions/latest/ext_force-fields.html) extension can be used to simulate wind effects.

luju97, 2025-01-08:
> Thanks for submitting this question. The [Omniverse Force Fields](https://docs.omniverse.nvidia.com/extensions/latest/ext_force-fields.html) extension can be used to simulate wind effects.

Thank you! Is it possible to simulate controlled air flows (e.g. pressurized air from a nozzle)?

RandomOakForest, 2025-02-15:
Thanks for following up. It may be possible, however, this is a question for the [Isaac Sim forums](https://forums.developer.nvidia.com/c/omniverse/simulation/69). Hope this helps.

closed 2025-02-15 by RandomOakForest