isaac-sim/IsaacLab #1650: Robot still thinks it is in collision even after resetting its articulation back to the origin of the simulation

brandon-lawrence96, 2025-01-09:
### Question

Basic questions, related to robot learning, that are not bugs or feature requests will be closed without reply, because GitHub issues are not an appropriate venue for these.

Advanced/nontrivial questions, especially in areas where documentation is lacking, are very much welcome.

For questions that are related to running and understanding Isaac Sim, please post them at the official [Isaac Sim forums](https://docs.omniverse.nvidia.com/app_isaacsim/app_isaacsim/isaac_sim_forums.html).

Hi. Currently, when a termination event occurs for my robot due to illegal contact with a wall for example, I make the robot "teleport" back to the origin position of the simulation but even after I do that, the robot still thinks that it is in collision as I can see this from the collision visualizations of the different contact sensors of the robot. Why is this so? This is my code snippet for teleporting the robot back in the event of a collision: 

obs, _, terminated, _ = env.step(actions_reordered)

if terminated.item() == 1:
                print("Restart Simulation!")
                # obs, _ = env.get_observations()  # Reset the environment
                root_state = env.env.scene.articulations["robot"].data.default_root_state.clone()

                env.env.scene.articulations["robot"].write_root_pose_to_sim(root_state[:, :7])

                env.env.scene.articulations["robot"].write_root_velocity_to_sim(root_state[:, 7:])
                
                joint_pos, joint_vel = env.env.scene.articulations["robot"].data.default_joint_pos.clone(), 
               env.env.scene.articulations["robot"].data.default_joint_vel.clone()
                env.env.scene.articulations["robot"].write_joint_state_to_sim(joint_pos, joint_vel)

                env.env.scene.articulations["robot"].reset()

Thank you for the assistance.

RandomOakForest, 2025-01-10:
Thanks for posting this. Can you clarify how you are observing the "robot still thinks that it is in collision"? Thanks.

brandon-lawrence96, 2025-01-11:
Hi. Thank you for the reply. I can see that the robot still thinks it is in collision from the red spheres in the isasc sim gui at the robot prims. I have attached a video as well.

https://github.com/user-attachments/assets/8c338646-14d6-4099-af20-bfc62ed67589

RandomOakForest, 2025-01-24:
Could you verify you are reading simulation data after a _PhysX_ or _sim_ step? otherwise, the data as before the reset persists.

closed 2025-02-15 by RandomOakForest