isaac-sim/IsaacLab #1653: [Question] Can visualization markers retain the markers displayed at each step like debug_draw and provide corresponding methods to clear existing markers?

amatsukaze233, 2025-01-09:
### Question

Currently, I need to visualize at least 60,000 and increasing sphere markers and be able to record RGB information with a camera for reinforcement learning observation. 

When using visualization markers, all marker coordinates need to be inputted at each step, and as the number of markers increases, the performance loss becomes greater. 
[image]

When using debug_draw, only the new marker coordinates need to be inputted at each step, making rendering relatively faster. 
[image]

However, the visual effect seen through the camera is very poor when using debug_draw.
[image]

 I didn't quite understand the usage of point instancer in the source code.

RandomOakForest, 2025-01-10:
Thank you for posting this. Could you help us reproduce this issue? could you share the relevant code you are using for this? Thanks.

amatsukaze233, 2025-01-13:
> Thank you for posting this. Could you help us reproduce this issue? could you share the relevant code you are using for this? Thanks.

Sure.Here is the revised code. The raycaster has been modified, but it should not affect reproduction. The robot has been replaced with Franka. To view from the camera's perspective, you need to hide the ee_marker and finger. The workpiece needs to be replaced with a mesh that includes rigid body and collision properties. If you switch to a rigid object, the visualization effect will not be as good.

  ```
  import argparse
    from omni.isaac.lab.app import AppLauncher
    parser = argparse.ArgumentParser(description="Tutorial on using the differential IK controller.")
    parser.add_argument("--num_envs", type=int, default=1, help="Number of environments to spawn.")
    AppLauncher.add_app_launcher_args(parser)
    args_cli = parser.parse_args()
    app_launcher = AppLauncher(args_cli)
    simulation_app = app_launcher.app

    import omni.isaac.lab.sim as sim_utils
    from omni.isaac.lab.assets import AssetBaseCfg,ArticulationCfg
    from omni.isaac.lab.controllers import DifferentialIKController, DifferentialIKControllerCfg
    from omni.isaac.lab.managers import SceneEntityCfg
    from omni.isaac.lab.assets import RigidObject, RigidObjectCfg
    from omni.isaac.lab.markers import VisualizationMarkers
    from omni.isaac.lab.scene import InteractiveScene, InteractiveSceneCfg
    from omni.isaac.lab.utils import configclass
    from omni.isaac.lab.utils.assets import ISAAC_NUCLEUS_DIR
    from omni.isaac.lab.utils.math import (subtract_frame_transforms,quat_mul,quat_from_euler_xyz,euler_xyz_from_quat,
    combine_frame_transforms,quat_from_angle_axis)
    from omni.isaac.lab.actuators.actuator_cfg import ImplicitActuatorCfg
    from omni.isaac.lab.sim.schemas import CollisionPropertiesCfg,define_collision_properties,modify_collision_properties
    from omni.isaac.lab.markers.visualization_markers import VisualizationMarkersCfg
    from omni.isaac.lab.sim.schemas import ArticulationRootPropertiesCfg,modify_articulation_root_properties
    import typing
    from omni.isaac.lab_assets import FRANKA_PANDA_HIGH_PD_CFG, UR10_CFG 
    from omni.isaac.lab.markers.config import FRAME_MARKER_CFG
    from pxr import Gf, Usd, UsdGeom,Sdf, UsdShade,UsdPhysics,Tf,PhysxSchema
    import omni.isaac.core.utils as prims_utils
    from omni.isaac.core.prims import XFormPrimView
    from xinsong import GCR_CFG
    import omni.usd 
    import omni.kit 
    import omni.physx
    import omni.replicator.core as rep
    import omni.replicator.isaac as dr
    from omni.isaac.lab.sensors.ray_caster import RayCasterCfg, patterns
    from omni.isaac.lab.sensors import ContactSensorCfg,TiledCameraCfg,TiledCamera,CameraCfg,Camera
    import numpy as np
    from omni.isaac.debug_draw import _debug_draw
    import carb
    import random
    import torch

    def get_local_transform_xform(prim: Usd.Prim) -> typing.Tuple[Gf.Vec3d, Gf.Rotation]:#, Gf.Vec3d]:
        xform = UsdGeom.Xformable(prim)
        local_transformation: Gf.Matrix4d = xform.GetLocalTransformation()

        translation = local_transformation.ExtractTranslation()
        rotation= local_transformation.ExtractRotation()
        rotation=prims_utils.rotations.gf_rotation_to_np_array(rotation)
        #scale: Gf.Vec3d = Gf.Vec3d(*(v.GetLength() for v in local_transformation.ExtractRotationMatrix()))
        return translation, rotation#, scale

    # Pre-defined configs
    @configclass
    class TableTopSceneCfg(InteractiveSceneCfg):
    
        # ground plane
        ground = AssetBaseCfg(
            prim_path="/World/defaultGroundPlane",
            spawn=sim_utils.GroundPlaneCfg(color=(1,1,1)),
            init_state=AssetBaseCfg.InitialStateCfg(pos=(0.0, 0.0,-1.05)),# 0.0)),#
        )

[... 168 lines cut ...]
                #     paint_marker.visualize(translations=all_sampled_data)
                #     #paint_marker.set_visibility(True)

                #debug_draw
                data_list = sampled_data.tolist()
                tuple_list = [tuple(item) for item in data_list]
                N=len(tuple_list)
                if N > 0:    
                    draw.draw_points(tuple_list, colors*N, sizes*N)

    def main():
        sim_cfg = sim_utils.SimulationCfg(dt=0.04)
        sim = sim_utils.SimulationContext(sim_cfg)
        sim.set_camera_view((-2.5, 2.5, 2.5), (0.0, 0.0, 0.0))
        scene_cfg = TableTopSceneCfg(num_envs=args_cli.num_envs, env_spacing=2.0)
        scene = InteractiveScene(scene_cfg)
        sim.reset()
        print("[INFO]: Setup complete...")
        run_simulator(sim, scene)

    if __name__ == "__main__":
        main()
        simulation_app.close()

```

RandomOakForest, 2025-02-15:
Thanks for following up. I'll move this to our _Discussions_ section for the team to follow up. Thanks.