isaac-sim/IsaacLab #1655: Unable to load the factory environment

yaoshiluo, 2025-01-09:
Hello everyone, 
I am working on`source/extensions/omni.isaac.lab_tasks/omni/isaac/lab_tasks/direct/factory/__init__.py`, I used the command `./isaaclab.sh -p source/standalone/workflows/rl_games/train.py --task=Isaac-Factory-PegInsert-Direct-v0 --num_envs 2` to load the environment, but I encountered the following error. Does anyone know why?
[image]

RandomOakForest, 2025-01-10:
Thanks for posting this. By any chance are you using a docker container? It may be related to how you are mounting your volumes (see [this issue, for example](https://github.com/isaac-sim/IsaacLab/issues/1522)).

yaoshiluo, 2025-01-11:
> 感谢您发布此信息。您是否正在使用 docker 容器？这可能与您安装卷的方式有关（[例如，请参阅此问题](https://github.com/isaac-sim/IsaacLab/issues/1522)）。

Thank you for your attention. I am not using Docker; I am running it locally.

closed 2025-01-30 by yaoshiluo