isaac-sim/IsaacLab #1659: [Bug Report] local_variable referenced before assignment in IsaacLab's articulation wrapper

AntoineRichard, 2025-01-10:
If you are submitting a bug report, please fill in the following details and use the tag [bug].

### Describe the bug

When using the articulation object from omni.isaac.lab, the functions `write_root_com_pose_to_sim` and `write_root_link_to_sim` both result in an error where a variable is called before being assigned.

### Steps to reproduce

Please try to provide a minimal example to reproduce the bug. Error messages and stack traces are also helpful.

One way to reproduce this issue is to modify the task Isaac-Cartpol-Direct-v0, inside `source/extensions/omni.isaac.lab_tasks/omni/isaac/lab_tasks/direct/cartpole/cartpole_env.py`.

Inside this file, modify the function `_reset_idx` (line 126 to 148).
In this function replace the line 146 and 147 by:
```python
        self.cartpole.write_root_com_pose_to_sim(default_root_state[:, :7], env_ids)
        self.cartpole.write_root_link_velocity_to_sim(default_root_state[:, 7:], env_ids)
```

This gives the following error:
```
Error executing job with overrides: []
Traceback (most recent call last):
  File "/workspace/isaaclab/source/extensions/omni.isaac.lab_tasks/omni/isaac/lab_tasks/utils/hydra.py", line 99, in hydra_main
    func(env_cfg, agent_cfg, *args, **kwargs)
  File "/workspace/isaaclab/source/standalone/workflows/rl_games/train.py", line 171, in main
    runner.run({"train": True, "play": False, "sigma": train_sigma})
  File "/workspace/isaaclab/_isaac_sim/kit/python/lib/python3.10/site-packages/rl_games/torch_runner.py", line 133, in run
    self.run_train(args)
  File "/workspace/isaaclab/_isaac_sim/kit/python/lib/python3.10/site-packages/rl_games/torch_runner.py", line 116, in run_train
    agent.train()
  File "/workspace/isaaclab/_isaac_sim/kit/python/lib/python3.10/site-packages/rl_games/common/a2c_common.py", line 1307, in train
    self.obs = self.env_reset()
  File "/workspace/isaaclab/_isaac_sim/kit/python/lib/python3.10/site-packages/rl_games/common/a2c_common.py", line 531, in env_reset
    obs = self.vec_env.reset()
  File "/workspace/isaaclab/source/extensions/omni.isaac.lab_tasks/omni/isaac/lab_tasks/utils/wrappers/rl_games.py", line 334, in reset
    return self.env.reset()
  File "/workspace/isaaclab/source/extensions/omni.isaac.lab_tasks/omni/isaac/lab_tasks/utils/wrappers/rl_games.py", line 234, in reset
    obs_dict, _ = self.env.reset()
  File "/isaac-sim/exts/omni.isaac.ml_archive/pip_prebundle/gymnasium/wrappers/order_enforcing.py", line 61, in reset
    return self.env.reset(**kwargs)
  File "/workspace/isaaclab/source/extensions/omni.isaac.lab/omni/isaac/lab/envs/direct_rl_env.py", line 274, in reset
    self._reset_idx(indices)
  File "/workspace/isaaclab/source/extensions/omni.isaac.lab_tasks/omni/isaac/lab_tasks/direct/cartpole/cartpole_env.py", line 147, in _reset_idx
    self.cartpole.write_root_link_velocity_to_sim(default_root_state[:, 7:], env_ids)
  File "/workspace/isaaclab/source/extensions/omni.isaac.lab/omni/isaac/lab/assets/articulation/articulation.py", line 471, in write_root_link_velocity_to_sim
    quat = self.data.root_link_state_w[local_env_ids, 3:7]
UnboundLocalError: local variable 'local_env_ids' referenced before assignment
```

### System Info

Describe the characteristic of your environment:

- Commit: de76c2e9d193ec42b834b920f00ebc7c2bff622a
- Isaac Sim Version: 4.2
- OS: 22.04
- GPU: A2000
- CUDA: 12.1
- GPU Driver: 535.183.01

### Checklist

- [x] I have checked that there is no similar issue in the repo (**required**)
- [x] I have checked that the issue is not in running Isaac Sim itself and is related to the repo

### Acceptance Criteria

RandomOakForest, 2025-01-10:
Thanks for submitting this. We will review.

closed 2025-01-14 by kellyguo11

RandomOakForest, 2025-01-23:
A fix has been merged. Thank you.

PR #1660 (2025-01-14): Fixes issue where the indices were not created correctly.
# Description

This PR fixes an issue in articulation object from omni.isaac.lab. The functions `write_root_com_pose_to_sim` and `write_root_link_to_sim` both result in an error where a variable is called before being assigned. I checked and there is a test for this, but I'm not sure why they don't catch it...

The two functions have been changed to match the behaviors of `write_root_link_pose_to_sim` and `write_root_com_velocity_to_sim`.

Fixes #1659 

## Type of change

- Bug fix (non-breaking change which fixes an issue)

## Checklis [... cut]
commit a73b63cb (2025-01-14): Fixes issue where the indices were not created correctly. (#1660)

# Description

This PR fixes an issue in articulation object from omni.isaac.lab. The
functions `write_root_com_pose_to_sim` and `write_root_link_to_sim` both
result in an error where a variable is called before being assigned. I
checked and there is a test for this, but I'm not sure why they don't
catch it...

The two functions have been changed to match the behaviors of
`write_root_link_pose_to_sim` and `write_root_com_velocity_to_sim`.

Fixes #1659 

## Type of change

- Bug fix (non-breaking change which fixes an issue)

## [... cut]
commit 0588b49c (2025-01-21): Fixes issue where the indices were not created correctly. (#1660)

# Description

This PR fixes an issue in articulation object from omni.isaac.lab. The
functions `write_root_com_pose_to_sim` and `write_root_link_to_sim` both
result in an error where a variable is called before being assigned. I
checked and there is a test for this, but I'm not sure why they don't
catch it...

The two functions have been changed to match the behaviors of
`write_root_link_pose_to_sim` and `write_root_com_velocity_to_sim`.

Fixes #1659 

## Type of change

- Bug fix (non-breaking change which fixes an issue)

## [... cut]
commit 93744bba (2025-04-14): Fixes issue where the indices were not created correctly. (#1660)

# Description

This PR fixes an issue in articulation object from omni.isaac.lab. The
functions `write_root_com_pose_to_sim` and `write_root_link_to_sim` both
result in an error where a variable is called before being assigned. I
checked and there is a test for this, but I'm not sure why they don't
catch it...

The two functions have been changed to match the behaviors of
`write_root_link_pose_to_sim` and `write_root_com_velocity_to_sim`.

Fixes #1659 

## Type of change

- Bug fix (non-breaking change which fixes an issue)

## [... cut]