isaac-sim/IsaacLab #1661: Implementing soft terrain for legged locomotion

jnskkmhr, 2025-01-11:
My question is somewhat similar to this issue https://github.com/isaac-sim/IsaacLab/issues/90.

I am currently working on implementing a soft terrain contact model (more like a force model) based on resistive force theory described in this paper: https://arxiv.org/abs/1303.7065.

As an initial test, I dropped a robot foot from a height of 0.5m and observed that the terrain can support the object with a certain level of penetration.

My question is: what could be a promising approach to simulate this soft terrain model for legged robot locomotion, particularly for bipedal locomotion? 
I found that simply applying force to the foot does not work effectively due to the resulting moments causing the robot to flip.

Do you have any suggestions for a practical way to implement this soft contact model in IsaacLab? 
I can share my current code, which essentially computes the contact force based on the paper's formulation and applies it to the robot using the articulation API.

https://github.com/user-attachments/assets/9cf388b6-521c-43d8-aa5f-c0d90f778ac4

jnskkmhr, 2025-01-11:
Env Cfg
```python
from __future__ import annotations

import math
import gymnasium as gym
import numpy as np
import torch
from collections.abc import Sequence
import carb

from omni.isaac.lab.envs.common import ViewerCfg
import omni.isaac.lab.sim as sim_utils
import omni.isaac.lab.terrains as terrain_gen
from omni.isaac.lab.assets import ArticulationCfg, AssetBaseCfg
from omni.isaac.lab.assets import RigidObjectCfg, RigidObject
from omni.isaac.lab.envs import DirectRLEnvCfg
from omni.isaac.lab.scene import InteractiveSceneCfg
from omni.isaac.lab.sim import SimulationCfg
from omni.isaac.lab.sensors import ContactSensorCfg
from omni.isaac.lab.terrains import TerrainImporterCfg
from omni.isaac.lab.markers import VisualizationMarkersCfg
from omni.isaac.lab.utils import configclass

# macros 
from omni.isaac.lab.utils.assets import ISAAC_NUCLEUS_DIR, ISAACLAB_NUCLEUS_DIR, NVIDIA_NUCLEUS_DIR
from omni.isaac.lab_assets import ISAACLAB_ASSETS_DATA_DIR
ENV_REGEX_NS = "/World/envs/env_.*"

##
# Pre-defined configs
##
from omni.isaac.lab_tasks.direct.hector.env_cfg.terrain_cfg import SoftTerrain
from .asset_cfg import SPHERE_CFG, HECTOR_FOOT, HECTOR_LEG_CFG, HECTOR_ARTICULATED_FOOT

@configclass
class SoftTerrainCfg(DirectRLEnvCfg):
    # ================================
    # Common configurations
    # ================================
    seed = 42
    episode_length_s = 5.0
    num_steps_per_env = 24 # see rsl cfg (horizon for rollout)
    
    dt=0.002 #500Hz 
    rendering_interval = 10 # 50Hz
    mpc_decimation = 5 # 100Hz
    decimation = 5 # 100Hz (RL)
    
    num_actions = 3
    num_observations = 3
    num_critic_observations = 3
    
    
    # ================================
    # Environment configurations
    # ================================
    
    # simulation settings
    sim: SimulationCfg = SimulationCfg(
        dt=dt, 
        render_interval=rendering_interval, 
        gravity=(0, 0, -9.81),
        disable_contact_processing=True, 
        )
    
    # terrain
    terrain = SoftTerrain
    
    is_articulation = True
[... 271 lines cut ...]
        alpha_z += self.cfg.A00*torch.cos(2*torch.pi*(0*beta/torch.pi)) #A00
        alpha_z += self.cfg.A10*torch.cos(2*torch.pi*(1*beta/torch.pi)) # A10
        alpha_z += self.cfg.B01*torch.sin(2*torch.pi*(1*gamma/(2*torch.pi))) # B01
        alpha_z += self.cfg.B11*torch.sin(2*torch.pi*(1*beta/torch.pi + 1*gamma/(2*torch.pi))) # B11
        alpha_z += self.cfg.B_11*torch.sin(2*torch.pi*(-1*beta/torch.pi + 1*gamma/(2*torch.pi))) # B-11
        
        # calculate alpha_x
        alpha_x = torch.zeros_like(depth)
        alpha_x += self.cfg.C01*torch.sin(2*torch.pi*(1*gamma/(2*torch.pi))) # C01
        alpha_x += self.cfg.C11*torch.sin(2*torch.pi*(1*beta/(torch.pi) + 1*gamma/(2*torch.pi))) # C11
        alpha_x += self.cfg.C_11*torch.sin(2*torch.pi*(-1*beta/(torch.pi) + 1*gamma/(2*torch.pi))) # C-11
        alpha_x += self.cfg.D10*torch.cos(2*torch.pi*(1*beta/(torch.pi))) # D10
        
        # calculate resistive force (feedforward)
        depth_mask = depth < 0 # apply resistive force only when foot is penetrating
        fz = alpha_z * (-depth) * dA * depth_mask * (1e6) #m^3 to mm^3
        
        fx = torch.zeros_like(fz)
        # fx = -alpha_x * (-depth) * dA * depth_mask *  (1e6) #m^3 to mm^3
        fy = torch.zeros_like(fz)
        
        force = torch.stack((fx, fy, fz), dim=-1)
        
        return force
```

jnskkmhr, 2025-01-17:
Hi @RandomOakForest 
Thank you for the response. 

I thought about that too, but I did not understand how to reflect the physics done in warp to IsaacLab simulation (how to integrate robot-particle interaction done by warp to Isaac?) 
Also, as far as I tried both warp and PhysX particle system, running particle sim with legged robot is too expensive. 

Is there a way to apply force computed in warp to every vertices of mesh (like foot)? 
It would be great if I can run sort of like reduced order terrain model (kind of like model which approximate foot-terrain interaction).

RandomOakForest, 2025-02-15:
Thanks for following up. I'll move this into our _Discussions_ section for the team to follow up.