isaac-sim/IsaacLab #1664: [Question] Plans to support Ubuntu 24.04

bhomaidan1990, 2025-01-11:
### Question

Hi, Is there any plans to support Ubuntu 24.04? thanks.

splendidsummer, 2025-01-12:
The same question, actually I am working on Ubuntu 24.04 currently

RandomOakForest, 2025-01-17:
Thanks for posting this. It would help to have more details on the issues you encountered with 24.04. Thank you.

Charles-CongFu, 2025-01-21:
> The same question, actually I am working on Ubuntu 24.04 currently with issacsim and issaclab

Hi. Mind if you share some tutorials on how to make isaac sim and isaac lab work on 24.04? Personally the only guide that I found is through [docker container](https://github.com/arambarricalvoj/nvidia_isaac-sim_ros2_docker) to use isaac sim. I guess by following the installation instruction from Nvidia, one can submit a new docker container with more packages in dockerfile for Isaac Lab, but it's definitely not ideal. 

My laptop has some compatibility issue with Ubuntu 22.04(typical for newer hardware with older kernel). So I'd very much like to install them on 24.04 or its variants.

bhomaidan1990, 2025-01-24:
The problem was in python version, I had to install older version to make it work:

```sh
sudo -E add-apt-repository -y 'ppa:deadsnakes/ppa'
sudo apt update
sudo apt install python3.10
sudo apt install python3.10-venv
python3.10 -m venv isaaclab
source isaaclab/bin/activate
pip install --upgrade pip
# CUDA 12
pip install torch==2.4.0 --index-url https://download.pytorch.org/whl/cu121
pip install isaacsim==4.2.0.2 isaacsim-extscache-physics==4.2.0.2 isaacsim-extscache-kit==4.2.0.2 isaacsim-extscache-kit-sdk==4.2.0.2 --extra-index-url https://pypi.nvidia.com
```

closed 2025-01-24 by bhomaidan1990