isaac-sim/IsaacLab #1666: [Question] How to turn off DLSS for visualizer?

Nate711, 2025-01-13:
### Question

I'm using a 3080 Ti and the environment rendering is quite poor, I think due to DLSS. For instance, I will see ghosting behind moving objects and thin objects like the rail in the cart-pole environment will look very poor. I've searched everywhere in the Isaac sim visualizer settings window for how to turn off DLSS, but I've only found how to change it's quality (performance, balanced, quality, etc). Any help appreciated thanks.

KyleM73, 2025-01-13:
The options are set in the simulation and rendering configs [here](https://github.com/isaac-sim/IsaacLab/blob/9ab52d47da0d2a43a492b835e866a4e75bbc62d1/source/extensions/omni.isaac.lab/omni/isaac/lab/sim/simulation_cfg.py#L174). This can be changed in the `__post_init__` method of the task config (assuming the manager based workflow) as:

`self.sim.render.antialiasing_mode = "TAA"  # doesn't require RTX DLSS`

RandomOakForest, 2025-01-17:
Thanks for submitting this. Let us know if you need further help. Sometimes this can be traced to a driver issue.

closed 2025-02-15 by RandomOakForest